openscad/openscad #2884: Add error command

adrianVmariano, 2019-03-20:
The only existing mechanism for code or libraries to generate errors is assert.  It is currently impossible to generate warnings from user code.  

When assert is used to produce an error it looks like this:

    assert(big_complex_condition, "message for the user explaining the problem")

produces

ERROR: Assertion 'big_complex_condition': "message for user explaining the problem" failed in file test.scad, line 10

This is a rather cluttered output with the text of the condition, extra quote marks, and the word "failed".  It does not look the same as a normal error message.  I propose a new error command that might work like this in a function:

    x <= 0 ? error("x must be positive in function foo"): 
                  do_something(x,y);

and the output would not have extra quote marks, or the word "failed", but would look the same as native error messages:

ERROR: x must be positive in function foo in file test.scad, line10

Perhaps the error message display could give the function name automatically.  

In a module it would be:

    if (x>0) { make_some_geometry } 
    else error("x must be positive for module foo");

For warnings the error command could take a flag, so 

    error("text",warn=true)

would produce a warning.   I was thinking that it made sense for error to be unconditional to make the forms above simple, but it's not clear that unconditional warnings make sense since the code needs to continue.   If it does take a condition the test should be the reverse of assert, namely that when the condition is true, the error or warning occurs.   Perhaps an unconditional error statement and a conditional warning statement?

t-paul, 2019-03-20:
I think we should go with separate `error()` and `warning()` as suggested by @doug-moen in the forum discussion.

Sharma-Hrishabh, 2020-03-12:
Hey @t-paul , I want to work on this issue. It will help me in understanding the parser code.

t-paul, 2020-03-12:
Sure, we can discuss details here and/or on IRC. But note that language changes will require a lot of discussion, even for this one which is relatively simple. The reason for that is those are pretty much one-way changes. There's not much chance to change the calling conventions once the feature is released as that would break all scripts using the feature. While not perfect, we try hard to prevent that.

dearmash, 2025-09-09:
It's been a while since this issue was reported and PR proposed. I'm currently looking for this exact thing, but there's no good way I know of to +1 the issue besides just commenting on it.

dearmash, 2025-09-10:
tbh, I'm hesitant to say adding a new function is absolutely correct, only because it will complicate things for anyone who for some reason already has something like error = translate() foo() bar(); Maybe modeling some sort of error / warning on a sign, who knows. Maybe something like echo_error, or echo_warn, that's a little less likely to already be in use.

I would entertain the idea of echo(level="warn|error|trace") and have more options to echo to act like the underlying LOG(), but I also know that echo has special parameter parsing handling already.

I'm mostly here to add my 2c to say if it were implemented, I would really appreciate it in whatever form it takes.

jordanbrown0, 2025-09-10:
Possible namespace conflicts are always an issue, especially when (as here) adding new syntax.  (let/echo/assert in expression context are **not** functions; they are their own unique syntax.)  It *might* be possible to rewrite that bit of the parser to have it be agnostic to the name, so that those names wouldn't need to be keywords.

AncientRoman, 2026-02-16:
+1. I think a warn() and an error() would be perfect.

I made #6665, but a warn() would be even better for me. See also #6005.

What needs to happen for warn() and error() to be added to the language?
I'd be happy to update my pr to include this or make a new one.

adrianVmariano, 2026-02-16:
At this point I see warnings in OpenSCAD as pointless, since we all run with "stop on first warning"---they are the same as errors.  From a library design perspective, a library author must assume that any warning is fatal for this reason.  The things that seem useful would be an error() that as noted above provides a cleaner output without the extra quotes and clutter, ideally with a syntax like error(cond,message) that displays the message and a traceback if cond is true.  And then some way to do messages that should only be displayed one time like deprecation messages.  But I'm not sure exactly how that could work.  

Of potential value also might be the ability to echo in other colors and hence to create something warning-like on the output, or a lower level of warning that never stops program execution.