openscad/openscad #2885: Option+Arrow keys don't work in 2019.03.03 in Mac OS X

MikeBenza, 2019-03-21:
I downloaded 2019.03.03 and discovered that the Option+Arrow keys don't work to move left and right on word boundaries.  I did a binary search in releases and found that it worked in OpenSCAD-2015.11.15.dmg but not in OpenSCAD-2015.12.02.dmg.  I looked at commits around those dates and found [#1471/c270bfc40d84fd721b6cbd36396e5b0840af8369](https://github.com/openscad/openscad/pull/1471/commits/c270bfc40d84fd721b6cbd36396e5b0840af8369), which appears to be the smoking gun.

I'm working through building locally to see what I can figure out around that change.

MikeBenza, 2019-03-21:
I'm sorry, but I just can't get it to build.  I followed the build steps, but there are still a lot of missing dependencies.  I'm not sure if the setup scripts are broken or my config is broken, but I've never been good with building C/C++.

kintel, 2019-03-21:
How did you install dependencies?

MikeBenza, 2019-03-21:
I tried with the homebrew script.  First it was missing libxslt (to compile), then it couldn't find qscintilla to link (I manually compiled and installed it), then it couldn't find double-conversion to link.  I tried brew link --force double-conversion but that didn't help.  Even after I compiled and installed double-conversion, it still won't build.

```
ld: warning: directory not found for option '-L/Users/mike/Sandbox/personal/openscad/../libraries/install/lib'
ld: library not found for -ldouble-conversion
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [OpenSCAD.app/Contents/MacOS/OpenSCAD] Error 1
```

I'm not sure how relevant the warning is.

kintel, 2019-03-21:
The homebrew script is pure homebrew. If that doesn't work, I'd say you have a bombed homebrew installation.
The error you show is from attempting to build against the locally built libraries.
Note: You cannot combine these two alternatives; it's either 100% homebrew or no homebrew at all.

MikeBenza, 2019-03-21:
> If that doesn't work, I'd say you have a bombed homebrew installation.

Pretty likely.  I'll clean it up and follow up here.

MikeBenza, 2019-03-21:
Follow up: If I undo that change it's unbroken.  I can't say for sure why.  Here's the rundown.

There are four events handled in an option+left arrow key press.

1. Modifiers: Qt::AltModifier, key = Qt::Key_Alt, type = QEvent::KeyPress
2. Modifiers: Qt::AltModifier | Qt::KeypadModifier, key = Qt::Key_Left, type = QEvent::KeyPress
3. Modifiers: Qt::AltModifier | Qt::KeypadModifier, key = Qt::Key_Left, type = QEvent::KeyRelease
4. Modifiers: 0, key = Qt::Key_Alt, type = QEvent::KeyRelease

OpenSCAD correctly identifies the key as an Alt+Left keypress on the second event, but it seems the navigate left isn't working.  It looks like in that PR you added functionality to skip left to the most significant digit or to zero-pad the number.

I think that's an OK functionality on Windows where the standard key to skip left/right works is Ctrl+Left / Ctrl+Right and Alt+Left / Alt+Right are free to use.  On OS X the standard key to skip left/right is Option+Left / Option+Right (where Option is encoded as Alt).  This makes it a real pain to use the keyboard to get around the editor on OS X.

So, I still view this as a bug, but I understand you may not.  Maybe a preference to enable the new behavior would be a good option (pun intended)?

MikeBenza, 2019-03-22:
Alternatively (pun also intended), make the new behavior Option+Shift+Arrow on Mac and Alt+Arrow elsewhere?  Sample commit [here](https://github.com/MikeBenza/openscad/commit/2af1dbb7a1833f27527112bfb97416681cd3ad03).  Let me know if you'd like me to open a PR / do it some other way.

MikeBenza, 2019-03-27:
@kintel any feedback on the above?

kintel, 2019-03-31:
The the OS X standard Option-Arrow keybindings documented anywhere?
In my OS X editor, Ctrl-left/right is bound do word navigation, so I'm unsure what the existing behavior would clash with.

MikeBenza, 2019-03-31:
Documented [here](https://support.apple.com/en-us/HT201236) (search for "Option–Left Arrow").

A quick test of apps I've got open shows the following navigate by word on Option+Left/Right arrow:
- Sublime Text
- Chrome
- IntelliJ
- Slack
- Slic3r
- OpenSCAD 2015.03 ;)
- Arduino
- iTerm2 (though I may have bound that myself)
- Macdown

kintel, 2019-03-31:
Thx. I'm on Emacs, so I never noticed :)

kintel, 2019-03-31:
We have to hijack some reserved combination, but Shift-Option-Arrows look reasonable to me.

omz, 2019-04-03:
Shift-Option-Arrows would clash with the standard way of extending the selection by word on macOS.

Maybe the new functionality could be either Ctrl-Cmd-Arrows or Ctrl-Option-Arrows instead? AFAIK these have no standard behavior on the Mac.

kintel, 2019-04-06:
@omz That's a pretty tricky key combo..

MikeBenza, 2019-04-06:
@kintel I don't find it too hard.  I'm open to most options.

QuackingPlums, 2019-07-05:
I agree with @omz and @MikeBenza; standard OS behaviour should not be overridden to support some application-specific functionality, even if the subset of affected users are in the minority.

I've opened a new issue on this (#2984) because we appear to flip-flop between two unacceptable solutions here. Both **Option-arrow** and **Shift-Option-arrow** are standard key combinations on OSX. Whatever functionality we're trying to enable here needs to go find another combination to hijack. ;)

gingerbeardman, 2021-05-18:
This trips me up every single day.

gingerbeardman, 2021-05-22:
I installed [sublime-openscad](https://github.com/adcurtin/sublime-openscad) plugin (syntax highlighting and code completition) for Sublime Text and now use that as an external editor. Problem solved!

QuackingPlums, 2021-05-22:
I’ve got several other code editors installed that are far better editors, but I get recurring errors that come and go whenever I use OpenSCAD in auto-render mode. Objects randomly fail to render, throw up errors, or cause other objects to fail to render. I’ve spent days down rabbit holes trying to find out why objects that rendered fine the night before suddenly won’t work, only to find that that if I just turn off the auto render option then the error goes away. 
This is one level of friction too far for something that should be the least stressful part of coding. 😁

nophead, 2021-05-22:
If you turn on OpenSCADs editor window do you see corrupt versions of your
file. Seems like OpenSCAD is seeing part written files.

QuackingPlums, 2021-05-22:
Not that I’ve noticed. F5 renders the object fine, but then I get an error when I hit F6. The error only appears once, but despite getting no errors on the second attempt to render, the object doesn’t appear. If I disable the auto refresh option and hit F6 again then the object renders fine. 
I guess it’s possible that OpenSCAD is seeing a partial file and that turning the option off reloads the full file but the error doesn’t indicate this. It refers to some include that is not part of my source.

nophead, 2021-05-22:
In that case I don't think it has anything to do with the external editor.
Some designs will error on F6 if you have done F5 first because F5 caches
polysets that have been grid snapped and are not manifold. If you do F6
before F5 or you clear the cache before F6 then the intermediate polysets
are not used and CGAL is happy.

The way to avoid it is to avoid close vertices that get grid snapped
together but sometimes that is almost impossible. For example, when two
curved surfaces intersect.