openscad/openscad #2888: Fontconfig warning, unknown element "blank"

obriencj, 2019-03-22:
When using openscad from the CLI from macOS, the following warning is emitted

```
certainly:keychain siege$ openscad cheapjeep.scad -o cheapjeep.stl
Fontconfig warning: "/Applications/OpenSCAD.app/Contents/Resources/fonts/fonts.conf", line 84: unknown element "blank"
```

I've taken to simply removing the ```<blank>...</blank>``` segment from that file, as it seems to have no impact. However I believe that it would be better if CLI interaction with openscad didn't produce spurious warnings.

```
OpenSCAD Version: 2019.03.03 (git abd9ae92)
System information: Mac OS X Version 10.14.3 (Build 18D109) iMac17,1 4 CPUs 24.00 GB RAM 
User Agent: OpenSCAD/2019.03.03 (git abd9ae92) (Mac OS X Version 10.14.3 (Build 18D109) iMac17,1)
Compiler: Clang "10.0.0 (clang-1000.10.44.4)" 64bit
MinGW build: No
Debug build: No
Boost version: 1_65_1
Eigen version: 3.3.7
CGAL version, kernels: 4.13, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.9.7
QScintilla version: 2.10.8
InputDrivers: 
GLib version: 2.56.3
lodepng version: 20180910
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.9.1
harfbuzz version: 2.3.1
lib3mf version: 1.8.0
Application Path: /Applications/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/siege/Documents
User Documents Path: /Users/siege/Documents
Resource Path: /Applications/OpenSCAD.app/Contents/Resources
User Library Path: /Users/siege/Documents/OpenSCAD/libraries
User Config Path: /Users/siege/Library/Application Support/OpenSCAD
Backup Path: /Users/siege/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/Users/siege/Documents/OpenSCAD/libraries
/Applications/OpenSCAD.app/Contents/Resources/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
/usr/share/fonts
/Users/siege/.local/share/fonts
/Users/siege/.fonts
/usr/X11/lib/X11/fonts
/System/Library/Fonts
/Library/Fonts
/Users/siege/Library/Fonts
GLEW version: 2.1.0
OpenGL Version: 2.1 ATI-2.4.10
GL Renderer: AMD Radeon R9 M395X OpenGL Engine
GL Vendor: ATI Technologies Inc.
RGBA(8888), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(888-1), depth(-1), stencil(-1)

```

BenBergman, 2020-09-22:
I found this particularly confusing as I was running openscad CLI on a Linux machine at first and my STL render only took a few minutes, but on macOS my render took considerably longer. This warning popped up near the start and had me thinking the process locked up or something.

gingerbeardman, 2021-06-16:
Still a problem in the latest builds.

AlexDep, 2023-06-27:
The problem is still present

Openscad File : 

```
message = "b";//[a,b,c,d]
textsize = 50;//
myfont = "Stencil";
height = 10;

linear_extrude(height){
    text(str(message), font=str(myfont),size=textsize);
}
```

Command line : 

` openscad -o "output.stl" "Example.stl"  -D message="a" -D textsize=10 -D myfont="Arial:style=Regular"`

Warning : 

`Fontconfig warning: "/Applications/OpenSCAD.app/Contents/Resources/fonts/fonts.conf", line 84: unknown element "blank"`

Still the font is applied to the text in the generated file.

fedekrum, 2024-02-07:
Is still there. Just saw it

docwhat, 2024-06-19:
From [FontConfig issue 250](https://gitlab.freedesktop.org/fontconfig/fontconfig/-/issues/250)

> Please simply remove those elements and also keep config files updated and/or use the same version of fontconfig on the system.
> ...
> As that warning says, blank element isn't used anymore. apparently fonts.conf bundled by OpenSCAD is totally outdated. they have to update it. that's not our issue.

The problem is that the OpenSCAD in Homebrew is using the Homebrew fontconfig but is supplying its own `fonts` configuration directory. This can be worked around in one of these ways:

- `export FONTCONFIG_PATH=$(brew --prefix)/etc/fonts/`
- `env FONTCONFIG_PATH=$(brew --prefix)/etc/fonts/ openscad ...`
- `mv /Applications/OpenSCAD.app/Contents/Resources/fonts /Applications/OpenSCAD.app/Contents/Resources/fonts.bak; ln -nsf $(brew --prefix)/etc/fonts /Applications/OpenSCAD.app/Contents/Resources/`

The real fix is to not supply our own `fonts` directory.

PR #5191 (2024-06-25): Update fontconfig configs for macOS bundling
Fixes #2888
closed 2024-06-25 by kintel