openscad/openscad #2892: library not found for -lqscintilla2_qt4 on Mac OSX

thboomsma, 2019-03-24:
Anybody ever managed to solve this error when compiling OpenScad for MAC?
Please help
>make
ld: library not found for -lqscintilla2_qt4
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [OpenSCAD.app/Contents/MacOS/OpenSCAD] Error

t-paul, 2019-03-24:
As said before in #2880, as the name indicates, this is trying to link a QScintilla built for Qt4 which is probably not installed.

Is this trying a clean build from the OpenSCAD master repository?

thboomsma, 2019-03-24:
Yes it is ...
Obviously my environment is messed up now, but i'm compiling from fresh git clone in different folder and trying to compile with the instructions given.

t-paul, 2019-03-24:
Ideally (if possible) try with a clean homebrew installation and make sure only Qt5 is installed. And `qmake -v` should give something like 'Using Qt version 5.11.3...'.

kintel, 2019-03-24:
Perhaps we need to kill Qt4 support for good and fail early with a loud error if Qt4 is being used?

t-paul, 2019-03-24:
Yep, first commit after release?

thboomsma, 2019-03-25:
Haha .. getting the memo ..
I stripped down the QT4 remnents on my Mac .. and yes it looks better, but i must have stripped abit too far i think ..
on MAKE:
In file included from src/MainWindow.h:6:
objects/ui_MainWindow.h:13:10: fatal error: 'QtGui/QAction' file not found
#include <QtGui/QAction>

t-paul, 2019-03-25:
That is still stuff generated by Qt4, I think. Try `make clean` and re-run `qmake`.

thboomsma, 2019-03-25:
That worked yea .. now stuck at:
dyld: Library not loaded: /Users/theoboomsma/Documents/OpenSCAD_src/../libraries/install/lib/libgettextsrc-0.19.8.dylib

kintel, 2019-03-25:
It still looks like you're using a combination of homebrew and locally built dependencies. Perhaps wipe OpenSCAD_src/../libraries/install and go homebrew-only?

Efroymson, 2019-05-13:
I had a problem similar to the one listed here, was eventually able to resolve it by re-installing home-brew, but my build crashes upon opening a scad file with 👍 
Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff78ef4b66 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff790bf080 pthread_kill + 333
2   libsystem_c.dylib             	0x00007fff78e501ae abort + 127
3   org.qt-project.QtCore         	0x00000001038611c9 0x103849000 + 98761
4   org.qt-project.QtCore         	0x0000000103862914 QMessageLogger::fatal(char const*, ...) const + 202
5   org.qt-project.QtWidgets      	0x0000000102d96890 QWidgetPrivate::QWidgetPrivate(int) + 544
6   org.qt-project.QtWidgets      	0x0000000102e53223 QFramePrivate::QFramePrivate() + 19
7   org.qt-project.QtWidgets      	0x0000000102e55bd5 QAbstractScrollArea::QAbstractScrollArea(QWidget*) + 37
8   libqscintilla2_qt5.13.dylib   	0x0000000100f05d97 QsciScintillaBase::QsciScintillaBase(QWidget*) + 23
9   libqscintilla2_qt5.13.dylib   	0x0000000100ef979c QsciScintilla::QsciScintilla(QWidget*) + 28
10  org.openscad.OpenSCAD         	0x0000000100aa7b0e ScintillaEditor::ScintillaEditor(QWidget*) + 126
11  org.openscad.OpenSCAD         	0x000000010084d3e1 MainWindow::MainWindow(QString const&) + 1617
12  org.openscad.OpenSCAD         	0x000000010083e1b6 gui(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >&, boost::filesystem::path const&, int, char**) + 1830
13  org.openscad.OpenSCAD         	0x0000000100842d05 main + 15957
14  libdyld.dylib                 	0x00007fff78da4015 start + 1

Perhaps I should open a new issue?  I think it is related ...

t-paul, 2019-07-07:
Closing as the original issue seems resolved.

closed 2019-07-07 by t-paul