openscad/openscad #2893: Animation dumps images with wrong file names.

t-paul, 2019-03-25:
Capturing discussion on IRC:

> anyone know how to make openscad dump animation png files as ascii numbers and not just increment past the alphabet
> it has non-character values in the filenames so there's newlines and backspaces in the middle of the filename lol

Fix suggestion:

> QString::arg requires a base as the third parameter for int. Without the base apparently that forced the char to an int, and matched it to arg(char a, int fieldWidth = 0, QChar fillChar = QLatin1Char(' '))
> Changing it to QString filename = QString("frame%1.png").arg(this->anim_step, 5, 10, QChar('0')); should fix it.

rcolyer, 2019-03-25:
Typo fix for clarity: the quoted comment from me explaining the theory of the bug should have read "forced the int to a char".

t-paul, 2019-03-25:
Thanks, updated.

t-paul, 2019-03-26:
Hmm, that seems to depend on some version of Qt or gcc or whatever. I can't reproduce that with a local build, but the AppImage fails.

t-paul, 2019-03-26:
Yep, that change seems to work fine... https://circleci.com/gh/openscad/openscad/2453#artifacts

commit b0214d7f (2019-03-26): Merge pull request #2897 from openscad/animation-filename-fix

Force selection of overloaded arg() method (fixes #2893).
commit ebc62b50 (2019-03-26): Force selection of overloaded arg() method (fixes #2893).
commit 141fb591 (2019-03-29): Force selection of overloaded arg() method (fixes #2893).
closed 2019-03-26 by t-paul