openscad/openscad #2898: Change current folder when opening file from file association or drag-and-drop

korpx, 2019-03-26:
When a .scad file is opened from within OpenSCAD (File, Open...) the folder path of the loaded file is remembered by OpenSCAD as current directory. This is great.

However when I open a .scad file by file association (double clicking the file in the file system) or drag-and-drop it into OpenSCAD the application does not update the current folder path. If I then chose File, Open... I am presented with the folder I last opened a file from, from within OpenSCAD, not the folder where the currently opened file is located.

Would it be possible to have it respect the folder path no matter which method is used to load the .scad file?

2019.01-RC4 on Ubuntu 18.04.2 LTS
2019.01-RC4 on Windows 10 1709

nophead, 2019-03-26:
Also when opened from the recently used list. That used to work but I don't think it does now.

leonplust, 2019-04-15:
I have made some changes to fix this issue. I tested on my machine it works and it's ready to make a PR. But how can I show the test visually?

t-paul, 2019-04-15:
Nice! I guess it's not really possible to show in images, I can give it a try.

leonplust, 2019-04-15:
Thanks. Let me know what I need to do. 
Also, Just to clarify, when saying currently opened file, is it referring to the file that's the main window the window on the top or the latest opened file? For example, you opened three different files accordingly A, B, C. There are three windows showing. Do you expect the file path to be C or you expect the file path to be different when you click open file on A you will get file path for A, on B you will get file path for B?
My current version is you will get file path from the latest opened file no matter which method is used.

t-paul, 2019-04-15:
I think that is stored along the window, so each MainWindow object (= GUI Top Level Window) has it's own storage (`export_paths`) to remember the last path by export type (implemented by @nophead in  #2648). The entry to calculate the name is: https://github.com/openscad/openscad/blob/5b6232c07ff61d3085b32e372658b8317ca97947/src/mainwin.cc#L3184-L3197

This issue should catch the remaining ways to open files that are not covered by that logic yet.

leonplust, 2019-04-16:
OK, I have made some changes. Now the path is stored along with the window. 

Sorry for the late reply. I have encountered some strange issue with the windows build that I have. The executable file that directly cross-built from the source behave the same as the old version which the path is the last opened file by clicking the open tab when openscad is opened by double-clicking .scad file. However, it works when directly open openscad.exe and do drag-drop or open recently files.

While this issue does not exist when I cross-build an installer and install openscad.exe.  If it's installed by installer it doesn't matter what way to open openscad, it works as expected. I don't know how to fix that issue. But I assume it's safe to make a PR since it works when it's installed by the installer?

leonplust, 2019-04-16:
Also, should I post a link of the branch that fixes this issue here?

t-paul, 2019-04-16:
A PR would be easiest, as we can discuss things directly linked to code snippets.

korpx, 2020-03-19:
I am using the latest Windows binary snapshot 2019.12.21.ci4161 (git 4ef45b00)

"Save as" work as expected. If I open a design by any method (drag and drop the .scad into a blank openscad window, double-click on the .scad, or use the recent- or jump list), and then choose "File, Save as..." I will end up in the folder of the currently open design.

It just don't work that way with "File, Open". It will go to the folder of the most recent .scad I opened from within OpenSCAD ("File, Open").