openscad/openscad #2899: Add a keystroke/button for interrupting preview or render

adrianVmariano, 2019-03-27:
It can be very annoying when OpenSCAD is stuck doing a long preview or render and you know you made a mistake in your file.  It should be possible to interrupt a preview or render without killing the application.

t-paul, 2019-03-27:
Render can be cancelled already (although there are limits to imposed by external libraries when that can happen). Preview is covered by #1195.

closed 2019-03-27 by t-paul

MichaelAtOz, 2019-03-27:
> Render can be cancelled already 
 
Via the `x` button next to the progress bar. But if it has entered the final union of everything, that is one call to CGAL and doesn't get interrupted.

For both render & preview, if your source has changed, it is saved, see [here](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#OpenSCAD_crashed/was_killed,_are_my_unsaved_changes_lost?), so you can just kill the process. Until #1195 eventuates.