openscad/openscad #2900: Relative translation does not work for 3D Mouse

DAKarp, 2019-03-27:
I'm able to use my SpaceNavigator 3D Mouse to work with OpenSCAD 2019.01-RC4 (Mac) thanks to help from the forum. But when I try to use relative translation and rotation as per these instructions:

http://forum.openscad.org/Inputdriver-on-MacOS-tp25528p25753.html

relative rotation works as expected, but relative translation is a no-op. The blue bars in the Axis Setup section continue to repond, but the model does not move with translation gestures.

t-paul, 2019-03-27:
Can you please add a screenshot from the preferences window showing the axis settings.

DAKarp, 2019-03-27:
[image]

MichaelAtOz, 2019-03-27:
Is it all axes or just Y? See [this comment](https://github.com/openscad/openscad/issues/2786#issuecomment-463066810).

DAKarp, 2019-03-27:
All axes - no translation in any direction

greyltc, 2019-06-03:
My experience disagrees with @DAKarp's comment above. I've tested relative translation with my 3D mouse (SpaceNav driver) and the only axis that doesn't work for me is Y.

yannpom, 2019-07-15:
I have the exact same issue as DAKarp with a Logitech 3D mouse, version 2019.05 / MacOS Mojave 10.14.4:

Translation works.
ViewPort Relative rotation works.
Zoom works.
but ViewPort Relative translation does not do anything, neither in X nor in Y nor in Z.

All my axis are trimed and responding correctly.

[image]

Would a video be helpful to show the problem? Or tell me how I can provide more information to help reproduce the problem?

BenBergman, 2020-04-03:
I had the same issue as @MichaelAtOz . Using regular Translation worked fine but using ViewPort rel. Translation killed translation response. I tried turning up sensitivity and that got it working, so perhaps it is mistakenly scaling to 0 when it says 1?