openscad/openscad #2902: When running on windows a File C:\dev\null is created on disk after running openscad

thesourcerer, 2019-03-28:
When running on windows a File C:\dev\null is created on disk after running openscad.
The file is locked while openscad is running, and I can manually delete it only after closing openscad. it won't get cleaned on its own.
My guess is that the code is routing output to dev/null (linux style) while in windows it will map to C:\dev\null.

On windows you should use NUL instead of dev/null.

t-paul, 2019-03-28:
I'm pretty sure I know where this is coming from (input driver trace file), but can you please still report which version you are running?

thesourcerer, 2019-03-28:
That would be version 2015.03-2

t-paul, 2019-03-28:
Oh? The old (and still latest) release version? That's surprising and needs a new theory... Can you please check if you see the same effect with the latest release candidate ([OpenSCAD-2019.01-RC4)](https://www.openscad.org/downloads.html#snapshots) of the upcoming release?

thesourcerer, 2019-03-28:
Can I run those side by side? or to I have to uninstall the official release first?

t-paul, 2019-03-28:
Yes, you can just use the ZIP version and extract that anywhere. It might register the nightly window icon, but otherwise there should not be any impact running those side-by-side.

MichaelAtOz, 2019-03-28:
I did not have C:\dev, when I created it, I got the null file with 2015.03.
Not with RC3.

There is not one on my system. No

thesourcerer, 2019-03-28:
Tried OpenSCAD-2019.01-RC4-x86-64, and did not get a repro, so I guess the issue was resolved in this version.

t-paul, 2019-03-28:
Thanks for testing. I'll close this then as we are pretty close to a new release finally, so there's not going to be any update on the old release.

closed 2019-03-28 by t-paul