openscad/openscad #2905: Release new version

kintel, 2019-03-30:
Release new version of OpenSCAD.
TODO:
* [ ] Decide on final name
* [ ] Make a final pass of the github Project: https://github.com/openscad/openscad/projects/1
* [ ] Make a decision which of the approved PRs should be included
* [ ] Go through release notes and complete/cleanup if necessary
* [ ] Do a test release to validate the release checklist and scripts. Update if needed
* [ ] Go through all necessary binaries we deliver and validate that they can be built automatically in release mode or be built manually

MikeBenza, 2019-03-30:
@kintel Please review #2904 for inclusion in the new release.  I think it'd be a real shame for there to be a major release with a significant (IMHO) UX issue on OS X.

kintel, 2019-03-30:
@MikeBenza Thx, will do

tpimh, 2019-04-15:
Flatpak already published by @cosimoc and it is very good, no problems with it.

t-paul, 2019-04-15:
@tpimh thanks, added a pointer in the checklist.

t-paul, 2019-04-15:
Flatpak is still just 2015.03 release though. I don't know if it's possible to also add RCs for the upcoming release without overwriting the old release.

donbright, 2019-05-02:
i would highly recommend not use Docker for official realase, its a massive security risk. 

they just got hacked, actually.

darkworks, 2019-05-15:
please update windows build exe on official site

thanks

closed 2019-05-20 by kintel

tpimh, 2019-08-17:
Flatpak is still not updated. @goncalossilva tried, but as far as I know there were no successful builds.

t-paul, 2019-08-17:
You already opened an issue on the flatpak repo, which is the correct place. Not much that can be done here other then pointing there.