openscad/openscad #2910: text() do not support Chinese

sonichy, 2019-04-01:


nophead, 2019-04-01:
Which version?

The latest version seems to:

[image]

t-paul, 2019-04-01:
Please share a specific example. Also note that OpenSCAD has no layout engine for `text()` which means only text shaping is supported, but no automatic layout and font substitution. So you have to specify a font which actually contains glyphs for the characters given.

sonichy, 2019-04-02:
I know, if non-english should set font parameter.
And preview is sucess, render is flat when do not rotate, disappear when rotate.

closed 2019-04-02 by sonichy

MichaelAtOz, 2019-04-02:
Note that text() is 2D, if you want it to have height make it 3D with linear_extrude()

sonichy, 2019-04-02:
**text() is 2D, if you want to make it 3D should add linear_extrude()** should be written in wiki.
It is OK now.
```
rotate([90,0,0])
linear_extrude(height=5)
text("海天鹰",font="华文行楷");
```

t-paul, 2019-04-02:
Where is that information missing? It's in the first paragraph in the [manual](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Text). Any other place that should be added?