openscad/openscad #2914: Trying to build openscad-2019.01-RC4 for windows: double-conversion/double-conversion.h: No such file or directory

reivaxy, 2019-04-04:
Issue #2765 requested a double-conversion to be added to macosX build script, it looks like it might also be necessary for windows?

I'm getting this message:
../src/value.cc:29:10: fatal error: double-conversion/double-conversion.h: No such file or directory

kintel, 2019-04-04:
Could try try just adding double-conversion to https://github.com/openscad/openscad/blob/master/scripts/mingw-x-build-dependencies.sh ?

reivaxy, 2019-04-04:
I'll let you know in a few hours :)

t-paul, 2019-04-04:
Oh, the docker images don't use the script but call `make` separately. I guess we should change that so we see that issues with the CircleCI builds.

kintel, 2019-04-04:
Yeah, the docker image doesn't depend on the OpenSCAD repo directly as that might be tricky to pull off, but it might be possible to trick Docker caching to make this fast somehow.

reivaxy, 2019-04-04:
Indeed, it didn't work.
I'm not at all familiar with this buildchain but is it really a dependency issue, since it seems to expect "double-conversion" to be in the src directory itself?

t-paul, 2019-04-04:
The library is now added to the script (https://github.com/openscad/openscad/commit/6bfcb5e036d2dc10716b5257f9b600efca148512). Please try again with the updated `mingw-x-build-dependencies.sh`.
Note that this can only work with the MXE branch `openscad-snapshot-build` from the OpenSCAD repo as this has some small updates that are not yet upstreamed to MXE.

reivaxy, 2019-04-05:
Thanks, I've tried building master with this change.
No error when running ./scripts/mingw-x-build-dependencies.sh 64

When running ./scripts/release-common.sh mingw64 I now get the error
Cannot find file: ../openscad.pro.

There still is no double-conversion directory in src

nophead, 2019-04-05:
double-conversion does not end up in src. The header ends up in \msys64\mingw64\include\double-conversion and \msys64\mingw64\include is in the include search path.

openscad.pro is part of the git project: https://github.com/openscad/openscad/blob/master/openscad.pro so you should definitely have it.

I have never attempted to run the release script.

reivaxy, 2019-04-05:
Oh ok thanks.

About openscad.pro yes I had it, but the script was not finding it.
I've made this PR to fix this: https://github.com/openscad/openscad/pull/2916

After that I got a message saying it couldn't find glib.h, which I worked around by creating a sym link "glib" to "glib-2.0" in /usr/include.

And then I was able to compile the installer \o/ 

(and found out the new release does not fix the issue I had with the 2015 build. Oh well :) )

Thanks a lot for your help!

closed 2019-04-06 by kintel