openscad/openscad #2915: ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /usr/include/CGAL/Nef_3/SNC_external_structure.h Line: 1106

gratefulfrog, 2019-04-04:
I have some simple code that won't compile and produces the above error.

Any help would be greatly appreciated.

If I execute ```all(20);```using the preview, it works perfectly, if I use Render, either it renders nothing, or gives the above error.

I am using the nightly builds on Ubuntu 16.04 version 2019.04.04.nightly (git 6bfcb5e) 

I would be happy to help if I can.
Thanks,
Bob

[donutTests.scad.zip](https://github.com/openscad/openscad/files/3045335/donutTests.scad.zip)

MichaelAtOz, 2019-04-04:
I don't have an answer yet, but interestingly, the first OpenSCAD version I tried it with happened to be 2018.11.04.ci339, which worked, both F5 & F6. Then tried 2015.03, errored, then 2019.01-RC4 errored.

MichaelAtOz, 2019-04-04:
Here is a workaround `$fn=32`.
I think this is probably another floating point misalignment, that changes the angle of incidence of the cylindrical shape with the squared edge.

MichaelAtOz, 2019-04-04:
But if you use View/Thrown-together, there is purple showing, so I suspect a STL will be bad...
Yep:
[image]

MichaelAtOz, 2019-04-04:
2018.11.04 doesn't get the error, but produces a bad STL.

The sphere, cylinder & donut is where the problem is.

MichaelAtOz, 2019-04-04:
I've pared it down to: (note I increased the size * 10)

```
$fn=32;
hullRad = 15;
angle   = 20;
cylH     = 400;
e=0.2;

rotate([0,angle/2.,0])
  translate([hullRad,0,0]){
    // donut
    #rotate([-90,angle/2.,0])
      rotate_extrude (angle = 180-angle)
        translate([1*hullRad,0,0])
          circle(r=hullRad);
    // angleCyl
    a=-angle; h=cylH;
    ratio = a == 0 ? 1 : abs(a)/a;
      translate([0,0,0])
        rotate([0,a/2.,0])
          translate([hullRad*ratio,0,-e]){
            cylinder(h=h,r=hullRad);
            *translate([0,0,h])
              sphere(r=hullRad);
          }
  }
```
with the -e, it definitely embeds the cylinder into the donut.
But the STL still has holes and degen faces.
I'm suspecting it is the nexus in the center, all triangles converging:
[image]

MichaelAtOz, 2019-04-05:
This shape really shows up the geometry deficiencies which seem to have creped into OpenSCAD.

By rotating the cylinder `rotate([0,0,360/$fn/2])` it moves the cylinders point away from the nexus.
That the exports the above 'J' shape properly.

Put that in the original (& comment out the sphere for simplicity) then the ends don't join to the middle bit.
[image]

This stuff just used to work (much of the time).

MichaelAtOz, 2019-04-05:
I give up.

I eliminated the nexus by moving the circle in the donut a bit.
It still can't make good geometry.

```
$fn=32;
hullRad = 15;
vAngle   = 20;
cH       = 400;
e=0.2;                    //  <----

module donut(hookAngle){
  //translate([0,0,hullRad])
  rotate([-90,hookAngle/2.,0])
    rotate_extrude (angle = 180-hookAngle)
      translate([1*hullRad+e/2,0,0]) //  <---- remove nexus
        circle(r=hullRad);
}
module angleCyl(h,a){
  ratio = a == 0 ? 1 : abs(a)/a;
    translate([0,0,0])
      rotate([0,a/2.,0])
        translate([(hullRad+e/2)*ratio,0,-e]){ //  <----
          //rotate([0,0,360/$fn/2])            //  <----
          cylinder(h=h,r=hullRad);
          *translate([0,0,h])       // for simplicity
            sphere(r=hullRad);
        }
}
module V(angle,cylH){
  rotate([0,angle/2.,0])
  translate([hullRad,0,0]){
    donut(angle);
    angleCyl(cylH,-angle);
    angleCyl(cylH,angle);
  }
}

module all(hM){
  *translate([0,hM,0])
     V(vAngle,cH);
  V(vAngle,cH);

  rotate([-90,0,0])
      linear_extrude(height=hM)
        projection(cut=true)
          rotate([90,0,0])
            V(vAngle,cH);
}
all(200);
```

Without the rotate of the cylinder:
[image]

With:
[image]

MichaelAtOz, 2019-04-05:
Well I call OpenSCAD broken.
After applying all the coincident face removal mojo, building it at a bigger scale (*10) to get over the 'small number' problem it's had for a while, I made one which Netfabb likes:
[image]

close up of ex-nexus:
[image]

However after scaling it back to the original size:
[image]

I think this floating point problem is insidious.

Code
```
$fn=32;
hullRad = 15;  // *10
vAngle   = 20;
cH       = 400; // *10
e=0.1;                    //  <----

module donut(hookAngle){
  //translate([0,0,hullRad])
  rotate([-90,hookAngle/2.,0])
    rotate_extrude (angle = 180-hookAngle)
      translate([1*hullRad+e*4,0,0]) //  <---- remove nexus
        circle(r=hullRad-e/2);
}
module angleCyl(h,a){
  ratio = a == 0 ? 1 : abs(a)/a;
    translate([0,0,0])
      rotate([0,a/2.,0])
        translate([(hullRad+e*4)*ratio,0,-e]){ //  <----
          //rotate([0,0,360/$fn/2])            //  <----
          cylinder(h=h,r=hullRad);
          translate([0,0,h])       
            sphere(r=hullRad);
        }
}
module V(angle,cylH){
  rotate([0,angle/2.,0])
  translate([hullRad,0,0]){
    donut(angle);
    angleCyl(cylH,-angle);
    angleCyl(cylH,angle);
  }
}

module all(hM){
    union() {
      translate([0,hM,0])
         V(vAngle,cH);
      V(vAngle,cH);
    }

  rotate([-90,0,0])
      translate([0,0,-e])             // overlap with ends
      linear_extrude(height=hM+e*2)   // "
        offset(e) // little bigger
        projection(cut=true)
          rotate([90,0,0])
            V(vAngle,cH);
}

//scale(0.1)  // <-- reduce to original size
  all(200); // *10
```

nophead, 2019-04-05:
Yes it does seem more broken than normal. I made a simpler version where I have only two junctions between three valid solids and I can easily change the overlap and shrinkage where they meet to avoid any numerically close vertices. I can get rid of degenerate triangles and holes but even with a big overlap and step I always seem to get some self intersections according to NetFabb.

```
$fn=32;
hullRad = 15;  // *10
vAngle   = 20;
cH       = 400; // *10
hM      = 200;
eps = 1;

module profile(h)
    hull()
        for(i = [-1, 1])
            translate([0, i * h / 2])
                circle(hullRad);
    
module bend(hookAngle, h){
    rotate([-90, hookAngle / 2, 0])
        rotate_extrude (angle = 180 - hookAngle)
            translate([hullRad + eps, 0])
                profile(h);
}

module V(angle, h){
    tw =  hM + 2 * hullRad;
    rotate([0, angle / 2, 0])
        translate([hullRad,0,0]){
            bend(angle, hM);
              
            for(a = [-1, 1])
                rotate([0, a * angle / 2, 0]) hull() {
                    translate([a * (hullRad + eps), 0, h])
                        rotate([-90, 0, 0])
                            rotate_extrude()
                                intersection() {
                                    profile(hM);
                                
                                    translate([0, -(tw + 1) / 2])
                                        square([hullRad *2, tw + 2]);
                                }
                                
                    translate([a * (hullRad + eps), 0])
                        linear_extrude(height = 1, center = true)
                            offset(-1)
                                profile(hM);
                }                    
        }
}

V(vAngle,cH);
```
[image]

I could make the bend and the sides seamless using sweep but I think I would have similar problems trying to union rounded endcaps.

gratefulfrog, 2019-04-06:
hi guys!
Thanks for all the follow-up! i don't really understand all of what you are
saying, but I am sad that openscad seems to be regressing. It is one of my
favorite tools, but I am facing odd behavior every day now. Huge
discrepancies between F5 and F6 results for example. Some objects display
fine with F5, but F6 produces no result at all...

Has something big changed in the code lately?

Cheers,
Bob

nophead, 2019-04-06:
F5 and F6 are rendered totally differently. F5 just draws pixels and uses graphics card tricks to simulate CGS ops. For example union simply draws both objects I think. F6 calculates all the new vertices and faces and then draws the resulting polyhedron. It fails if the geometry is not valid whereas F5 has no notion of validity. The worst you get is z-fighting.

I don't know if it is definitely a regression because I have a large library of parts that I am continually testing and nothing is broken. I don't understand why a union with large overlap and no close vertices should fail though. 

The only unusual thing compared with my designs is this is the ends of the corner are both inclined polygonal planes, so the polygons will not be exactly co-planar because none of the vertices are rational numbers. It gets triangulated to keep CGAL happy and so do the curved faces meeting it so. So the resulting union might have fairly close vertices that get merged by the grid snap.

The big bug in OpenSCAD is that it switches numerical types all over the place and even deliberately snaps to a grid and this will break topology whenever distinct vertices get merged. It is a major problem because all the GGAL errors reported here are either caused by it or by bad STL files. We desperately need mesh healing to end this constant stream of bug reports.

MichaelAtOz, 2019-04-06:
@gratefulfrog I'll try to summerise what we are going on about, but It'll take a little while.
Basically there are situations which 3D operations don't 'understand' and you need to adjust things to give it a hint. TBC

nophead, 2019-04-06:
CGALs operations are pretty rock solid. The problem is OpenSCAD doesn't use its number representation and naively down converts to floats, which is a hard thing to do without corrupting topology.  This can normally be worked around by ensuring vertices are either exactly coincident or far enough apart to not get merged. In this case the normal workarounds don't seem to work because the mating surfaces are too finely triangulated.

nophead, 2019-04-06:
This version seems to work. I can't see any errors in NetFab
```
$fn=32;
hullRad = 15;  // *10
vAngle   = 20;
cH       = 400; // *10
hM      = 200;
eps = 1/128;
irad = 0.5;

module profile(h)
    hull()
        for(i = [-1, 1])
            translate([0, i * h / 2])
                circle(hullRad);
    
module bend(hookAngle, h){
    rotate([-90, hookAngle / 2, 0])
        rotate_extrude (angle = 180 - hookAngle)
            translate([hullRad + irad, 0])
                profile(h);
}

module V(angle, h){
    tw =  hM + 2 * hullRad;
    rotate([0, angle / 2, 0])
        translate([hullRad,0,0]){
            bend(angle, hM);
              
            for(a = [-1, 1])
                rotate([0, a * angle / 2, 0]) hull() {
                    translate([a * (hullRad + irad), 0, h])
                        rotate([-90, 0, 0])
                            rotate(180)
                            rotate_extrude(angle = 180)
                                intersection() {
                                    profile(hM);
                                
                                    translate([0, -(tw + 1) / 2])
                                        square([hullRad *2, tw + 2]);
                                }
                                
                    translate([a * (hullRad + irad), 0])
                        linear_extrude(height = 1, center = true)
                            offset(delta = eps)
                                profile(hM);
                }                    
        }
}

V(vAngle,cH);
```
I expanded the profile where it meets slightly instead of contracting it. That means the part that meets is the end, avoiding the long thin triangles down the sides.

MichaelAtOz, 2019-04-07:
Not quite.
[image]
Degenerate faces & a self intersection.

When you go to the original size,
first by scale(0.1)
[image]
No S.I. interestingly.

Or by changing the variables (hullRad = 1.5; cH=40; hM=20; irad=0.05; & make the intervening slice `linear_extrude(height = eps*2...`)
[image]
Again no S.I.

This isn't 2019 related, this is with above small numbers & 2015.03, noting that it does not have the `angle=` parameter to `rotate_extrude()`
[image]

Which is why I think it is broken, going to two different methods, both avoiding the 'traditional' problem areas and not being able to generate good geometry, when at the heart of it, a `union()`, which the OP code is built with, should not have a problem with coincident faces. If it was `difference()` maybe.

@gratefulfrog I'll get to the explanation today.

MichaelAtOz, 2019-04-07:
Explaination. 
Let me preface this with, I-am-not-a-qualified-mathematician so I'm likely to misuse some terminology.

Firstly the CGAL error.
ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /usr/include/CGAL/Nef_3/SNC_external_structure.h Line: 1106

So that is 
`CGAL error in CGALUtils`
`applyBinaryOperator union`
`assertion violation! Expr: itl != it->second.end()`

CGAL does a lot of internal checks, this is an assertion failure during a `union()`.
It is expecting `itl` (whatever that is) to be equal to `it->second.end()` (whatever that is - other errors are more decipherable).
This means OpenSCAD has passed it geometry which is off.

Experience has shown us what this is likely to be.

Originally OpenSCAD used [quotients](https://doc.cgal.org/latest/Number_types/classCGAL_1_1Quotient.html), which are an exact mathematical representation of real numbers. 
However that imposes significant memory & CPU overheads, so to improve performance, later versions introduced some floating point, which as[ CGAL says](https://doc.cgal.org/latest/Number_types/index.html):
> Like with all number types with finite precision representation which are used as approximations to the infinite ranges of integers or real numbers, the built-in number types are inherently potentially inexact. Be aware of this if you decide to use the efficient built-in number types: you have to cope with numerical problems. For example, you can compute the intersection point of two lines and then check whether this point lies on the two lines. With floating point arithmetic, round-off errors may cause the answer of the check to be false.

Basically due to such inaccuracies, two objects very close to each other *may* be seen to be touching, or two which touch exactly*, called coincident faces, *may* or *may* *not* be joined.

 *e.g. Here the right side of the first cube & left of the second occupy the same space
```
  cube(10);
  translate([10,0,0])
    cube(10);
```
That is obvious with nice integer numbers, but if you rotate that say 33 degrees, then any point of those faces, in floating point will be either exactly the same, or slightly closer, or further apart. This is also complicated by OpenSCAD internals which tries to help, which works some of the time. but can introduce issues too.

So now we need to help CGAL know what we want it to do. 
Note this is not limited to OpenSCAD, most CAD implementations have this problem.
Basically you have to make it blatant that two objects are either together, or apart. 
This is done by either embedding one into the other, or ensuring there is a meaningful gap.

In your original code you had a bunch of objects butting up against each other, each join could introduce these uncertainties.
Starting with your donut(), being without a hole, meant that lots of increasingly small triangles were all trying to point into one singularity in the center, what I called the nexus. Plus the corner of the cylinders also at the nexus. Add floating point and there will be a small cloud of coordinates not quite at the same singularity.
So firstly I moved the circles & cylinders apart by a small amount (the e=0.2), ie a meaningful gap.

I then embedded the cylinder end into the donut, both having the same external faces. This will often work, particularly with simple integer flat faces. In the above cube example this would be done by `translate([10-0.05,0,0])`.
Unfortunately, that didn't work, so I want to make the surface clearly distinct, I reduced the size of the donut circle, so the donut & cylinder faces were not coincident.
Then added 
```
      translate([0,0,-e])             // overlap with ends
      linear_extrude(height=hM+e*2)   // "
        offset(e) // little bigger
``` 
to the center-piece so it was slightly bigger than and enveloped the the end-pieces.
No coincident points, lines or faces.

That is usually sufficient for CGAL to produce good geometry, which it probably did.
There is one final step, export of the STL, this is pure floating point, and for this design, produced degenerate faces and/or self intersections, as highlighted in the Netfabb images above.

This is compounded by what I called the 'small number' problem. This is often seen with curves, if you start with small objects then the differences in curves are even smaller differences, leading into the uncertainties I describe above.
Your original size of `hullRad=1.5` is starting small.
Until this is fixed, the workaround is to do it larger, then scale the final result back down.

My solution [above](https://github.com/openscad/openscad/issues/2915#issuecomment-480160201) worked, until I reduced it in size...but
Sometimes OpenSCAD doesn't work.
You could export the bigger object, then scale it in another tool.

An alternative is to use less detail, the code below, has a slightly larger donut internal radius and a smaller `$fn`, this has fewer faces and the differences between them are larger, so two points are more distinct.
Turns out that it exports cleanly, but that is probably more to do with lucky...
```
$fn=24;
hullRad = 15;
vAngle   = 20;
cH       = 400;
iRad=hullRad/15;
[... 25 lines cut ...]
    angleCyl(cylH,angle);
  }
}

module all(hM){
    union() {
      translate([0,hM,0])
         V(vAngle,cH);
      V(vAngle,cH);
    }

  rotate([-90,0,0])
      translate([0,0,-e])             // overlap with ends
      linear_extrude(height=hM+e*2)   // "
        offset(e) // little bigger
        projection(cut=true)
          rotate([90,0,0])
            V(vAngle,cH);
}

scale(0.1)
  all(200);
```

I'm happy to clarify anything if needed.

MichaelAtOz, 2019-04-07:
See also [CGAL FAQ](https://www.cgal.org/FAQ.html#inexact_NT).

gratefulfrog, 2019-04-07:
@Michael

Thank you for the detailed and understandable explanation of what is going
on here. I realized that the triangle generation must be complex but had no
ideas as to the details.

If I understood correctly, you advise to design at a larger scale, say 10x
what I was doing, and then use other applications in the 3D printing
tool-chain, e.g. Slic3r, to scale the object back down to the correct size.
Is that correct? Will that generally give better results?

In any case, thanks for your hard work on my little problem and especially
for taking the time to explain to me what was going on!

Cheers,
Bob

MichaelAtOz, 2019-04-07:
If you're operating in the sub-ten size with detailed curves, that is a problem area.
So work even in the x100 space, then scale(0.01), will *often* work in OpenSCAD without requiring external tools.
But if the STL is bad then yes, scaling with Slic3r etc would be the simplest.
Hopefully we'll get to the bottom of this and improve the internal grid handling etc. but the floating point issue will always be hovering in the background.

praetp, 2024-03-11:
any update on this ? hitting this with [OpenSCAD](https://www.openscad.org/) version 2024.03.10.ai18783 (git f96ef9b83)

kintel, 2024-03-11:
@praetp This is very input-dependent. I suggest opening a new ticket with your failing example. Ideally, as small as possible, then we can look at it and advice from there.

kintel, 2024-03-30:
Status as of 2024-03-24: The initial example now works with CGAL, but it's broken with Manifold

MichaelAtOz, 2026-01-31:
Status as of 2026-01-25, both CGAL & Manifold produce STL with holes. 
Manifold render to 3mf is non-manifold with SI's. Aroung the singularities in the donut.