openscad/openscad #2917: How to export big image?

sonichy, 2019-04-06:
How to export big image in Preview and Render mode ?

kintel, 2019-04-06:
Please use #openscad on freenode or the mailing list as a support channel

closed 2019-04-06 by kintel

MichaelAtOz, 2019-04-07:
@sonichy See [this](http://forum.openscad.org/Registration-amp-Subscription-Info-READ-THIS-FIRST-td21660.html) to register to the [forum](http://forum.openscad.org/).

t-paul, 2019-04-07:
I've seen that question before, so I've added a [FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_can_I_export_screenshots_with_higher_resolution_than_the_current_window) for this too.

sonichy, 2019-04-07:
Why not support a gui to set export ?

kintel, 2019-04-07:
It's not that simple. As a Qt developer I'm sure you can follow the threads starting here: https://github.com/openscad/openscad/blob/master/src/mainwin.cc#L2598

We do accept patches though :)

sonichy, 2019-04-08:
After seen https://github.com/openscad/openscad/blob/master/src/mainwin.cc#L2598 , the GUI exprot image is
```
QImage QOpenGLWidget::grabFramebuffer()
```
Where is the command line export source code ?
```
openscad --imgsize 16000,16000 -o CSG.png CSG.scad
```

kintel, 2019-04-08:
https://github.com/openscad/openscad/blob/406dd378003018fd01281787382acd3c1037c57d/src/export_png.cc#L54

sonichy, 2019-04-09:
openscad build failed, fatal error: glib.h: No such file or directory
But I made a demo: https://github.com/sonichy/QOpenGL
```
QSize size = openGLWidget->size();
openGLWidget->resize(spin_width->value(),spin_height->value());
QImage image = openGLWidget->grabFramebuffer();
image.save(path,0,100);
openGLWidget->resize(size);
```