openscad/openscad #2921: Constant 4% cpu load from rapid poll syscalls with 1-25 ms timeout

richfelker, 2019-04-07:
Running `strace` on an idle openscad (gui) process shows it constantly calling poll() with extremely short timeouts. This yields a constant 4% load on my laptop and likewise fast battery drain. Interspersed with the polls are stat syscalls on the open file, so I tried turning off automatic reload, but the polls remain. I haven't traced where they're coming from but it looks like somewhere in Qt or another library, possibly as a result of incorrect use of QTimer or internal bugs in QTimer.

Ideally automatic reload should use `inotify` with no timeout in the `poll`, rather than short timeouts, on systems where `inotify` is available. Short of that, the `poll` interval should be the whole refresh rate (currently hard coded as 200 ms), not 25 ms, and it would be nice if it were configurable.

kintel, 2019-04-07:
Using Linux `inotify` and the equivalent mechanisms on other OS'es is definitely the right way to go.
In the meantime, perhaps @nophead has a comment on the poll timing?

nophead, 2019-04-07:
AFAIK the stat calls are only every 200ms and only when auto reload is on, that is certainly the intention. On my windows desktop with 147 files being polled it only takes 0.5% of the CPU. The OS caches the file accesses, so it is really only polling RAM.

This is with the latest master. It would be a lot worse with nested includes before #2812 was merged on 25th Feb. 

Rather than being configurable I think you would just turn it off when running from battery if it causes a problem.

I don't know where 25ms comes from.

kintel, 2019-04-07:
@richfelker What version of OpenSCAD are you using?

richfelker, 2019-04-08:
According to git-describe, 2019.01-RC4. The 25ms poll timeouts don't go away even with auto-reload disabled. Just the stat syscalls do. I was experiencing the problem with just a single file open, no includes.

kintel, 2019-04-08:
That's odd - nophead's fixed should be included in that one.

MichaelAtOz, 2019-04-08:
On Windows, using Microsoft Sysinternals Process Monitor, 2019.01-RC4 with one file, access the directory entry every approx 200ms, here is 4 consequtive of these
[image]

8/04/19 15:50:06.3951496
8/04/19 15:50:06.6013485
8/04/19 15:50:06.8003349
8/04/19 15:50:07.0067210

It does a bunch of registry lookups in between. 
There is no I/O, purely checking the cached directory entry.
It is active, e.g. from
8/04/19 15:50:06.7994435 to
8/04/19 15:50:06.8008588 = 1.5ms in that ~200ms period.

Process explorer shows it using ~0.55% CPU on a 6 core-thread* VM, so that is ~3% of a 1 core non threaded CPU, probably on the high side for idling. But only a problem on batteries...

*VMware treats each thread of a multi-thread core as a virtual CPU, so in this case 3 cores are allocated, in the VM it sees 6 CPUs. So a process at 100% is consuming 17% (16.66 rounded).

Perhaps we can apply some smarts like minimising the window, pauses the auto-reload??
Or if there has not been a reload for some period, increase the timeout??

MichaelAtOz, 2019-04-08:
Note that I first thought that it was still checking every 200ms after I turned Auto-reload off. 
But there was a second document open in the MDI, once I deselected that it stopped checking the directory.
However, CPU didn't change...?? 
Process Monitor showed nothing happening...
Turns out if defaults to filtering 'process profiling' events.
Most of that ~0.55% CPU is Windows profiling the process.

nophead, 2019-04-08:
I don't think it is anything to do with with auto reload as that only uses stat().

The only fast QTimer I can see is the InputEventMapper at 30ms. Does that run all the time and would it call poll() on Linux?

richfelker, 2019-04-08:
Yes, that's clearly it. 25ms is likely the remainder of the timer period after the process waits to get scheduled and runs the handler. Short of a proper fix, is there somewhere I can just prevent this from ever getting instantiated? I'm not using joysticks. That would also confirm that this is the source of the problem.

kintel, 2019-04-08:
@t-paul This may be something you know about

richfelker, 2019-04-08:
I confirmed that commenting out `timer->start(30);` makes the problem go away, and seems to have no ill effects for non-joystick users. WIth that change, I see constant 200ms-timeout polls with auto-reload enabled, and completely idle strace with auto-reload disabled, as expected.

Probably starting the timer should be made conditional on having a joystick device enabled. Ideally it would not wake up every 30ms even then, but would only enable the timer when an input event puts an axis in a position where movement would be produced, and disable it again when the axis is back centered.

ijackson, 2020-02-17:
I observed this problem in Debian bullseye, 2019.05-3.  It irritates me so I am motivated to fix it :-).  If no-one else is working on this I will take a look.

I'm hoping that I can make patches against 2019.05 and then rebase them, because that will be easiest for me to build.  Is that likely to work well ?

kintel, 2020-02-17:
@ijackson Nobody else is working on this, to my knowledge. Patching against the latest release will work. If there is anything we can do to help you build master would be better as it might help other people in the future.

richfelker, 2020-02-17:
If you get a solution against an older version and have trouble rebasing it against master, I'd be glad to help with the latter.

ijackson, 2020-02-17:
Thanks, both.  I will take a look.

Ian.

-- 
Ian Jackson <ijackson@chiark.greenend.org.uk>   These opinions are my own.

If I emailed you from an address @fyvzl.net or @evade.org.uk, that is
a private address which bypasses my fierce spamfilter.

t-paul, 2020-02-17:
There's literally just 5 lines of code change (and one comment line :smile:)  in `src/input` so there should not be any problem getting things into master.
I guess the main question is how to change/replace `InputEventMapper::onTimer()` in a way that it still polls quick enough if there are input events, but goes quiet if nothing happens.

richfelker, 2020-02-17:
I think the issue is that the timer should be armed only when joystick is non-centered (to enable auto-repeat); this change might be more invasive but I'm not sure. But a simple partial fix is just disabling it when joystick control is not enabled.

ijackson, 2020-02-17:
> I think the issue is that the timer should be armed only when joystick is
> non-centered (to enable auto-repeat); this change might be more invasive but
> I'm not sure. But a simple partial fix is just disabling it when joystick
> control is not enabled.

Precisely.

I may take a look at using inotify for input files too.  I often have
a dozen openscads sitting about on my laptop for weeks on end...

Ian.

-- 
Ian Jackson <ijackson@chiark.greenend.org.uk>   These opinions are my own.

If I emailed you from an address @fyvzl.net or @evade.org.uk, that is
a private address which bypasses my fierce spamfilter.

t-paul, 2020-02-17:
I don't see much point in having a 10% solution that works only for the case when there's no input device at all (well, unless it's like 2 lines of code and obvious).

It's probably not just a Linux issue, so while it's possible to have different solutions based on platform, it would be better if we can find something that works independently. Something like @richfelker mentioned might work that we start translating as soon as there's events but stop after some quiet period. That might need some additional tracking of event times but should work for all cases independent of the platform.

That said, I think the logic mainly came in to do rate limiting of input events as some devices may have a very high event rate which does not make sense for OpenSCAD usage. So another option would be stepping back a bit and see if it can be handled on a lower level.

richfelker, 2020-02-17:
When I last read the code, the short-duration timer was not about polling input events. It was about synthesizing rapid motion events from a non-moving joystick being held in a non-centered position, for smooth scrolling/rotation/whatever. That makes sense. The only problem was keeping the timer running when that's not happening.

t-paul, 2020-02-17:
It's probably both, at least the GUI update part comments
```
//update the UI on time, NOT on event as a joystick can fire a high rate of events
```

ijackson, 2020-02-18:
I'm making good progress on the 30ms joystick timer.  More updates on that soon.

I took a look at using inotify for the reload timer.  That doesn't seem quite so simple.  As far as I can see, there isn't a list of the loaded filenames anywhere.  Instead, we just reparse all the files every 200ms?  I am doubting my own reading of the code since it seems to me that that would be too slow.  Anyway, if someone could point me in the right kind of directions and/or contradict me where I'm wrong, that might help.  Thanks.

ijackson, 2020-02-18:
Please see #3228 for my proposed fix to the 30ms timer issue.

Maybe the desire for inotify should be a separate issue?  If so I would be happy to file one.

richfelker, 2020-02-18:
Yes I think `inotify` is a separate issue. It only came up here because I initially suspected auto-reload was behind the problem.

ijackson, 2020-02-18:
Rich Felker writes ("Re: [openscad/openscad] Constant 4% cpu load from rapid poll syscalls with 1-25 ms timeout (#2921)"):
> Yes I think inotify is a separate issue. It only came up here because I
> initially suspected auto-reload was behind the problem.

Right.  OK, I have opened #3230 to track the desire for inotify
support.

Thanks,
Ian.

-- 
Ian Jackson <ijackson@chiark.greenend.org.uk>   These opinions are my own.

If I emailed you from an address @fyvzl.net or @evade.org.uk, that is
a private address which bypasses my fierce spamfilter.

PR #3228 (2020-05-23): Only enable 30ms joystick timer when needed
Fixes #2921 

The effect is that the timer is now only enabled when (i) there is a joystick and (ii) either (a) it has been moved or buttons pressed recently and we haven't processed that yet or (b) the joystick is sufficiently off-centre and out of the dead zone that we must generate input events.  When this ceases to be true, the timer is disabled again.  So this should solve the cpu usage (and battery wastage) problem for all users, even those with joysticks.

I have tested this by rebasing these patches onto Debian's 2019.05-3 package, as found i [... cut]
closed 2020-05-23 by t-paul