openscad/openscad #2923: A single root modifier should not give warnings

nophead, 2019-04-07:
If a single root modifier is added to code that is called more than once it generates erroneous warnings.

```
for(i = [0 : 10])
    translate([i * 10, 0]) !cube();
```

```
Compiling design (CSG Tree generation)...
WARNING: Root Modifier (!) Added At Line2 
WARNING: Root Modifier (!) Added At Line2 
WARNING: Root Modifier (!) Added At Line2 
WARNING: Root Modifier (!) Added At Line2 
WARNING: Root Modifier (!) Added At Line2 
```

If stop on first warning is set in preferences then it prevents rendering.
```
WARNING: Root Modifier (!) Added At Line2 
Execution aborted
```

MichaelAtOz, 2019-04-08:
Was the original intent to warn when more than one root modifier was present?

t-paul, 2019-04-09:
What about
```
for(i = [0 : 10])
    !translate([i * 10, 0]) cube();
```
Would we define that if there's a single modifier *in the source code* then it should not generate the warning, but it still renders only one of the instances?

kintel, 2019-04-09:
The intent was to warn in cases where someone added a second `!`. ..so it should probably warn about more than one literal `!` in the source.
I don't think this warning is very important though, mostly a helper

leonplust, 2019-04-09:
My new commit is it only generates a warning for each line when there is more than one root modifier is called. Is that what you want?

nophead, 2019-10-15:
I would prefer no warning at all because with Stop on First Warning it prevents anything bring rendered.  It is better to show something as it is then visually obvious where the root is.

If @leonplust's commit fixes the case where there is only one root modifier then that is a big improvement and should be merged but I would prefer if it used the deepest modifier when one is a on a parent node of the other, and only warn if they are separate branches. 

The reason is I might isolate something and then isolate a component of that, fix it and then want to pop back up the tree. I don't want to have to remove one modifier and put it back again.

PR #3183 (2020-03-09): Fix root modifier warning
Fixes #2923:
* Prints warning if more than one root modifier is found
* This does no longer include multiple instances of the same object
* Added tests
closed 2020-03-09 by kintel