openscad/openscad #2933: OpenScad 2019.04.07 hangs on "New" or "Open File" on Intel Graphics / Linux

KlausKnopper, 2019-04-13:
Recent issues of openscad (also nightly builds) hang indefinitely in the editor/preview view on 32bit Debian, intel i915 graphics. Stable version 2015.03 built with exactly the same libraries works on the same machine.

strace of version 2019.04.07 keeps looping in
3848  ioctl(11, DRM_IOCTL_I915_GEM_MADVISE, 0xfff0178c) = 0
3848  ioctl(11, DRM_IOCTL_I915_GEM_BUSY, 0xfff01724) = 0
...repeating...

The error might be similar to the previously reported QT5 crash on start issue on Windows machines, though the workaround of changing QT_OPENGL=(anything...) does not work here.

t-paul, 2019-04-14:
Please add the details asked for in that FAQ entry: [How_do_I_report_bugs?](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs?)

That sounds like a tricky one, so we need to collect all the information we can. I don't see how to easily reproduce, or is it reproducable in some VM?

KlausKnopper, 2019-04-14:
>    [1]How_do_I_report_bugs?
> References
>    Visible links

Hallo Torsten Paul,

Same thing happens with the openscad version currently in
Debian/unstable (i386), so I'm somewhat puzzled about the fact that I'm
the first one to report.

I can provide access to a bootable live ISO image by PM on request, for
easy testing. With this image, I can reproduce the same behavior in kvm,
so it may be unrelated to intel graphics after all.

The requested (lengthy) additional information:

1. The OpenSCAD version: 2019.01-RC2 and so far all versions in
Debian/unstable since at least february 2019 an openscad-nightly from
http://download.opensuse.org/repositories/home:/t-paul/Debian_9.0/ .

2. The Operating System name and version:
Debian Debian GNU/Linux buster/sid (i386/32bit)

3. A description of the scenario that produces the issue:
Starting openscad and hitting "New" or "Open" from the starter window,
the followup openscad window shows editor, preview, console and the new
"customizer", and freezes immediately after. The text cursor in the edit
window does not blink as usual and text input is not possible. openscad
consumes 100% cpu in one thread until killed (force-closing by the
window manager does not end the program).

4. In case of graphics issues, the OpenGL driver information:

$ glxinfo
name of display: :0.0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_no_error, 
    GLX_ARB_create_context_profile, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_libglvnd, GLX_EXT_no_config_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, GLX_OML_swap_method, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_profile, GLX_ARB_create_context_robustness, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
[... 281 lines cut ...]
    GL_OES_vertex_half_float, GL_OES_EGL_image, GL_OES_depth_texture, 
    GL_OES_packed_depth_stencil, GL_EXT_texture_type_2_10_10_10_REV, 
    GL_OES_get_program_binary, GL_APPLE_texture_max_level, 
    GL_EXT_discard_framebuffer, GL_EXT_read_format_bgra, GL_EXT_frag_depth, 
    GL_NV_fbo_color_attachments, GL_OES_EGL_image_external, GL_OES_EGL_sync, 
    GL_OES_vertex_array_object, GL_ANGLE_texture_compression_dxt3, 
    GL_ANGLE_texture_compression_dxt5, GL_EXT_robustness, GL_EXT_texture_rg, 
    GL_EXT_unpack_subimage, GL_NV_draw_buffers, GL_NV_read_buffer, 
    GL_NV_read_depth, GL_NV_read_depth_stencil, GL_NV_read_stencil, 
    GL_EXT_draw_buffers, GL_EXT_map_buffer_range, GL_KHR_debug, 
    GL_KHR_robustness, GL_OES_required_internalformat, 
    GL_OES_surfaceless_context, GL_EXT_separate_shader_objects, 
    GL_EXT_compressed_ETC1_RGB8_sub_texture, GL_EXT_draw_elements_base_vertex, 
    GL_EXT_polygon_offset_clamp, GL_EXT_texture_border_clamp, 
    GL_KHR_blend_equation_advanced, GL_KHR_context_flush_control, 
    GL_OES_draw_elements_base_vertex, GL_OES_texture_border_clamp, 
    GL_KHR_no_error, GL_EXT_shader_framebuffer_fetch_non_coherent

5. If possible, a trimmed down script reproducing the issue:

echo 'cube(10);' > test.scad
openscad test.scad

Regards
-Klaus

t-paul, 2019-04-14:
I can't reproduce on a fresh installed debian-unstable, so an ISO showing the issue would be useful. I think that installed Gnome on Wayland, not sure if that is important.
```
Linux debian-unstable-32bit 4.19.0-4-686-pae #1 SMP Debian 4.19.28-2 (2019-03-15) i686 GNU/Linux
```
Library Info... (interesting memory amount ;-)
```
OpenSCAD Version: 2019.01-RC2
System information: Linux 4.19.0-4-686-pae #1 SMP Debian 4.19.28-2 (2019-03-15) i686 Debian GNU/Linux buster/sid 2 CPUs 16777216.00 TB RAM
User Agent: OpenSCAD/2019.01-RC2 (Linux i686; Debian GNU/Linux buster/sid)
Compiler: GCC "8.3.0" 32bit
MinGW build: No
Debug build: No
Boost version: 1_67
Eigen version: 3.3.7
CGAL version, kernels: 4.13, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.11.3
QScintilla version: 2.10.4
InputDrivers: 
GLib version: 2.58.3
lodepng version: 20180910
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.9.1
harfbuzz version: 2.3.1
lib3mf version: 1.8.1

GLEW version: 2.1.0
OpenGL Version: 3.1 Mesa 18.3.4
GL Renderer: llvmpipe (LLVM 7.0, 128 bits)
GL Vendor: VMware, Inc.
RGBA(8880), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8880), depth(0), stencil(0)
```

donbright, 2019-05-02:
does it work with software rendering? (Is that even still a thing?) 

IIRC you can test by running this:

     $ LIBGL_ALWAYS_SOFTWARE=1 openscad

KlausKnopper, 2019-05-02:
> LIBGL_ALWAYS_SOFTWARE=1 openscad-nightly

hangs the same way.

KlausKnopper, 2019-07-23:
>    Yes, I'm seeing that in my test VM too. I'm a bit stuck as it's not
>    reproducible on a "normal" 32bit Debian VM. The only item remaining on my
>    idea list is to get an AppImage going for 32bit and see what that is doing
>    when run the Knoppix image.

Hello Torsten,

After tons of testing, library updates/downgrades and compiles from
source, I can narrow down a possible cause.

A) Booting Knoppix with the (default) 64bit kernel on a 64bit
architecture, but running the (default) 32bit userspace including 32bit
openscad -> openscad hangs when opening the editor/preview window. This
is the case you also observed when booting the Knoppix image in a 64bit
VM (which will autoselect the 64 bit kernel and run in 32bit
compatibility mode).

B) Booting Knoppix with the (manually typed in the boot comamndline
"knoppix") 32bit kernel on a 64bit architecture, running the (default)
32bit userspace including 32bit openscad -> openscad works when opening
the editor/preview window.

C) Booting Knoppix on a 32bit architecture with 32bit kernel and 32bit
userspace -> openscad works when opening the editor/preview window.

All cases with the same openscad 32bit Debian/Linux installation version
2019-RC1 or RC2 or nightly.

In scenario A), where openscad hangs (64bit kernel and 32bit userspace
on 64bit hardware), its is not sufficient to start openscad by:

setarch i386 openscad

, it will still hang the same way. Usually, the setarch i386 trick,
which is also changing the output of uname -a, works fine solving
bitness problems with software trying to determine wheter to use 64 or
32bit builtin instructions.

Browsing the openscad source, I did not find any bitness-specific code
yet, but I may have missed issues that assume
kernel-architecture == userspace architecture without explicit check.

Running the 32bit kernel on 64bit platforms, only supporting 4GB of RAM,
or installing a 63bit userspace environment which won't work on 32bit
computers, is no option. 

So it seems do narrow down to openscad having problems with mixed kernel
vs. userspace bitness starting from versions higher than (probably) the
2015 editions which still worked in scenario A.

Regards
-Klaus

KlausKnopper, 2019-07-30:
Changed issue topic to reflect the recent findings.

t-paul, 2019-10-24:
I still can't reproduce in any development environment. I've now installed a Debian 10/32-bit VM and updated the kernel to an `amd64` variant.
Interestingly that totally broke booting into gdm so that seems to be affected by the issue too. After installing `lxdm` graphical login was possible. Compiling and running current `master` of OpenSCAD on that system works fine.

KlausKnopper, 2019-10-24:
Hello Torsten,

In the very same environment you mentioned (Debian 10 32bit running on a
Debian 64bit kernel including modules), openscad hangs for me when
opening the main (editor/Preview) window (the "intro" window is still
responsive).  Are you sure that the 64bit kernel was actually booted in
your VM? (uname -a)

Debian will normally refuse to install a 64bit kernel package on a 32bit
system, unless you added the 64bit architecture package option or did a
force install.

Also, aside from the kernel, did you also install the kernel modules? If
not, this may explain why Gnome failed to work: the accellerated
graphics (KMS) modules are missing, so the system max fall back to VESA,
and you end up with a fallback window manager.

Running the 32bit openscad version in a 32bit environment and a 32bit
kernel works.

Since you mentioned you compiled openscad from the git master branch: I
did this a while ago, but later ran most tests from the Debian 32bit
package you provided at
http://download.opensuse.org/repositories/home:/t-paul/Debian_Unstable/
. To make sure that I didn't miss an important change in the source, I
now checked out the git master branch again and compiled it by myself,
but the result is unfortunately still the same.

Regards
-Klaus

t-paul, 2019-10-24:
Yep, I did:
```
dpkg --add-architecture amd64
apt-get update
apt-get --no-install-recommends install linux-image-amd64:amd64
uname -a
Linux debian10-32bit 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u1 (2019-09-20) x86_64 GNU/Linux
```
But maybe later booted into the wrong kernel again. I can now reproduce the issue and it seems it hangs inside `swrast_dri.so`. OpenSCAD does run when disabling the axis markers but the display is still not correct.
Unfortunately I don't think there's much that can be done on OpenSCAD side to fix the issue. Maybe there's a way to disable DRI but I could not find a setting for that yet.

KlausKnopper, 2019-10-25:
Hi Torsten,

I'm glad that you are now finally able to reproduce the problem in a -
more or less - "standard" Debian setup with mixed Kernel/userspace
bitness; I was almost wondering if I was missing some compile time
option.

Disabling the axis markers manually in openscads config file before
starting, indeed delays the freezing to a certain point before rendering
an object. But since all other programs using DRI, including more
complex graphics like tuxracer, blender etc. work flawlessly in the same
setup, I don't think the bug is related to DRI or software rastering
itself.

Also, the old openscad version 2015.3 will run nicely without rendering
problems with and without hardware-supported DRI, no freezes, using the
same X11 and library versions with the possible exception of (editor)
libqt5scintilla2 (just install openscad and libqt5scintilla2 from
Debian/oldstable in your test setup), which makes me think that the
newer versions of openscad must do something different in the preview
window rendering than 2015.3 did.

Regards
-Klaus

On 24.10.19, Torsten Paul wrote:
> Yep, I did:
>
>  dpkg --add-architecture amd64
>  apt-get update
>  apt-get --no-install-recommends install linux-image-amd64:amd64
>  uname -a
>  Linux debian10-32bit 4.19.0-6-amd64 #1 SMP Debian [...]

t-paul, 2019-10-25:
True, it's triggered by a change in OpenSCAD. Other applications are probably not a good reference as OpenSCAD still uses ancient OpenGL, partially even fixed pipeline which is probably (hopefully?) not used much anymore.
I tried to check with `RenderDoc` but this requires OpenGL 3.2+ context. So it might be OpenSCAD doing something wrong with the API but without tools that's going to be hard to find. Maybe someone reading this has pointers to other ways of validating OpenGL API calls?

t-paul, 2019-10-26:
Hmm, `git bisect` produced
```
git bisect start
# bad: [6ec97fda154b0578688ab98723685c66af7a7a71] More MacOS builds.
git bisect bad 6ec97fda154b0578688ab98723685c66af7a7a71
# good: [be4fd23dc6b9f4de44c5a4c5cee94894dc139393] Cleaned out old 32-bit Mac builds
git bisect good be4fd23dc6b9f4de44c5a4c5cee94894dc139393
# good: [bf6153592bebd0299722b4c535ef827bdaeec5f1] make the default value of getVec3 explicit by overloading getVec3
git bisect good bf6153592bebd0299722b4c535ef827bdaeec5f1
# bad: [1434fb9ec370294aa18205c3bbdb8d21db89750e] Check only actual parameter sets for isEmpty().
git bisect bad 1434fb9ec370294aa18205c3bbdb8d21db89750e
# good: [c31b627be142f3304b8fa6870a53a3b10ce2c17a] Merge pull request #2638 from MichaelPFrey/para
git bisect good c31b627be142f3304b8fa6870a53a3b10ce2c17a
# bad: [9147d940f3464ae59b113689f8fde1c4df003cfc] Merge pull request #2708 from openscad/3d-printing-updates
git bisect bad 9147d940f3464ae59b113689f8fde1c4df003cfc
# good: [ed93d3fec5d11bedc06f5200a3a740a72c7b6c56] Merge pull request #2672 from openscad/3d-printing-init-dialog
git bisect good ed93d3fec5d11bedc06f5200a3a740a72c7b6c56
# good: [155f90406cb0a86840208567531596c4da03e810] move the "try catch" for readability and to make it more modular
git bisect good 155f90406cb0a86840208567531596c4da03e810
# good: [88417b32b873912ded901630c4198c0828911916] clean up
git bisect good 88417b32b873912ded901630c4198c0828911916
# bad: [69bf5a55475102be43c2ddc698198e85821332b3] Merge pull request #2697 from MichaelPFrey/ParaRangeCheck
git bisect bad 69bf5a55475102be43c2ddc698198e85821332b3
# good: [8b6e905f68d7ce19db1ef3e649dc85c25b315cd9] Merge pull request #2706 from openscad/amf-remove-experimental-flag
git bisect good 8b6e905f68d7ce19db1ef3e649dc85c25b315cd9
# good: [0d1bc651f0d4e407905f477d99965b17e2e2b144] clean up
git bisect good 0d1bc651f0d4e407905f477d99965b17e2e2b144
# bad: [d26ea931d7bb0583496e1245ef9c7fa3de871939] Merge pull request #2710 from openscad/input-driver-remove-experimental-flag
git bisect bad d26ea931d7bb0583496e1245ef9c7fa3de871939
# bad: [2b505f1dd943d0051db2130d5029bdb7ebf6d830] Change input-driver experimental flag to cover only the DBus driver (fixes #2704).
git bisect bad 2b505f1dd943d0051db2130d5029bdb7ebf6d830
# first bad commit: [2b505f1dd943d0051db2130d5029bdb7ebf6d830] Change input-driver experimental flag to cover only the DBus driver (fixes #2704).
```
Which seems pretty strange as the offending commit does not touch any of the display logic. Assuming it could have something to do with Qt initalizing stuff, I disabled the change in the bad commit and it worked afterwards. But after applying the similar change to `master` it did not help. And now even recompiling some of the good commits are broken. So I'm a bit at a loss here.

t-paul, 2019-10-27:
So the issue is triggered when processing the custom `QEvent`s created from the input driver code via `QCoreApplication::postEvent()`. It's not clear how that would deadlock in the OpenGL driver and I don't see a way to fix that yet.

KlausKnopper, 2022-02-03:
Workaround: In input/InputDriverManager.cc (openscad version 2021.01), replace

```
void InputDriverManager::postEvent(InputEvent *event){
 ...  QCoreApplication::postEvent(window, event);
}

```

by an empty function, and openscad works again compiled in 32bit running on a 64bit kernel, no freezes anymore. As far as I can tell, keyboard and mouse are still working without the call to  postEvent().

My best guess is that for some reason, input events may get retransmitted in an endless loop in the combination of 32bit application+64bit kernel.

So, the DRM messages observed in the log may have been unrelated or caused by openscad trying to readjust the graphics canvas too frequently because of exceed input events.

KlausKnopper, 2022-02-03:
Hello Torsten,

Your hint about QCoreApplication::postEvent(), which I first thought of
being unrelated to an alleged graphics problem, actually yields _the_
workaround for me. Wish I had tried to just remove the call just to
check on what happens much earlier. Please see my comment in issue
#2933.

Frankly, I do not understand why input events should get retransmitted
to the same window they came from, which is what the function seems to
do in my thinking. Maybe it has something to do with the new support for
space navigation drivers and translation/modification of events, but it
seems to create an indirectly recursive event loop somehow.

After simply removing the call, the issue is gone. Though it is probably
not the correct solution, I can live with that private modification for
Knoppix. :-)

While picking through the source, I also noticed

src/export_stl.cc:
  if (triangle_count > 4294967295)

comparison is _always_ false due to the limited range of triangle_count
in 32bit architecture. I believe there are several places in openscad
where 64bit integers are assumed implicitly. Probably unrelated to
issue #2933 .

Regards
-Klaus

t-paul, 2022-02-03:
Yes, this is only used for 3D-Mouse / Gamepad and so on. As those will be non-functional, it might make sense to also make sure those devices are disabled.

I think the easy config options were only added after the release for the `cmake` build. The options would be
* `ENABLE_JOYSTICK`
* `ENABLE_QGAMEPAD`
* `ENABLE_HIDAPI`
* `ENABLE_SPNAV`
* `ENABLE_DBUS`

closed 2026-03-28 by github-actions[bot]