openscad/openscad #2934: modulecache-tests - failed?

MichaelAtOz, 2019-04-15:
I just grabbed the first test mentioned here:
[openscad/testdata/modulecache-tests/README.txt ](https://github.com/openscad/openscad/blob/master/testdata/modulecache-tests/README.txt)

use.scad & used.scad

> Test1: Basic cache
o Turn off Design->Automatic Reload and Compile
o Open use.scad
o Compile twice (F5) - check that module reference is the same and that ModuleCache only says "Compiling library: /path/to/used.scad" once.
o touch used.scad
o Compile (F5) - check that ModuleCache says "Recompiling cached library: /path/to/used.scad".

After running 2019.01-RC4-32 with the debug options mentioned I did not get the `recompiling cached library:` mentioned, just `compiled module:` (& not `Compiling library:`)

Note adding the `ECHO: "used.scad: module used(r=", 20, ") - green sphere."` was my 'touch', which does indicate it loaded the new used.scad.

Readme out of date?

```
Loaded design 'C:/Users/MeB/Documents/3D-REPRAP/OpenSCAD/GUI Testbed/git_tests/use.scad'.
Parsing design (AST generation)...
ModuleCache: compiled module: C:/Users/MeB/Documents/3D-REPRAP/OpenSCAD/GUI Testbed/git_tests/used.scad
FileModule: C:/Users/MeB/Documents/3D-REPRAP/OpenSCAD/GUI Testbed/git_tests/used.scad: 0 -> 0x96b0f50
Used file cache size: 1 files
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 1
Geometry cache size in bytes: 62032
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

Parsing design (AST generation)...
FileModule: C:/Users/MeB/Documents/3D-REPRAP/OpenSCAD/GUI Testbed/git_tests/used.scad: 0x96b0f50
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 1
Geometry cache size in bytes: 62032
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

Parsing design (AST generation)...
ModuleCache: compiled module: C:/Users/MeB/Documents/3D-REPRAP/OpenSCAD/GUI Testbed/git_tests/used.scad
FileModule: C:/Users/MeB/Documents/3D-REPRAP/OpenSCAD/GUI Testbed/git_tests/used.scad: 0x96b0f50
Used file cache size: 1 files
Compiling design (CSG Tree generation)...
ECHO: "used.scad: module used(r=", 20, ") - green sphere."
Compiling design (CSG Products generation)...
Geometries in cache: 1
Geometry cache size in bytes: 62032
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Compile and preview finished.
```

nophead, 2019-04-15:
Did you build a debug version from source? Release versions don't print debug information even if you turn it on with command line.

MichaelAtOz, 2019-04-15:
Snapshots & the RC's seem to be debug versions, as above shows, e.g. :
```
ModuleCache: compiled module: C:/Users/MeB/Documents/3D-REPRAP/OpenSCAD/GUI Testbed/git_tests/used.scad
FileModule: C:/Users/MeB/Documents/3D-REPRAP/OpenSCAD/GUI Testbed/git_tests/used.scad: 0x96b0f50
Used file cache size: 1 files
```

nophead, 2019-04-15:
This is what I get with a debug version I compiled with MSYS2
```
OpenSCAD 2019.03.24
https://www.openscad.org/

Copyright (C) 2009-2019 The OpenSCAD Developers

This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.

Loaded design 'D:/msys64/home/ChrisP/openscad/testdata/modulecache-tests/use.scad'.
Parsing design (AST generation)...
ModuleCache: Compiling library 'D:/msys64/home/ChrisP/openscad/testdata/modulecache-tests/used.scad'.
ModuleCache: compiled module: D:/msys64/home/ChrisP/openscad/testdata/modulecache-tests/used.scad
FileModule: D:/msys64/home/ChrisP/openscad/testdata/modulecache-tests/used.scad: 0 -> 0x74779d00
Used file cache size: 1 files
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 1
Geometry cache size in bytes: 62072
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

Parsing design (AST generation)...
FileModule: D:/msys64/home/ChrisP/openscad/testdata/modulecache-tests/used.scad: 0x74779d00
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 1
Geometry cache size in bytes: 62072
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

Parsing design (AST generation)...
ModuleCache: Recompiling cached library: D:/msys64/home/ChrisP/openscad/testdata/modulecache-tests/used.scad (5cb45187.55)
ModuleCache: compiled module: D:/msys64/home/ChrisP/openscad/testdata/modulecache-tests/used.scad
FileModule: D:/msys64/home/ChrisP/openscad/testdata/modulecache-tests/used.scad: 0x74779d00
Used file cache size: 1 files
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 1
Geometry cache size in bytes: 62072
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds
```

The code is here: https://github.com/openscad/openscad/blob/b58875721e2c98ac4b9bbb31d0968671beb507e7/src/ModuleCache.cc#L84

It is protected by ifdef DEBUG. I doubt that snapshots are debug versions because they run very slowly.

t-paul, 2019-04-15:
Theory says:
- Debug versions (usually only built by developer locally): DEBUG = Yes, Experimental = As Needed
- Snapshot versions: DEBUG = No, Experimental = Yes
- Release versions (and RCs): DEBUG = No, Experimental = No

At least for RC4 that is supposed to be true, earlier RCs were not correctly following that.

The DEBUG status is now shown in the Library Info, Experimental is enabled if the Features tab is shown in Preferences, maybe we could add that to Library Info too, just for completeness.

nophead, 2019-04-15:
Yes I think the confusion comes from PRINTD, which works in release mode when enable on the command line but not if it guarded with #ifdef DEBUG. So you get a subset of the messages in release mode.