openscad/openscad #2937: render() affects CGAL rendering

Ghostkeeper, 2019-04-25:
*Summary:* The `render()` function prevents a part of my model from being rendered in render mode, but not in preview mode.

I'm seeing a difference between preview mode and render mode here which I can't explain. Preview mode shows correctly, but render mode does not. The funny thing is though that the bug only occurs when I'm using the `render()` function to speed the preview mode performance up.

Versions I've seen this in are: 2017.04.05 on Windows, and 2019.01 RC2 on Ubuntu (via Apt).

This bug originally occurred for me in [this](https://github.com/Ghostkeeper/Stormcrest/blob/master/models/gutter_lift.scad) model but I've stripped it down to a minimal example as well as I could:

```
cylinder(r=30, h=40);

render(convexity=4) {
	translate([0, 0, 12.1]) {
		minkowski() {
			cube([10, 10, 37.9]);
			sphere(r=10);
		}
	}

	translate([0, 0, 50]) {
		cylinder(r=11, h=1);
	}
}
```

If you remove any element here the bug will not reproduce any more. Also if you change numbers related to the Z coordinate, it will not reproduce any more. This tells me that it has to do with an edge case in CGAL somewhere. Always fun. But more surprisingly it tells me that putting that `render()` call there broke my model, which is something that you may be able to fix.

This is further aggravated by an error that gets posted in the log when you render this:
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: pe_prev->is_border() || !internal::Plane_constructor<Plane>::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate() File: /opt/mxe/usr/x86_64-w64-mingw32.static/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h Line: 251
```

Here's how it looks in preview mode as well as when you remove the `render()` function (correct):
[image]

Here's how it looks in render mode (incorrect):
[image]

MichaelAtOz, 2019-04-25:
If you Design/Flush_Caches before Render it works.
There has been some issues in the weeds with F5 'rendered' cache geometry being slightly off compared to CGAL render.

With RC4 I get a different error
> ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 426

kintel, 2024-03-30:
Status as of 2024-03-24: Still broken both for CGAL and Manifold

pca006132, 2024-09-05:
Works with manifold now.

MichaelAtOz, 2026-03-29:
With 2026.02.01 works with both CGAL & Manifold. Closing as complete.