openscad/openscad #2939: Make the preview area into a new window

LincolnAuster, 2019-04-29:
I would like to be able to pop the preview section into a new window that I can have take up another monitor to see more detail on the model and have more room for console/preferences/code. This would hopefully be similar to how Cryengine lets you expand toolbars into total new windows and allows saving layouts.

Can I do this or does this need to be implemented?

t-paul, 2019-04-29:
You can separate/undock the code and console window.

LincolnAuster, 2019-04-29:
> 
> 
> You can separate/undock the code and console window.

How? I'm pretty new. I've looked in the 'View', 'File', 'Edit', etc menus and  I haven't seen anything.

t-paul, 2019-04-29:
Just by dragging the sub windows at the title bar (you may need to check Preferences as there's a setting that enables undocking - in the Advanced section).

For more general questions, please use the [forum/mailing list](https://www.openscad.org/community.html).

LincolnAuster, 2019-04-29:
That's exactly what I was looking for! Thanks!
Sorry; I was unaware of the forum and mailing list.

closed 2019-04-29 by LincolnAuster

t-paul, 2019-04-29:
No problem. Nice to hear it works :smile:.