openscad/openscad #2940: Copy viewport rotation produces scientific notation numbers with low precision

nophead, 2019-05-01:
[image]

[ 2.6e+02, 0, 3.3e+02 ]

It seems to occur when the numbers exceed 99.

nophead, 2019-05-01:
Looks like this commit broke it. https://github.com/openscad/openscad/commit/2bed5189ddeb1fccc7c6c709ac6c881d9d38d639

donbright, 2019-05-02:
automatic regression test for the GUI would help catch this stuff. 

sprintf  should be replaced with asprintf not something that behaves differently. 

https://doc.qt.io/qt-5/qstring-obsolete.html

t-paul, 2019-05-02:
It actually says: "Use asprintf(), arg() or QTextStream instead.". So `arg()` is one of the options officially proposed as replacement. The problem seems to be that `arg()` is too heavily overloaded so in some cases it's picking a different implementation not the intended one (see #2897).

commit d6b7b8a7 (2019-05-05): Use 'f' format instead of 'g' to prevent scientific number notation (fixes #2940).
commit e98a364e (2019-05-05): Merge pull request #2941 from openscad/number-format-fix

Use 'f' format instead of 'g' to prevent scientific number notation (fixes #2940)
closed 2019-05-05 by kintel