openscad/openscad #2944: Update docs for latest version of OpenSCAD

kintel, 2019-05-06:
* [ ] Cheat sheet: Go through, add new features, clean up if necessary
* [ ] SVG: Go through and clean up, specifically consider DPI and unit docs
* [ ] Replace "required nightly build" with requires <new version number>

MisterAwesome23, 2019-06-05:
I would like to work on this. Is anyone on this already?

kintel, 2019-06-05:
This is still up for grabs. I think the cheat sheet is pretty much up to date though.

MisterAwesome23, 2019-06-05:
> This is still up for grabs. I think the cheat sheet is pretty much up to date though.

Where can I review these files? 
Sorry I am super new here.

kintel, 2019-06-05:
The documentation is all in Wikibooks, linked from the website; I would start by looking at that structure and see how to best fit the missing docs.

MisterAwesome23, 2019-06-05:
I found the cheatsheet:
https://github.com/openscad/openscad.github.com/tree/master/cheatsheet
but as mentioned SVG needs more attention. I'm not able to track SVG (code location) and exactly and what exactly is expected.

t-paul, 2019-06-05:
I started with documenting SVG, but it's by far not complete.
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/SVG_Import

dabc312GitHub, 2020-03-29:
I want to contribute to this project or issue, although I am new into these kind things of open source, I read both links https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/SVG_Import and https://github.com/openscad/openscad.github.com/tree/master/cheatsheet... So I think that you want to increase information about SVG files in OpenSCAD to show in Wikibooks, right?

t-paul, 2020-03-29:
Correct, specifically the issue is about adding information about the new features supported, maybe also including the feature which is not yet in the release but only nightly builds (with a note in the docs about this, so people don't expect it to work in the release version).

gingerbeardman, 2021-02-12:
"jump to source code from 3D preview" on right click seems to be missing from docs.

kintel, 2024-03-30:
Closing this as the ticket was meant for OpenSCAD-2019.05

closed 2024-03-30 by kintel