openscad/openscad #2946: Cannot export model to 3mf, but stl works fine

jbcpollak, 2019-05-09:
When running on the command line, I am finding that I can export an STL, but I get a CGAL error when exporting to 3MF. I tried using the gui and was able to preview and render fine, but exporting to 3MF from the menu gave me the error `Intersection_of_constraints_exception: Intersection of constraints while using No_intersection_tag`.

```
$ openscad --version
OpenSCAD version 2019.01-RC4
```

Here is what I see on the terminal:

```
$ openscad -o quote.3mf /var/folders/_4/7rhjy87148140x058565h26m0000gn/T/uk-70mtyf/d3f73a4b-3894-409f-b326
-de748e595061/d3f73a4b-3894-409f-b326-de748e595061.scad
Fontconfig error: Cannot load default config file
\libc++abi.dylib: terminating with uncaught exception of type CGAL::Constrained_triangulation_2<CGAL::Projection_traits_xy_3<CGAL::Cartesian<CGAL::Gmpq> >, CGAL::Triangulation_data_structure_2<CGAL::Triangulation_vertex_base_2<CGAL::Projection_traits_xy_3<CGAL::Cartesian<CGAL::Gmpq> >, CGAL::Triangulation_ds_vertex_base_2<void> >, CGAL::Constrained_triangulation_face_base_2<CGAL::Projection_traits_xy_3<CGAL::Cartesian<CGAL::Gmpq> >, CGAL::Triangulation_face_base_2<CGAL::Projection_traits_xy_3<CGAL::Cartesian<CGAL::Gmpq> >, CGAL::Triangulation_ds_face_base_2<void> > > >, CGAL::Default>::Intersection_of_constraints_exception: Intersection of constraints while using No_intersection_tag
/Users/.../bin/openscad: line 2: 19657 Abort trap: 6           /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD $*

$ openscad -o quote.stl /var/folders/_4/7rhjy87148140x058565h26m0000gn/T/uk-70mtyf/d3f73a4b-3894-409f-b326
-de748e595061/d3f73a4b-3894-409f-b326-de748e595061.scad
$
```

t-paul, 2019-05-09:
Can you add enough information (= code snippets) to reproduce the issue. Otherwise I'll close as duplicate of #2847 which is most likely the cause. STL behaves differently as there's currently special code for STL only which can skip the conversion step that has the issue.

jbcpollak, 2019-05-10:
@t-paul please see this repo, I added reproductions of both the .3mf and .stl issues:

https://github.com/jbcpollak/openscad-import-problems

Note that both the .3mf and .stl were generated by `OpenScad 2019.01-RC4`

MichaelAtOz, 2019-05-10:
The stl generated is bad. 
Degenerate faces
[image]

Self intersections (top edge)
[image]

Care to share the code that made the base?

jbcpollak, 2019-05-11:
> Care to share the code that made the base?

Sure, its this file: https://github.com/jbcpollak/KeyV2/blob/master/oem-r3-w1-h1.scad

You'll need to check out the entire KeyV2 repo, then change into that directory. From there all you need to do is:

`openscad -o base.3mf oem-r3-w1-h1.scad` and it should work (or generate an stl if you want). Let me know if that doesn't work.

Also, what are you using to visualize that the stls are bad? Thanks.

MichaelAtOz, 2019-05-11:
> Also, what are you using to visualize that the stls are bad? Thanks.

Netfabb studio basic, it used to be a free limited version, [currently](https://www.autodesk.com/products/netfabb/free-trial) it's a trial, I think the trial used to go to the limited scope afterwards but I'm not sure.

jbcpollak, 2019-05-11:
Thanks! It's interesting Cura had no problem slicing this stl. Are the degenerate faces a problem with the project code, or with openscad itself?

MichaelAtOz, 2019-05-11:
> Also, what are you using to visualize that the stls

BTW in Netfabb, degenerate faces is via Part/Repair click the 'show...' box, then self intersection via Repair/Detect_self_intersection

> Are the degenerate faces a problem with the project code, or with openscad itself?

Mainly openscad, it is the 'small numbers problem', it is here

```
module hull_shape_hull(thickness_difference, depth_difference, extra_slices = 0) {
  for (index = [0:$height_slices - 1 + extra_slices]) {
    hull() {
      shape_slice(index / $height_slices, thickness_difference, depth_difference);
      shape_slice((index + 1) / $height_slices, thickness_difference, depth_difference);
    }
  }
}

module shape_slice(progress, thickness_difference, depth_difference) {
  skew_this_slice = $top_skew * progress;
  depth_this_slice = ($total_depth - depth_difference) * progress;
  tilt_this_slice = -$top_tilt / $key_height * progress;

  translate([0, skew_this_slice, depth_this_slice]) {
    rotate([tilt_this_slice,0,0]){
      linear_extrude(height = 0.001){
        key_shape(
          [
            total_key_width(thickness_difference),
            total_key_height(thickness_difference)
          ],
          [$width_difference, $height_difference],
          progress
        );
      }
    }
  }
}
```
Working in such small steps, openscad gets confused about precise positions, think of Brownian motion.
I fixed that by scale it up then down by 1000.

Self intersections are harder...

MichaelAtOz, 2019-05-11:
I'm suspecting offset() as in rounded_square.scad, similar to #2868.
Can't prove it yet, but I really think clipper integration problem...

jbcpollak, 2019-05-11:
Would it temporarily work around the problem to scale the whole KeyV2 project up by an an amount, render, and then scale it back down before final render or after import?

nophead, 2019-05-11:
I avoid such problems by ensuring none of my vertices are less that 1/128 units apart. I use that faction because it is an exact number in floating point and is bigger that the OpenSCAD grid snap. It also avoids Z-fighting in preview at normal viewing distances.

You simply can't have very close vertices in OpenSCAD because they collapse and cause either degenerate triangles, or self intersections. Even taking the hull of close vertices can make a non-manifold shape whereas mathematically it shouldn't be.

Fortunately 1/128mm is an order of magnitude smaller than anything noticeable in 3D printing.

jbcpollak, 2019-05-11:
Sorry, I'm not very proficient with Openscad yet, how do you ensure that?

nophead, 2019-05-11:
I keep the number of vertices in curves under control with $fa = 6 and $fs = extrusion_width / 2 = 0.25.

I have a constant called eps = 1/128. If I get problems with vertices too close I add or subtract it to move them apart.

If I union or difference 3D items that aren't numerically exact then I overlap by at least eps.

For a more complex example I had a stack of polygons, each the same size and layer thick but with variable numbers of vertices. I got problems with them lining up almost but not exactly, so I made each one eps smaller to avoid them ever being close.

MichaelAtOz, 2019-05-12:
The degeneratons are introduced with differencing a smaller key from the bigger key to hollow out the bottom. I was unable to get rid of them, so I looked at the basic key shape.

I've cut it down to this, still a bit verbose. Scaled * 1000 to remove any possible 'small number problems'.

[key-cutdown.scad.txt](https://github.com/openscad/openscad/files/3169536/key-cutdown.scad.txt)

Basically it is hull()ing pairs of slices to build the block, the slices are linear_extrude()'ed offset() square()'s.
[image]

Export as stl is bad. No degenerates or self intersects. (2019.01.RC4)

stl: [key-cutdown.zip](https://github.com/openscad/openscad/files/3169543/key-cutdown.zip)

Netfabb shows what I think must be from a hole (the opening moves around as you rotate the object), but it doesn't show any holes in its stats.
[image]

Something funky in OpenSCAD...

I removed the offset() still funky
[image]

stl: [key-cutdown-NoOffset.zip](https://github.com/openscad/openscad/files/3169546/key-cutdown-NoOffset.zip)

MichaelAtOz, 2019-05-12:
> No ... or self intersects. 

Spoke too soon, meshlab shows everything self intersecting.
[image]

MichaelAtOz, 2019-05-12:
Getting warmer...
Removed the hull()'s and added this to `rounded_square_shape_1000`
```
    w=width - extra_width_this_slice - extra_corner_radius_this_slice * 2;
    h=height - extra_height_this_slice - extra_corner_radius_this_slice * 2;
    echo(progress,w=w,h=h);
```
got
```
ECHO: 3, w = -1.84, h = 4.16
```

square([-1.84, 4.16]) is not good.

[image]

It does nothing standalone, so not sure that's up yet...

MichaelAtOz, 2019-05-12:
I did this, with & without the `*` & exported stls. Which is replicating the above slices.
```
union() {
linear_extrude(height=0.1)
scale(1000)
square([4.16, 8.16], center=true);
*linear_extrude(height=0.1)
scale(1000)
square([-1.84, 4.16], center=true);
}
```
In Meshlab the front looks normal:

[image]

But the back looks like this:

[image]

Meshlab does not show self intersections, but with Netfabb Repair/Detect-self-intersections the back shows (both stl's look the same):

[image]
And a close look at the edges in normal mode looks different.

Does anyone have any idea what that means?
stl with `*`: [test-single.zip](https://github.com/openscad/openscad/files/3169812/test-single.zip)
stl without: [test.zip](https://github.com/openscad/openscad/files/3169813/test.zip)

I then removed the union(), shows the same, so:
```
linear_extrude(height=0.1)
scale(1000)
square([4.16, 8.16], center=true);
```
Is producing that bad geometry, something funky in OpenSCAD???

MichaelAtOz, 2019-05-12:
> Would it temporarily work around the problem to scale the whole KeyV2 project up by an an amount, render, and then scale it back down before final render or after import?

Sorry @jbcpollak, missed that.
Normally that would get around the 'small number problem', but I tried that at scale 1000 and it still had problems. As you can see above there is something else going on. As those slices in `shape_slice()` are broken.

MichaelAtOz, 2019-05-12:
Removed the scale, so
```
linear_extrude(height=0.1)
square([4160, 8160], center=true);
```
Is badly broken.

nophead, 2019-05-12:
Actually it seems to be NetFabb that is broken. If you edit the STL file
and replace all the 0.1 z ordinates with 1 then NetFabb says it is OK. The
topology is unchanged so it can't be self intersecting.

Also a cube with the same dimensions is broken according to NetFabb.

MichaelAtOz, 2019-05-12:
What about Meshlab, that doesn't look normal.

nophead, 2019-05-12:
Both sides look fine for me in Meshlab. Your image looks like some form of Z fighting due to the very close proximity of the two large faces when zoomed out far enough to see the whole object.

The extreme ratio of size to thickness is what seems to confuse NetFabb. Odd because it seems well within the resolution of a float.

MichaelAtOz, 2019-05-13:
Round 2.

I just looked at it in Netfabb with Triangle Mesh ticked. 
[image]

I'm now thinking all those triangles converging may be the source of the self intersections.

nophead, 2019-05-13:
The question is are they real or false positives in netfabb as we have seen
how easily it is fooled by a thin cube.

MichaelAtOz, 2019-05-13:
The point of the OP was that export of that to stl, then import of that stl for further CSG got CGAL errors. 
So the self intersections are a real problem.
I'm trying to pin down why OpenSCAD is generating a bad stl.

Meshlab, shows the same triangle convergence, AND the whole thing is flagged as self intersections:
[image]
And the self intersections, in the post above Netfabb image, are co-located to those convergence areas.

There was one likely issue, of co-planer faces in `hull_shape_hull()`, where  the adjacent pairs of slices were a problem, I can't see them causing the degenerate faces shown above. 
However the smoothness of the curved face is causing the current triangulation to go to a singularity at those corners. There is also two-fold similar triangulation in the inside cut-out. 
I can think of a much better triangulation of those side faces, like alternating rather than converging.

nophead, 2019-05-13:
It Just looks like $fn is far too high. I don't get these types of problem
because I set $fa and $fs to values suitable for 3D printing. I never get
vertices so close together is a simple shape like that.

MichaelAtOz, 2019-05-13:
Obviously trying to get a work-around and changing resolution is something to look at.
But should OpenSCAD be limited to geometry suitable for current 3D printers?
If it creates an export that it can't use for import, then it is broken, surely.

nophead, 2019-05-14:
Yes  OpenSCAD is broken because it reduces resolution and snaps vertices
without preserving topology in many places. Fixing it is difficult.

MichaelAtOz, 2019-05-14:
Getting warmer.
By embedding the slices so they are not co-incident & reducing the facet resolution I got rid of self-intersections (SI), except for one. Focusing on the basic key shell shape, I've still left off the stem etc. for now.

But I'm at an impasse.

This is the key shape, with a smaller key shape differenced.
[image]

stl: [key-g.zip](https://github.com/openscad/openscad/files/3176070/key-g.zip)

The remaining SI is next to that blue line, on the inside top back left corner.
Note the lines on the inside top curved surface are yellow, joined to green vertical lines.
With opposite colours on the outside.

Netfabb:
[image]

Close up.

[image]

Problem is when you look close in wireframe in OpenSCAD, 
[image]
Those lines, next to the blue ones, slanting top down left to right are the upper surfaces, yellow on the inside, green on the out. The opposite slants are the verticals, just inside showing.

Note the squared off tops of the green lines, both the horizontal curved surface and the inside walls (1/2 of them) are polygons, not triangles. At that top end, at the other end they go to a point. That top squared end is 0.2 units, not that small.

So when the quad gets triangulated it turns to two tiny slivers. 

[image]

With what I think is reasonable compromise of curved face size, and no co-incident faces, how can you control for that?

nophead, 2019-05-14:
Not sure why you would need slices. I made a mock up with my library, the code looks like this:

```OpenSCAD
include <NopSCADlib/lib.scad>

width = 20;
depth = 15;
height = 10;

taper = 3;
rad = 2;
indent = 1;

d = (depth - taper);
r = indent / 2 + sqr(d) / 8 / indent;

module shape(offset = 0)
    difference() {
        hull() {
            rounded_rectangle([width - taper - 2 * offset, depth - taper - 2 * offset, height- offset], rad - offset, center = false);
            
            rounded_rectangle([width - 2 * offset, depth - 2 * offset, eps], rad - offset, center = false);
        }
        translate_z(height - indent + r)
            rotate([0, 90, 0])
                cylinder(r = r + eps + offset, h = width + 1, center = true, $fa = 2);
    }

difference() {
    shape();
    translate_z(-eps)
        shape(1);
}
```
 It is just the normal way I model in OpenSCAD, no special tricks.

This is what it looks like.

[image]

Wireframe looks like this:

[image]

How do you get such fine lines in your wireframe view?

The STL looks fine in NetFabb and I can import it back into OpenSCAD and union with a unit cube().

The way I use OpenSCAD I don't get problems like this because I avoid close vertices.

MichaelAtOz, 2019-05-14:
I'm examining the OP issue, by @jbcpollak 
I'm doing this investigation to see where the problems are, being original .scad code or OpenSCAD internal issues.

I like to help, I like a challenge, but I also have an interest in mechanical keyboards, which have 'historical' design parameters.

The OP is a fairly sophisticated open source suit with a wide range of keyboard key features.
So far I have back-tracked to the core key-shape, trying to get a valid stl. There is more complexity beyond my current findings.

I would like to make it work to match REAL WORLD KEYCAPS.

So, I'm not having a go at your mock up, but it is not matching the complexity of real keycaps.
I have examined the code, found things which my experience show are problematic and corrected them. (I have not posted all these code changes yet)

But with what I think is reasonable code, OpenSCAD is producing bad stl's.

If we can't make something like real keycaps, something is wrong.

I'm trying to find where OpenSCAD is failing in this endeavour.

> The STL looks fine in NetFabb and I can import it back into OpenSCAD and union with a unit cube().

OK, I hadn't imported it, but did you do Repair/Detect-self-intersections? Netfabb does not do that as its automatic check. That is where the above SI images are from.

What is the point if we do not examine where these problems come from.

jbcpollak, 2019-05-14:
Thank you @MichaelAtOz - I appreciate you digging into this. I agree if OpenSCAD can't reproduce a real-world example like this, its a problem with the tool. Its great if there are work arounds and ways to make the modeling language produce valid output, but I think it is reasonable to expect that if you can model it, OpenSCAD should be able to render it, or give an error explaining why its impossible.

> I have examined the code, found things which my experience show are problematic and corrected them. (I have not posted all these code changes yet)

Are these updates to KeyV2 or OpenSCAD?

I'm at a bit of a loss on how to dig in to help here, but I am following along and learning quite a bit about OpenSCAD reading the example code posted here.

nophead, 2019-05-14:
Yes I am doing Repair/Detect-self-intersections and then hunting for coloured bits but as it gives a false positive for a thin cube I have lost confidence in its results. The fact CGAL is happy with it being imported again indicates it is a good STL as it surely barfs at any non-manifoldness.

Yes it doesn't help find the problem in the real library but that is far too complex for me to analyse. What I am trying to show is a style of modelling in OpenSCAD that just works without these problems. If you intersect or even take the hull of curves with too close vertices you will run into problems with OpenSCAD. It just can't do that, so there is not much point trying to analyse further. My mock up looks detailed enough for real world use but has far less vertices and in particular thin sliver triangles. The problem with those is if you intersect or difference near to the point you generate two vertices very close together.

[image]

MichaelAtOz, 2019-05-15:
> Are these updates to KeyV2 or OpenSCAD?

KeyV2. I'll list them eventually, I want to make sure as they may be some interrelated causes.

I mistakenly had the degenerated box un-checked in Netfabb, it does show one, next to the corner self-intersection, red arrow, next to the SI, yellow arrow.
[image]

While Meshlab shows nothing. The above posts Meshlab red bits was my bad, un-checking unify dupl. vertices. :/

I then just rendered the smaller keycap by itself, ie what is differenced. 
[image]

It shows a degenerated face in the same spot as in the inside, but no SI. 
Meshlab shows nothing. So I'm now with Nophead, not trusting Netfabb for this.

So it looks like the basic shell is OK now. I'll add the stem etc. next & see...

nophead, 2019-05-15:
Your internal view doesn't seem to be showing all the triangulation. It has quads and polygons. It doesn't match the external triangulation of the bit you subtracted.

My STL looks like this in NetFabb. Everything is triangulated.

[image]

MichaelAtOz, 2019-05-18:
> Your internal view doesn't seem to be showing all the triangulation.

Yeh, I had triangulation off and show edges at 0 deg.

This is with triangulation:

[image]

> Meshlab shows nothing. So I'm now with Nophead, not trusting Netfabb for this.

Now I'm not sure.

Progress update: Aaarrrghh!

So we have the inside solid which is subtracted from the larger key solid.
Netfabb shows one degenerated face. Note AFAIK Meshlab can't show those??
No SI's. (this is outside the small solid)
[image]
The hollowed keycap then shows no SI's in Meshlab, but one in Netfabb + one degenerated face

But when I added the tines, which puncture that face, SI's again:
[image]

That point is shared with a triangle which has another point in common with the Netfabb SI line.

So now I'm sure Netfabb is correctly showing a problem.

So I'm back to why the smaller solid has the degeneration v's the larger which doesn't...

Progress...(?)

I think this is the definition of degeneration in Netfabb:
[image]

There are two large triangles, one slim triangle (green) and the really skinny <0.001mm triangle (yellow).
This is down in the [Brownian motion ](https://en.wikipedia.org/wiki/Brownian_motion)level...

My next step is just to 'fiddle' with $fn etc until it produces a good stl. Unless anyone has a scientific approach?

MichaelAtOz, 2019-05-18:
[image]
The red dot at the top corner is the problem area.
Close up:
[image]

That cube is 0.004. T=triangle, Q=quad.

This is the base object un-`dish()`'ed, `cube(0.4)`
[image]

Zoom, `cube(0.04)`
[image]

Zoom, `cube(0.004)`
[image]

Zoom yellow box
[image]
That is a ledge where the two slices meet. That makes a small ridge running down (on the right).
When that get `dish()`'ed, chopped horizontally, who knows what happens, but I think the ridge is artificially flattened.
Anyway the result is a tiny sliver quad, which gets triangled on export into the tiny 'degenerated' face according to Netfabb, ie one edge <0.001.

Root cause, probably the OpenSCAD grid??
Controllable cause, the join of the slices.

TBC...

MichaelAtOz, 2019-05-18:
Hmm...seem `hull()` may have a problem.
The ledge was caused by my addition of a small embedding overlap
When I removed that and only used two slices, the base & top hull()'ed, there was no ledge, but there was the vertical ridge.

Load this CSG
```
//hull() 
{
  color([0.501961, 0.501961, 0.501961, 1]) {
    group() {
      group();
      multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
        multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
          linear_extrude(height = 0.015625, center = false, convexity = 1, scale = [1, 1], $fn = 0, $fa = 12, $fs = 0.1) {
            group() {
              group() {
                group() {
                  group() {
                    group() {
                      group();
                      group();
                      offset(r = 1, $fn = 0, $fa = 12, $fs = 0.1) {
                        square(size = [13.16, 13.16], center = true);
                      }
                    }
                  }
                }
              }
            }
          }
        }
      }
    }
  }
  #color([1, 0, 0, 1]) {
    group() {
      group();
      multmatrix([[1, 0, 0, 0], [0, 1, 0, 1.7], [0, 0, 1, 10.5], [0, 0, 0, 1]]) {
        multmatrix([[1, 0, 0, 0], [0, 0.994522, -0.104528, 0], [0, 0.104528, 0.994522, 0], [0, 0, 0, 1]]) {
          linear_extrude(height = 0.015625, center = false, convexity = 1, scale = [1, 1], $fn = 0, $fa = 12, $fs = 0.1) {
            group() {
              group() {
                group() {
                  group() {
                    group() {
                      group();
                      group();
                      offset(r = 1, $fn = 0, $fa = 12, $fs = 0.1) {
                        square(size = [7.16, 9.16], center = true);
                      }
                    }
                  }
                }
              }
            }
          }
        }
      }
    }
  }
}
```
This is the two slices, with hull commented out.
[image]

Close up of top slice in problematic area, -x,+y +z corner.
[image]

Same corner on bottom.
[image]

Then uncomment hull(). Top join.
[image]

In Netfabb
[image]

stl: [hull-problem.zip](https://github.com/openscad/openscad/files/3193735/hull-problem.zip)

MichaelAtOz, 2019-05-18:
Then again maybe not...

My 3D perspective is being challenged.

Here it is with a thicker top slice, the base of the top protrudes outwards more as it is tilted.
stl: [hull-problem-thick.zip](https://github.com/openscad/openscad/files/3193762/hull-problem-thick.zip)

[image]

Is that a valid triangle? It seems twisted.
I'm going to rest my brain...

nophead, 2019-05-18:
I don't think tiny triangles are actually degenerate if they have finite
area. They shouldn't cause a problem as long as they don't collapse when
exported to STL and imported again.

I avoided the top ridge by making the top rounded rectangle the full
height. My bottom shape is a thin slice but not too thin but as it makes a
convex ridge it doesn't create thin slivers up the edge.

 It is a shame hull can't work on 2D shapes. To get a true hull without the
ridge one can lower the bottom slice and then slice it off with difference.
That gives a clean shape with no close vertices before subtracting the
finger depression from the top.

[image: image.png]

[image: image.png]

MichaelAtOz, 2019-05-18:
@nophead care to tilt (6 deg) your top face & see what happens?

MichaelAtOz, 2019-05-18:
It's lucky I like a challenge.
I'm off to drink red wine while watching election results...that should reset my brain.

nophead, 2019-05-18:
It still looks like a clean mesh and imports correctly but NetFabb showed
lots of degenerate triangles and a few self intersections. I reduced the
tolerance in the settings by a factor of 10 and the degenerate triangles
are gone. I still have a few self intersections but they can't be real as
CGAL doesn't object to them.

[image: image.png]

MichaelAtOz, 2019-05-18:
@nophead your images aren't showing.

nophead, 2019-05-18:
Github seems a bit broken ATM

nophead, 2019-05-18:
Hmm, when I edit it I see a message "Email replies don't support Markdown" so it doesn't show the images inline.

nophead, 2019-05-18:
I don't think tiny triangles are actually degenerate if they have finite area. They shouldn't cause a problem as long as they don't collapse when exported to STL and imported again.

I avoided the top ridge by making the top rounded rectangle the full height. My bottom shape is a thin slice but not too thin but as it makes a convex ridge it doesn't create thin slivers up the edge.

 It is a shame hull can't work on 2D shapes. To get a true hull without the ridge one can lower the bottom slice and then slice it off with difference. That gives a clean shape with no close vertices before subtracting the finger depression from the top.

[image]

[image]

nophead, 2019-05-18:
It still looks like a clean mesh and imports correctly but NetFabb showed lots of degenerate triangles and a few self intersections. I reduced the tolerance in the settings by a factor of 10 and the degenerate triangles are gone. I still have a few self intersections but they can't be real as CGAL doesn't object to them.

[image]

MichaelAtOz, 2019-05-19:
@jbcpollak 

The root cause is the slicing in `hull_shape_hull()` & the 'small number problem' (SNP*) which is triggered by the small $fs.

\* SNP is a really annoying, but difficult issue where floating point conversion loses detail.

Re the slicing.
https://github.com/jbcpollak/KeyV2/blob/5e07df75d3ec208198c4a7bb6f476dcdd8bba8ef/src/key.scad#L47
`// extra_slices is a hack to make inverted dishes still work`

That plus the slicing method, introduced 'discontinuities' that along with small $fs lead to long sliver like triangles particularly on the corners, when exported, due to SNP, caused self intersections. Which is why you couldn't `import()` the stl.

The slicing method, of `hull()`'ing the two slices and just butting those pieces up to each other causes one of the major things you want to avoid, co-incident faces. The call to `shape_slice()` at the top of one layer is identical to the bottom of the next. In CSG the algorithms find it hard to know whether a point is meant to be in one object, or the other, or both. 

The usual approach is to embed the joining face slightly inside the other object.
I tried that, but with the slope it introduced a little edge, which exacerbated the SNP.
Also note that the extra_slices didn't work properly, as the top slice had a negative width, and hence didn't exist.

The height of the base key shape didn't help, making the slivers thinner.
[image]

This is the hack I ended up using. Obviously it should be cleaned up.
```
module hull_shape_hull(thickness_difference, depth_difference, extra_slices = 0) {
  //for (index = [0:$height_slices - 1 + extra_slices]) 
  {
  index=0;
    hull() 
    {
      shape_slice(index / $height_slices, thickness_difference, depth_difference);
      shape_slice((index +2) / $height_slices, thickness_difference, depth_difference);
    }
  }
}
```
I changed `index +2` to `index +1.5` later, you will want to make that as small as possible to fit whatever other things sit on top of the keyface.

Re small $fs.

 I note in
https://github.com/jbcpollak/KeyV2/blob/5e07df75d3ec208198c4a7bb6f476dcdd8bba8ef/src/dishes/cylindrical.scad#L1
```
 // .5 has problems starting around 3u
$fa=.25;
```

With the global $fs=0.1 the combination of the curved upper face and the corners make this:
[image]

You can see how the SNP is going to be triggered, particularly in places like the single points all those triangle converge to, note the degenerate faces radiating from there. That lead to self intersections on export, and degenerate faces (which sometimes are not a problem, but sometimes are).

Changing that to $fa=5 and global $fs (in `key.scad`) to $fs=0.3 (with the `index +1.5`) I was able to export a clean stl, no SI's no degenerate faces.
[image]
stl: [key8.zip](https://github.com/openscad/openscad/files/3194710/key8.zip)

I think the curves are reasonable compromise. You could experiment with different $fa/$fs for the top curve & corners. If there are problems with wider keys (`// .5 has problems starting around 3u`), you should try making $fa proportional to the key width, not just small overall.

Obviously it would be better if OpenSCAD coped with those issues, but the SNP is a hard nut, as it is a performance trade off too.
I suspect making the whole thing bigger by say 1000 and scaling the final object down externally *may* help if you want a finer curve.

Note I only checked the code paths based on the `settings.scad` in `src`, there could be other issues hiding.

Other things in passing.

This does nothing, presumably a left over
https://github.com/jbcpollak/KeyV2/blob/5e07df75d3ec208198c4a7bb6f476dcdd8bba8ef/src/key.scad#L293

In `tines_support()` you are also just butting the ends of the tines against the inside surface. As they are flat surfaces aligned to the axis it works ATM, if there were a rotation involved the points on the coincident faces would be a problem. If you extend the length by 1/4 `$wall_thickness`, then the ends will embed into the key.

As per hophead's post near the top you may want to use 1/128 etc instead of 0.005 etc. Depending on the overall size you need to make those values proportionally bigger, eventually they don't work for the cross-hatching etc. BTW whenever you see cross-hatching ([z-fighting](https://en.wikipedia.org/wiki/Z-fighting)) you should examine it, as it means coincident faces.

The top curve was not extending to the full width of the face, at
https://github.com/jbcpollak/KeyV2/blob/5e07df75d3ec208198c4a7bb6f476dcdd8bba8ef/src/key.scad#L132
I changed it to -0.05.

MichaelAtOz, 2019-05-19:
I had a look at the bigger scope of the project.
I suspect the `hull_shape_hull()` above is not going to work for other key profiles.

nophead, 2019-05-19:
>I suspect making the whole thing bigger by say 1000 and scaling the final object down externally may help if you want a finer curve.

Perhaps, but the nature of floating point is scaling doesn't change much. For example if you scaled by 1024 the only thing that would change is the exponents. The mantissa resolution would be exactly the same.

You would be more likely ensure vertices don't snap to the same OpenSCAD grid point, but the grid will produce identical floats for different cells when the numbers get big enough. It shouldn't snap to a constant grid, it should snap to points resolvable in floats.

MichaelAtOz, 2019-05-19:
Hmmm...~~re-examining this,~~ Oops I was about to go through that loop again.

I missed that @jbcpollak repository was a fork, of https://github.com/rsheldiii/KeyV2. 
I'm wondering if @rsheldiii has had problems with bad stl's?

Also concentrating on the bad stl, I was looking at the generic problem rather than the OP problem with `oem-r3-w1-h1.scad`, I'll have a look tomorrow, but it's probably the same base issues.

jbcpollak, 2019-05-19:
On Sun, May 19, 2019, 4:14 AM Michael <notifications@github.com> wrote:

> Hmmm...re-examining this, Oops I was about to go through that loop again.
>
> I missed that @jbcpollak <https://github.com/jbcpollak> repository was a
> fork, of https://github.com/rsheldiii/KeyV2.
> I'm wondering if @rsheldiii <https://github.com/rsheldiii> has had
> problems with bad stl's?
>
Other than my addition of the example problem stl, I don't think there are
any differences between my repo and upstream. Also @rsheldiii may not have
noticed any problems because the generated STL files seem to slice fine in
3D printing software. They just don't re-import into OpenSCAD well. It's
only when I tried to use the stls as an external cache that I ran into
problems.

>

rsheldiii, 2019-05-19:
Wow, I'm honored by how much effort everyone's put into dissecting this. I'll respond to the things I've seen that might need extra context.

@jbcpollak is right, I haven't had any slicing issues, so I haven't been aware of these problems. I've always had issues with importing STLs in openSCAD, which I now know is because I'm often working on small models that would exhibit float rounding, so when KeyV2 exports wouldn't import cleanly it didn't set off any alarms.

I knew that co-incident faces are best avoided, but given the purpose of the shape slices it was the only solution I could think of besides implementing a `skin`-based approach. This library is an attempt to create extremely accurate recreations of keycap profiles that already exist, and some keycaps have progressively rounded shapes: 

[image]

which is where the shape slicing comes in. `$linear_extrude_shape` can be set to `true` to avoid `hull_shape_hull`, but it gets rid of this rounding. I think the final solution is to convert all shapes from 3d objects to functions that produce polygons, then use a skinning library to join them together, which also allows for concave shapes (ISO enter keys).

I'm not seeing how the `extra_slices` causes any additional problems, as it just continues the shape of the keycap slightly further than normal. In keycaps with normal dishing, these slices will be a safe distance within the dish and shouldn't contribute any geometry. They could create issues with inverted dishes, though without them the inverted dish doesn't exist. It's easy to disable them for normal keycaps however, and will make rendering faster.

As for the dishing itself, the rounding is high due to scaling issues with larger key widths / heights; as @MichaelAtOz suggested I'm going to play with scaling the rounding value for larger keys. I think the final solution is to generate a polyhedron that is the exact shape of the dish, which would allow you to optimize vertex distribution across the surface. That's a fair bit of math though.

The codebase is definitely a bit sprawling, but I tried to separate it as much as I could to make understanding it easier. I wrote up a document about the constituent parts of a keycap: https://github.com/rsheldiii/KeyV2/wiki/Technical-Design-of-a-Keycap which might help a little. 

Approaching this from a different perspective, what are you trying to do @jbcpollak that would need the keycaps as an external cache? could you use `render()` around the keycap to cut down on successive render calls?

jbcpollak, 2019-05-19:
@rsheldiii - to keep the thread on topic I'll reach out directly to explain what I'm working on.

MichaelAtOz, 2019-05-20:
> Also concentrating on the bad stl, I was looking at the generic problem rather than the OP problem with `oem-r3-w1-h1.scad`, I'll have a look tomorrow, but it's probably the same base issues.

Just changing `dishes\cylindrical.scad` $fa=3  exports a stl that will `import()` and `union()` with  `cube()`, so passes CGAL checks, it has degenerate faces in Netfabb, but no SI's in Netfabb or Meshlab.
[image]

However 3mf export fails CGAL on import & union.
` CGAL ERROR: assertion violation! Expr: G.mark(v1,0)==G.mark(v2,0)&& G.mark(v1,1)==G.mark(v2,1)`

A bit of searching suggest that is due to self intersections, and found a reference to
[`CGAL::Polygon_mesh_processing::does_self_intersect()`](https://doc.cgal.org/4.7/Polygon_mesh_processing/group__PkgPolygonMeshProcessing.html#ga0beba25d03a16010b20e2c9b6771cd12)
I'm wondering it that may be useful somewhere?
 
I don't have any tools to look at 3mf. Anyone know of any for Windows 7?

MichaelAtOz, 2019-05-20:
Changing to $fs=0.3 in `key.scad` (plus the $fa=3 above) exports a 3mf which imports and unions.

Also found [`CGAL::Polygon_mesh_processing::self_intersections()`](https://doc.cgal.org/4.7/Polygon_mesh_processing/group__PkgPolygonMeshProcessing.html#ga0afb9e365a2f16f0591255b8df468885)
> detects and records self-intersections of a triangulated surface mesh

thehans, 2020-06-07:
@jbcpollak I just reviewed this issue and I was able to reproduce an error in OpenSCAD 2019.05, but it appears to be fixed in the latest nightly builds. I ran the script from the repo in your [2nd comment](https://github.com/openscad/openscad/issues/2946#issuecomment-491149289) 

I'm not sure who or which commit apparently fixed this, but can you confirm if it works for you, running the most recent dev snapshot for your platform?  http://www.openscad.org/downloads.html#snapshots

kintel, 2024-02-27:
Closing as fixed

closed 2024-02-27 by kintel