openscad/openscad #2947: svg import: ring renders as dot on a certain threshold

akust, 2019-05-09:
**pre message**
first let me THANK YOU very much for the the whole project, clean language and continued developement. I love the new (compared to release) and super helpful features - it's so satisfying working and learning CAD and creating 3D objects this way. Customizer, for loop inside variable, extended import - phantastic features I never want to miss anymore. If you ask me you should release it, so everyone can use it. The following bug is just a tiny issue ;)

## Bugreport
**OpenSCAD Version**
2019.03.03 (git abd9ae92) 
**OS**
macOS Sierra 10.12.6

**Bug Description**
trying to extrude special QR-Code (qrcode-monkey.com) downloaded .svg, I found that rings sometimes/often interpreted wrongly in OpenSCAD. After intese trying I can offer an edge case as svg (attached) you can see both, 1 correct and 1 wrong extruded ring.

**steps reproduce**
import attach svg and linear extrude it, preview in openSCAD

**expected behavior**
it should extrude 2 rings of almost same size

**observed behavior**
it extrudes 1 ring (correctly)  and 1 dot (incorrect and diameter is smaller)

_details about the content in svg_
- rings are made by standard cirlce/elipse tool
- working in Inkscape (v0.91) I use select tool (F1) to select a ring and enter the diameter size of it via value entries in the toolbar (width = hight)
- left ring outer diameter is 19.1**20**
- right ring outer diameter is 19.1**19**
yes: this is the only difference what causes the wrong interpretation in OpenSCAD so far I can see it..

**workaround**
.. is possible but I'm not sure yet how exactly. I guess encapsulating bigger rings into group and then scale the group down to wished size .. or something like that..)

**attachment**
[ring-test.svg.zip](https://github.com/openscad/openscad/files/3162190/ring-test.svg.zip)

t-paul, 2019-05-09:
I don't have OpenSCAD at hand right now, but one observation is that both are not-closed arcs with < 360 degrees. I'm not sure if that's intended but using visual tricks will always be a corner case with SVG import. The save way should be to create two circles (with the "non-arc" mode selected) and difference the two. That should ensure it's a closed path with defined outline.

I'll have a look later what's going on with those specific arcs. It might be a rounding thing that they are behaving differently.

akust, 2019-05-09:
indeed! thank you for the hint. This was one of the many triels to find a workaround before I desided to report.
I closed the rings - now everything is completely different again, none of the example rings are extruded correctly anymore. hm..

akust, 2019-05-09:
..and trying your suggestion: yes, that works too. It reminds me on the function "convert countour to path" what works too in the case of circle. Is that result the same as subtraction? If yes, why doesn't it work in the case of my original import (I called 'object to path' and 'contour to path'). It's something different then the circle but it is the initial problem what caused me to start with redesigning the rings.. I don't know what the problem is..: [ring test path.svg.zip](https://github.com/openscad/openscad/files/3162651/ring.test.path.svg.zip)

(anyway: I like the idea that OpenSCAD supports the 'native' instruction)

henwist, 2019-05-31:
Seems to have to do with if paths are closed or not (see attached svg). If paths are closed, the object is fillled and if paths are open, the object is hollow. The big polygon is Inkscape's native polygon and cannot be split into nodes and therefor seems to be closed and not modifiable (to show difference in OpenSCAD).
[closed-path.zip](https://github.com/openscad/openscad/files/3240643/closed-path.zip)

henwist, 2019-06-01:
More on this topic: found that src/import_svg.cc does some strange things when importing two triangles - one that has a closed path and the other that is open. Try the attached svg and linear_extrude it. When debugging specific line in function: 
_Polygon2d *import_svg(const std::string &filename, const double dpi, const bool center, const Location &loc)_ 
line 152 in current master:  _outline.vertices.push_back(Vector2d(x, y));_
the output from gdb for x, y for the two imported triangles is:

**tri1:**
x:-18, y:-9
x:-5, y:-9
x:-13, y:-4
x:-18, y:-9

**tri2:**
x:17, y:5
x:16, y:4
x:3, y:4
x:3, y:6
x:13, y:6
x:8, y:9
x:5, y:6
x:4, y:8
x:8, y:11
x:17, y:5

[closed-path-triangle.zip](https://github.com/openscad/openscad/files/3244175/closed-path-triangle.zip)

nophead, 2019-06-01:
Surely and open path makes no sense in OpenSCAD. All 2D outlines must be closed.

t-paul, 2019-06-01:
Closed path are imported as solid 2d shapes, open polygons are converted to closed 2d shapes using the stroke-width.

nophead, 2019-06-01:
So does that mean stroke width is ignored for closed paths but used for open paths? Sort of makes sense but not very intuitive behaviour. Is it documented anywhere?

henwist, 2019-06-02:
Back to the two rings...  I have found that one ring is doing offset and the other is not, This is decided at line src/libsvg/path.cc:513

> 511             if (!path_closed && !path_list.empty()) {
512                     path_t path = path_list.back();
513                     if (is_open_path(path)) {
514                             path_list.pop_back();
515                             offset_path(path_list, path, get_stroke_width(), get_stroke_linecap());
516                     }
517             }

If you do a tbreak at line 514 and jump there, the two rings will draw as intended.

t-paul, 2019-06-02:
There's still a number of things that need to be updated in the documentation now that 2019.05 is out. I do like to have examples included so preparing that takes quite some time. The still quite incomplete page for the SVG import is: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/SVG_Import - I just added the open/closed.

t-paul, 2019-06-02:
Argh, and just noticed the error in the image that should say "Stroke only" for the bottom row...

t-paul, 2019-11-14:
Documentation is updated to explain the behavior, so I think we can close this. If there's anything I'm missing, please describe the remaining issue and reopen.

closed 2019-11-14 by t-paul

3Dietrich, 2019-11-14:
Your [Documentation](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/SVG_Import 
) describes the difference very well. Actually the wiki is one of the great parts of OpenSCAD.
However, there is an obstacle in this topic. My usage (and help tip for every new user) is:
- open [cheatsheet](http://www.openscad.org/cheatsheet/index.html) (have it as bookmark)
- find your what you're looking for, open it --> Wikibook

so easy, soo good.
But this way the 2D and 3D **import("….ext")** leads to a [different page](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Importing_Geometry#import). I would appreciate if the cheatsheet leads to "SVG_Import" as well (e.g. import it or just set a link in "Importing_Geometry"). 

Thank you!

thehans, 2019-11-14:
2D import also supports DXF, so having it go directly to SVG page would not be great.  I went ahead and made the SVG bullet point on the main import description into a link to the SVG page.  Does that help @3Dietrich ?

3Dietrich, 2019-11-14:
Hi Hans,

> I went ahead and made the SVG bullet point on the main import description into a link to the SVG page. Does that help @3Dietrich ?

Yes. But since you ask me.. (on mac, Chrome & Safari) the Link has no underscore and the color is close to black.. hard to see that it's a link. I found it only because I searched it. So if you can and want to change the formatting I'd suggest to set it more classic ((allways) blue-ish and underscore, like in the cheatsheet).
But it's cosmetic. 

here is a screenshot:
[image]