openscad/openscad #2950: echo() formating error

RonaldoCMP, 2019-05-19:
I have found this:

echo(1.00006e-20); //ECHO: 1.00006e-2
echo(1.00000e-20);//ECHO: 1e-20

using:

Windows 7/64
OpenSCAD 2019.01-RC4

t-paul, 2019-05-19:
Confirmed on Debian/Testing. Same output.

thehans, 2019-05-19:
Ah you found a bug in my "trim trailing zeroes" code.  I just made a quick patch for it.  The code is a little bit weird because for efficiency, I originally tried to loop through the output string no more than once.  Now with the added check for exponent, it has to scan the string in sort of redundant way now.  It could probably be written better, but I don't have time at the moment to work out another method.

I still wish the main double-conversion library supported this but the maintainer didn't think it was important.
https://github.com/google/double-conversion/issues/87

kintel, 2019-06-23:
Fixed in #2957

closed 2019-06-23 by kintel