openscad/openscad #2952: Invalid Warning: argument [arg_name] supplied more then once

nickjbauer, 2019-05-22:
Version: 2019.01-RC4 

Getting an invalid warning like this:
WARNING: argument [arg_name] supplied more then once, in file [file_name].scad, line [123]

Exact scenario:
module1 has named argument "x".
module2 also has named argument "x".
module1 calls module2, passing: x=x
```
module module1(x=1) {
// do stuff
  module1(x=x);
// do more stuff
};

module module2(x=2) {
// do stuff
};
```

I suspect this might be any variable named the same as the named arg.

nickjbauer, 2019-05-22:
I'd like to add that when this is called in a loop, it blasts the console logs.

MichaelAtOz, 2019-05-22:
I presume you mean his;
```module module1(x=1) {
// do stuff
  module2(x=x);
// do more stuff
};

module module2(x=2) {
// do stuff
};
```
I blew away all the RC's after the 2019.05 release.
The above does not log anything with the release, can you confirm, as there was a bit of fine tuning towards the end.

MichaelAtOz, 2019-05-22:
edited code snip

nickjbauer, 2019-05-22:
I appreciate your speculation, but I did put what i meant.. Your code could also have be affected...
Just installed 2019.05... unfortunately, the issue is still present.

MichaelAtOz, 2019-05-22:
You said "module1 calls module2, passing: x=x", your code snippet has module1 recursively calling module1.

What operating system?

MichaelAtOz, 2019-05-22:
Could you check for a typo, `module2(x,x)` would cause that.

nickjbauer, 2019-05-22:
Sheesh.  yup, you are right.  updated original post.  I think i was thrown by the email vs your update.   Admittedly, i don't regularly contribute to github.

MichaelAtOz, 2019-05-22:
Could you copy/paste the exact line of code `in file [file_name].scad, line [123]` please.
Also what operating system are you on?

t-paul, 2019-05-22:
I'm not getting any warning with the snippet either (Release version 2019.05  on Debian/Testing). Please update with complete code snippet that shows the issue.

nickjbauer, 2019-05-22:
OS: MacOs Mojave 10.14.5 (18F132)
Here is an extracted snippet from my larger project that exhibits the issue.  Sorry for the false simplified version.

```
module make_hole(start_layer=0, end_layer=0, xy=[0, 0], hole_slop=1.0) {
    height_multiplier = end_layer-start_layer+1;
    translate(xy)    
    cylinder(d=1.0 * hole_slop, h=2.0*height_multiplier);    
}

MIN_LAYER_HEIGHT = 0.1;
module make_slot(start_layer=0, end_layer=0, xy=[0, 0], slots=[], hole_slop=1.0) {
    height = 2.0;
    height_multiplier = end_layer-start_layer+1;
    wall = 2 * 0.4;
    width= 0.5;
    outer_diameter = 3.0;
    inner_diameter = 1.0;
    translate(xy){
        difference() {
            cylinder(d=outer_diameter, h=height*height_multiplier);
            cylinder(d=inner_diameter + 2 * width + wall, h=height*height_multiplier);   
        }
        difference() {
            cylinder(d=outer_diameter, h=height*height_multiplier);
            make_hole(xy, start_layer=start_layer, end_layer=end_layer, hole_slop=hole_slop);
            line_width = 0.2;
            r2 = (inner_diameter * hole_slop)/2;
            if(start_layer == 0)
            cylinder(r1=r2+2*line_width, r2=r2, h=MIN_LAYER_HEIGHT*6);    

            
            if(len(slots) > 0)
            for(i = [0:len(slots)-1]) {
                rotate([0, 0, slots[i]]) {
                    translate([-width/2, 0, height*height_multiplier-width*1.1])
                    cube([width, outer_diameter/2, width*1.1]);

                    rotate([0, 0, 90])
                    translate([inner_diameter/2, 0, 0])
                    cylinder(d=width/2, h=height*height_multiplier);
                }
            }       
        }
    }
}

make_slot();
```

nophead, 2019-05-22:
The problem is in you code. It needs to be:
```
            make_hole(xy = xy, start_layer=start_layer, end_layer=end_layer, hole_slop=hole_slop);
```

If pass xy unnamed it become the first parameter, which is start_layer.

MichaelAtOz, 2019-05-22:
> 
> I'd like to add that when this is called in a loop, it blasts the console logs.

Also note that Preferences/Advanced has some options for warning/error, including turning this specific one off, and stopping on first warning.

nickjbauer, 2019-05-23:
nophead is correct.  A refactoring error left xy positional instead of named.  Sorry for the false report.

closed 2019-05-23 by nickjbauer