openscad/openscad #2953: Preview performance regression on Linux

ali1234, 2019-05-24:
Somewhere around February the performance on the preview window fell dramatically. With moderately complicated objects the framerate when rotating drops to about 0.5 fps. Prior to the regression it would be about 15 fps. Due to a large number of commits that fail to build, I was only able to narrow it down to 15 commits with bisect.

Library information from the last good version 27ce666 is attached. The library info for the first bad version c02dfb9 is identical.

[libs-good.txt](https://github.com/openscad/openscad/files/3214647/libs-good.txt)

```
# only skipped commits left to test
# possible first bad commit: [c02dfb9c128396a7c03dc48fc08485ffdb196818] get CMakeLists.txt working on Linux again after changes for Windows.
# possible first bad commit: [d57a1d09248d79706f59eed1d4588320d5721bc4] Merge branch 'master' of http://github.com/openscad/openscad into msvc_build
# possible first bad commit: [997928e0fdf4eac579fc387dce096ee3250dc98a] Successfully set stack size, and fix utf8 string literal.
# possible first bad commit: [97250402325c747250cb6b896dd9d48a967bcdfd] Merge branch 'master' of https://github.com/openscad/openscad into msvc_build
# possible first bad commit: [65f1a2e15708334369806fa17540b9acc994c2a3] revert bison.pri change
# possible first bad commit: [97917551327fb4eb8936cca6590718e5b01a93ab] revert to parser name command line setting
# possible first bad commit: [a9c59b3c77db424693c6d0a69d8a9342cdec5aea] Build openscad.exe for windows subsystem (GUI, not console).
# possible first bad commit: [4955b37793cfee63c989eb9f1e95d950b6640e3a] Make eigen std::vector alignment conditional on MSVC
# possible first bad commit: [a3d89d17ba8aef7317bdfb7bb61c8082f63014a3] Remove deprecated name-prefix in favor of api.prefix
# possible first bad commit: [f3c5cdd78a3c5114bcd0606f79550b9d42f90bf7] Reduce excess warnings, mostly type coercion
# possible first bad commit: [04f32befab5c6356658ad8c8a8955f927b381b61] Fixes for fixed-size vectorizable Eigen objects, alignment issues. https://eigen.tuxfamily.org/dox/group__DenseMatrixManipulation__Alignement.html
# possible first bad commit: [172faa36c445ca62a72108bf12ccc0bdff58c23e] Build fixes for MSVC
# possible first bad commit: [2bed5189ddeb1fccc7c6c709ac6c881d9d38d639] Removed -DNULLGL option, replaced obsolete QString::sprintf and fixed other compile errors
# possible first bad commit: [6d8da4f1c60bceb842e0abf823ea04f2ce3b6abb] Found gettext aka libintl
# possible first bad commit: [ea8ed6b28ef835d5456559b5f8d88c0e84fb4d96] MSVC build testing, WIP
```

ali1234, 2019-05-24:
This also seems to affect headless rendering of the final objects, with c02dfb9 taking over twice as long as 27ce666 to render the same scenes - 3m33s vs 1m24s.

t-paul, 2019-05-24:
Please add at least some kind of code example to reproduce the issue.

ali1234, 2019-05-24:
```
for (i = [-5:5]) for (j = [-5:5]) {
    translate([i*2, j*2, 0]) sphere(1, $fn=32);
}
```

```
time /home/al/Source/openscad/bisect/openscad-bad/openscad -o test.stl test.scad
    Real: 0:10:46, User: 0:10:45, Sys: 0:00:00, CPU: 99%, Memory: 2,314.39MB
time /home/al/Source/openscad/bisect/openscad-good/openscad -o test.stl test.scad
    Real: 0:04:48, User: 0:04:47, Sys: 0:00:00, CPU: 99%, Memory: 2,308.23MB
```

t-paul, 2019-05-24:
Thanks, I'll try to reproduce with the referenced commits. There were some changes trying to optimize performance using an external library. Maybe that caused regressions in other scenarios...

t-paul, 2019-05-24:
I can't reproduce this. The times are pretty much the same:
```
openscad-27ce666e
real 5m24,886s, user 5m23,861s, sys 0m0,940s
real 5m2,163s, user 5m1,205s, sys 0m0,952s

openscad-92550dec (current master)
real 5m13,735s, user 5m12,248s, sys 0m0,972s
real 5m11,368s, user 5m10,016s, sys 0m0,940s
```
FWIW using flags `qmake CONFIG+=experimental CONFIG-=debug`

ali1234, 2019-05-24:
I can only reproduce with cmake builds. qmake builds run at almost exactly the same speed as cmake build before the regression, and continue to run at the same speed after it.

For the record I am not using any extra configuration flags with either build tool.

t-paul, 2019-05-24:
Added a note to #2518 tracking open issues with the cmake build (which is not officially supported yet).

closed 2019-05-24 by t-paul

ali1234, 2019-05-24:
This was all down to CONFIG+=debug as you predicted.

Specifically qmake builds behaviour does not change before/after the regression. However, `qmake`, `qmake CONFIG-=debug` and `qmake CONFIG+=debug` all behave differently:
* `qmake CONFIG-=debug` does not print CGAL debugging and runs fast.
* `qmake` prints CGAL debugging and runs fast.
* `qmake CONFIG+=debug` prints CGAL debugging and runs slow.

When building with cmake with no config flags it never prints CGAL debugging. It runs fast before the regression, and slow after it. Note that printing CGAL and running slow is a combination that can't be reproduced by manipulating the debug flag in the qmake build.

So it seems like something got changed around that affected exactly which debugging bits are turned on and off in the cmake build.