openscad/openscad #2956: Running GCAL blocks the UI thread

lukehutch, 2019-05-30:
When GCAL is being run, the UI thread is blocked. If GCAL hangs, there's no way to still save the open project in OpenSCAD. GCAL should be run asynchronously in a different thread.

lukehutch, 2019-05-30:
An example of a project that hangs OpenSCAD:

[example.zip](https://github.com/openscad/openscad/files/3235033/example.zip)

t-paul, 2019-05-30:
Render (F6) is running in a separate thread, if you mean in Preview (F5) that is covered by #1195 already.

closed 2019-05-30 by t-paul

lukehutch, 2019-05-30:
@t-paul OK, but please check the project I attached anyway, since it is a lock-up bug, and it may indicate a separate problem.

t-paul, 2019-05-30:
Lock-up where and how? Looking at the example it has multiple `minkowski()` with `$fn = 80` spheres which explodes the number of vertices to calculate. Using `$fa = 5; $fs = 0.5;` instead of `$fn = 80` renders the model in 15 seconds on my system.

lukehutch, 2019-05-30:
Sorry, you're right, `$fn` should have been overridden for `minkowski()`. I wasn't patient enough. I guess this is just an example of #1195.

t-paul, 2019-05-30:
No problem. Maybe we can find a way to warn about some of those specific cases. Even with #1195 implemented it still takes very long to compute. `$fn = 80` just finished after 10 minutes 43 seconds.