openscad/openscad #2958: localization in parameters panel

JSkrat, 2019-05-31:
Can I please have dot as decimal point to be allowed in parameters panel instead of or along with comma despite openscad picked russian language? 
Not-english decimal point is useless as code in english with dot as decimal point.

Btw, disabling localization does not affect that, delimiters are still localized(

t-paul, 2019-05-31:
Can you please clarify if you see a bug that makes the Customizer parameter non-functional or if this is more about fixing the display to be the same in both places.

Using this example...
[image]
I see the display effect you describe, but it does generate the correct values for the model generation, e.g. while the Customizer GUI shows `3,4`, it's actually using `3.4` value for the cube.

t-paul, 2019-11-14:
Closing, no feedback.

closed 2019-11-14 by t-paul

JSkrat, 2019-11-16:
sorry for long reply. 
It is about consistency. I always should remember that this field requires comma, despite everywhere else (in code) i'm using dot. I can get used to it, but why?

t-paul, 2019-11-16:
I still don't get what you mean, sorry.
Setting the locale to english, the customizer shows `3.4` in the customizer window and uses value 3.4. When setting the language to German (which has ',' as decimal separator), the customizer shows `3,4` and the value used is 3.4.

So are you saying you want to set the language to russian (which I suppose also uses `,` as decimal separator) but still want the GUI control to use 3.4?

JSkrat, 2019-11-16:
at the computer with russian system locale customizer shows and (the most importantly) accepts only comma, despite the localization settings of the openscad itself. Dot even can not be entered.

Also, while trying i've found that if initial value of the parameter is round (tried even 25.0), you can not enter part after the comma/dot, field does not accept anything after comma. Not sure if it is bug or not

JSkrat, 2019-11-16:
and yes, personally i hate that different locales has same digits, but different decimal separator, so for me it would be much more convenient if the field for every locale will accept dot too :)

JSkrat, 2019-11-30:
@t-paul can you reopen please? 9_9

t-paul, 2019-11-30:
You are asking for a change in Qt, so reopening here would not help.