openscad/openscad #2962: Bug in Find feature of OpenSCAD

RomitKumar, 2019-06-04:
The count of matched words in Find option displays wrong value. Steps to reproduce the error:
1. Launch Openscad and clear the contents of editor(if any) 
2. Press Ctrl+f to view find panel and clear the contents of text field in find panel 
3. Press 'a' 9 times in the editor
4. In text field of search panel, write 'a'(count of matched words is 9)
5. In editor write on more 'a' (now count of 'a' is 10)
6. In text field of search panel, append another 'a' such that 'aa' is being searched for(count of matched words is 5 now)
7. In text field of search panel press a backspace, so 'a' is again being searched for
However the count of matched words shown is 1

henwist, 2019-06-04:
I can confirm the bug - but the commit 92550dece3b0d15edae2982f80a0b344dc7d0640 (just a few days ago) in master don't count to ten when another 'a' is added in the editor.

RomitKumar, 2019-06-04:
The find in OpenSCAD does not do real time finding. So the count of matched word does not update automatically. What I meant by step 5 was that there are 10 'a' in editor. Updated the issue to make it clear.

kintel, 2019-06-23:
Fixed by #2974

closed 2019-06-23 by kintel