openscad/openscad #2963: Q: Implementation of OpenSCAD

scheming-pony, 2019-06-05:
I posted this comment on the Libfive Github issue page, would someone here care to comment on its accuracy or other details?

> OpenSCAD uses CGAL as a geometry (meshing) engine with B-rep modeling. Aside from the complexity of doing CSG operations on boundary triangles, it uses exact arithmetic ("exact computation paradigm") and can be quite slow with the tradeoff of increased accuracy and robustness. Libfive uses F-rep modeling, which by itself is probably faster, and the mesh is generated with IEEE 754 floating point arithmetic--which is again faster, but can produce degeneracies which have to be addressed (by the library maintainer).
>
> [Edit]: It does look like OpenSCAD uses OpenCSG to do some of the CSG calculation, for display I assume--it would be interesting to know how the mesh rendering work is partitioned between CGAL and OpenCSG.

kintel, 2019-06-05:
Looks accurate. We only use OpenCSG for display, not geometry evaluation.

closed 2019-06-05 by kintel

MichaelAtOz, 2019-06-05:
> Aside from the complexity of doing CSG operations on boundary triangles, it uses exact arithmetic ("exact computation paradigm") and can be quite slow with the tradeoff of increased accuracy and robustness

That applies to CGAL.
OpenSCAD also uses other methods ([Clipper](https://sourceforge.net/projects/polyclipping/)) for geometry evaluation based on floating point, which can produce degenerations.