openscad/openscad #2964: ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 329

l29ah, 2019-06-06:
`OpenSCAD version 2019.06.03`

```
fan_d = 135;
flange_thickness = 2.5;
flange_offset = 6;

thickness = 1;

grille_hole_width = 1.4;
grille_spoke_width = 0.6;

linear_extrude(thickness) {
	intersection() {
		union() {
			for (x = [-fan_d / 2 : grille_hole_width + grille_spoke_width : fan_d / 2]) {
				translate([x, 0]) square([grille_spoke_width, fan_d], center = true);
			}
			for (y = [-fan_d / 2 : grille_hole_width + grille_spoke_width : fan_d / 2]) {
				translate([0, y]) square([fan_d, grille_spoke_width], center = true);
			}
		}
		circle(d = fan_d - flange_offset * 2);
	}
	difference() {
		circle(d = fan_d + thickness * 2);
		circle(d = fan_d - flange_offset * 2);
	}
}
translate([0, 0, thickness]) {
	linear_extrude(flange_thickness + thickness) {
		difference() {
			circle(d = fan_d + thickness * 2);
			circle(d = fan_d);
		}
	}
}
```

nophead, 2019-06-06:
Change line 24 to 
```
	difference() {
		circle(d = fan_d + thickness * 2);
		circle(d = fan_d - flange_offset * 2 - 1/128);
	}
```

Fixes the problem by making the ring slightly overlap the mesh. Without it you get vertices that are very nearly coincident but not quite due to floating point representation.

l29ah, 2019-06-06:
Thank you.
It's interesting that it works fine if i remove the whole `translate([0, 0, thickness])` part that is irrelevant to the problem you've outlined.

nophead, 2019-06-06:
Yes that is because CGAL is not involved at all until you do the 3D union. The grid is calculated in 2D, which uses the clipper library. Linear extrude converts it to a polyhedron and if you don't do an operations on the that the result looks fine. If you union it with just a unit cube then CGAL is used to compute the union and that doesn't like the mesh produced by linear extrude.

kintel, 2019-06-23:
Closing as resolved as well as duplicate of #791

closed 2019-06-23 by kintel