openscad/openscad #2965: OpenSCAD crashes on invalid (empty) STL or PNG import

hroncok, 2019-06-07:
```
$ touch empty.png
$ touch empty.stl
$ echo 'import("empty.stl");' > test.scad
$ openscad --version
OpenSCAD version 2019.05
$ openscad test.scad  # [F6]
/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = Eigen::Matrix<float, 3, 1>; _Alloc = std::allocator<Eigen::Matrix<float, 3, 1> >; std::vector<_Tp, _Alloc>::reference = Eigen::Matrix<float, 3, 1>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Aborted (core dumped)
[tmp]$ echo 'surface("empty.png");' > test.scad
[tmp]$ openscad test.scad   # [F6]
/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = Eigen::Matrix<float, 3, 1>; _Alloc = std::allocator<Eigen::Matrix<float, 3, 1> >; std::vector<_Tp, _Alloc>::reference = Eigen::Matrix<float, 3, 1>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Aborted (core dumped)
```

More info incl. backtrace in the downstream issue: https://bugzilla.redhat.com/show_bug.cgi?id=1717625

This is Fedora 30, x86_64.
OpenSCAD was built with Eigen 3.3.7.

```
$ rpm -q --requires openscad | grep ^lib
libCGAL.so.13()(64bit)
libGL.so.1()(64bit)
libGLEW.so.2.1()(64bit)
libGLU.so.1()(64bit)
libQt5Core.so.5()(64bit)
libQt5Core.so.5(Qt_5)(64bit)
libQt5Core.so.5(Qt_5.12)(64bit)
libQt5DBus.so.5()(64bit)
libQt5DBus.so.5(Qt_5)(64bit)
libQt5Gamepad.so.5()(64bit)
libQt5Gamepad.so.5(Qt_5)(64bit)
libQt5Gui.so.5()(64bit)
libQt5Gui.so.5(Qt_5)(64bit)
libQt5Multimedia.so.5()(64bit)
libQt5Multimedia.so.5(Qt_5)(64bit)
libQt5Network.so.5()(64bit)
libQt5Network.so.5(Qt_5)(64bit)
libQt5Widgets.so.5()(64bit)
libQt5Widgets.so.5(Qt_5)(64bit)
libX11.so.6()(64bit)
libboost_filesystem.so.1.69.0()(64bit)
libboost_program_options.so.1.69.0()(64bit)
libboost_regex.so.1.69.0()(64bit)
libc.so.6()(64bit)
libc.so.6(GLIBC_2.11)(64bit)
libc.so.6(GLIBC_2.14)(64bit)
libc.so.6(GLIBC_2.2.5)(64bit)
libc.so.6(GLIBC_2.3.4)(64bit)
libc.so.6(GLIBC_2.4)(64bit)
libdouble-conversion.so.1()(64bit)
libfontconfig.so.1()(64bit)
libfreetype.so.6()(64bit)
libgcc_s.so.1()(64bit)
libgcc_s.so.1(GCC_3.0)(64bit)
libglib-2.0.so.0()(64bit)
libgmp.so.10()(64bit)
libharfbuzz.so.0()(64bit)
libm.so.6()(64bit)
libm.so.6(GLIBC_2.2.5)(64bit)
libm.so.6(GLIBC_2.29)(64bit)
libmpfr.so.4()(64bit)
libopencsg.so.1()(64bit)
libpolyclipping.so.22()(64bit)
libpthread.so.0()(64bit)
libpthread.so.0(GLIBC_2.2.5)(64bit)
libqscintilla2_qt5.so.15()(64bit)
libstdc++.so.6()(64bit)
libstdc++.so.6(CXXABI_1.3)(64bit)
libstdc++.so.6(CXXABI_1.3.3)(64bit)
libstdc++.so.6(CXXABI_1.3.5)(64bit)
libstdc++.so.6(CXXABI_1.3.7)(64bit)
libstdc++.so.6(GLIBCXX_3.4)(64bit)
libstdc++.so.6(GLIBCXX_3.4.11)(64bit)
libstdc++.so.6(GLIBCXX_3.4.14)(64bit)
libstdc++.so.6(GLIBCXX_3.4.15)(64bit)
libstdc++.so.6(GLIBCXX_3.4.18)(64bit)
libstdc++.so.6(GLIBCXX_3.4.19)(64bit)
libstdc++.so.6(GLIBCXX_3.4.20)(64bit)
libstdc++.so.6(GLIBCXX_3.4.21)(64bit)
libstdc++.so.6(GLIBCXX_3.4.22)(64bit)
libstdc++.so.6(GLIBCXX_3.4.26)(64bit)
libstdc++.so.6(GLIBCXX_3.4.5)(64bit)
libstdc++.so.6(GLIBCXX_3.4.9)(64bit)
libxml2.so.2()(64bit)
libxml2.so.2(LIBXML2_2.4.30)(64bit)
libxml2.so.2(LIBXML2_2.5.0)(64bit)
libxml2.so.2(LIBXML2_2.6.0)(64bit)
libzip.so.5()(64bit)
```

henwist, 2019-06-07:
This bug cannot be reproduced on Debian at least. 

> OpenSCAD Version: 2019.05.30
System information: Linux 4.19.0-4-amd64 #1 SMP Debian 4.19.28-2 (2019-03-15) x86_64 Debian GNU/Linux 10 (buster) 8 CPUs 11.66 GB RAM

OpenSCAD linked to libraries:
lib3MF.so.1
/lib64/ld-linux-x86-64.so.2
libasyncns.so.0
libboost_atomic.so.1.67.0
libboost_filesystem.so.1.67.0
libboost_program_options.so.1.67.0
libboost_regex.so.1.67.0
libboost_system.so.1.67.0
libboost_thread.so.1.67.0
libbsd.so.0
libbz2.so.1.0
libcap.so.2
libCGAL.so.13
libcrypto.so.1.1
libc.so.6
libdbus-1.so.3
libdl.so.2
libdouble-conversion.so.1
libexpat.so.1
libFLAC.so.8
libfontconfig.so.1
libfreetype.so.6
libgcc_s.so.1
libgcrypt.so.20
libGLdispatch.so.0
libGLEW.so.2.1
libglib-2.0.so.0
libGL.so.1
libGLU.so.1
libGLX.so.0
libgmp.so.10
libgpg-error.so.0
libgraphite2.so.3
libharfbuzz.so.0
libICE.so.6
libicudata.so.63
libicui18n.so.63
libicuuc.so.63
liblz4.so.1
liblzma.so.5
libmpfr.so.6
libm.so.6
libnsl.so.1
libogg.so.0
libopencsg.so.1
libpcre2-16.so.0
libpcre.so.3
libpng16.so.16
libpthread.so.0
libpulsecommon-12.2.so
libpulse.so.0
libqscintilla2_qt5.so.13
libQt5Concurrent.so.5
libQt5Core.so.5
libQt5DBus.so.5
libQt5Gamepad.so.5
libQt5Gui.so.5
libQt5Multimedia.so.5
libQt5Network.so.5
libQt5PrintSupport.so.5
libQt5Widgets.so.5
libresolv.so.2
librt.so.1
libSM.so.6
libsndfile.so.1
libspnav.so.0
libssl.so.1.1
libstdc++.so.6
libsystemd.so.0
libuuid.so.1
libvorbisenc.so.2
libvorbis.so.0
libwrap.so.0
libX11.so.6
libX11-xcb.so.1
libXau.so.6
libxcb.so.1
libXdmcp.so.6
libXext.so.6
libXi.so.6
libxml2.so.2
libXtst.so.6
libzip.so.4
libz.so.1
linux-vdso.so.1

t-paul, 2019-06-07:
I can't reproduce this either. Maybe related to gcc-9? Debian currently has `gcc version 8.3.0 (Debian 8.3.0-6)`.

Also: `You are not authorized to access bug #1717625`

hroncok, 2019-06-07:
Oh, the reporter has checked the magical "private" checkbox. I've asked whether I can make it public.

knielsen, 2019-06-09:
Was able to reproduce in a fedora 30 systemd-nspawn container.

Stack trace:

Core was generated by `openscad test.scad'.
Program terminated with signal SIGABRT, Aborted.
#0  0x00007f70fb860eb5 in raise () from /lib64/libc.so.6
[Current thread is 1 (Thread 0x7f70f9304900 (LWP 281))]
Missing separate debuginfos, use: dnf debuginfo-install CGAL-4.14-1.fc30.x86_64 boost-filesystem-1.69.0-8.fc30.x86_64 boost-program-options-1.69.0-8.fc30.x86_64 boost-regex-1.69.0-8.fc30.x86_64 boost-system-1.69.0-8.fc30.x86_64 bzip2-libs-1.0.6-29.fc30.x86_64 double-conversion-3.0.0-4.fc30.x86_64 elfutils-libelf-0.176-2.fc30.x86_64 expat-2.2.6-2.fc30.x86_64 fontconfig-2.13.1-6.fc30.x86_64 freetype-2.9.1-7.fc30.x86_64 glib2-2.60.3-1.fc30.x86_64 glibc-2.29-12.fc30.x86_64 gmp-6.1.2-10.fc30.x86_64 graphite2-1.3.10-7.fc30.x86_64 harfbuzz-2.3.1-1.fc30.x86_64 jbigkit-libs-2.1-16.fc30.x86_64 lcms2-2.9-5.fc30.x86_64 libGLEW-2.1.0-4.fc30.x86_64 libICE-1.0.9-15.fc30.x86_64 libSM-1.2.3-2.fc30.x86_64 libX11-1.6.7-1.fc30.x86_64 libX11-xcb-1.6.7-1.fc30.x86_64 libXcursor-1.1.15-5.fc30.x86_64 libXdamage-1.1.4-16.fc30.x86_64 libXext-1.3.3-11.fc30.x86_64 libXfixes-5.0.3-9.fc30.x86_64 libXinerama-1.1.4-3.fc30.x86_64 libXrandr-1.5.1-9.fc30.x86_64 libXrender-0.9.10-9.fc30.x86_64 libXxf86vm-1.1.4-11.fc30.x86_64 libdrm-2.4.98-1.fc30.x86_64 libedit-3.1-27.20190324cvs.fc30.x86_64 libffi-3.1-19.fc30.x86_64 libgcc-9.1.1-1.fc30.x86_64 libicu-63.2-2.fc30.x86_64 libjpeg-turbo-2.0.2-1.fc30.x86_64 libpng-1.6.36-1.fc30.x86_64 libselinux-2.9-1.fc30.x86_64 libtiff-4.0.10-4.fc30.x86_64 libuuid-2.33.2-1.fc30.x86_64 libxml2-2.9.9-2.fc30.x86_64 libxshmfence-1.3-4.fc30.x86_64 libzip-1.5.2-1.fc30.x86_64 llvm-libs-8.0.0-6.fc30.x86_64 mesa-dri-drivers-19.0.5-2.fc30.x86_64 mesa-libGL-19.0.5-2.fc30.x86_64 mesa-libGLU-9.0.0-17.fc30.x86_64 mesa-libglapi-19.0.5-2.fc30.x86_64 mpfr-3.1.6-4.fc30.x86_64 openssl-libs-1.1.1c-2.fc30.x86_64 pcre-8.43-2.fc30.x86_64 pcre2-10.33-2.fc30.x86_64 polyclipping-6.4.2-6.fc30.x86_64 qscintilla-2.11.1-1.fc30.x86_64 qt-4.8.7-49.fc30.x86_64 xz-libs-5.2.4-5.fc30.x86_64 zlib-1.2.11-15.fc30.x86_64
(gdb) bt
#0  0x00007f70fb860eb5 in raise () from /lib64/libc.so.6
#1  0x00007f70fb84b895 in abort () from /lib64/libc.so.6
#2  0x000055c62de6cf18 in std::__replacement_assert (__file=__file@entry=0x55c62e270608 "/usr/include/c++/9/bits/stl_vector.h", __line=__line@entry=1042, 
    __function=__function@entry=0x55c62e27c2d0 "std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = Eigen::Matrix<float, 3, 1>; _Alloc = std::allocator<Eigen::Matrix<float, 3, 1>"..., 
    __condition=__condition@entry=0x55c62e2703e0 "__builtin_expect(__n < this->size(), true)") at /usr/include/c++/9/x86_64-redhat-linux/bits/c++config.h:2533
#3  0x000055c62def572d in std::vector<Eigen::Matrix<float, 3, 1, 0, 3, 1>, std::allocator<Eigen::Matrix<float, 3, 1, 0, 3, 1> > >::operator[] (__n=0, this=0x7ffd0232ce88) at /usr/include/c++/9/bits/stl_iterator.h:806
#4  Reindexer<Eigen::Matrix<float, 3, 1, 0, 3, 1> >::getArray (this=<optimized out>) at src/Reindexer.h:49
#5  PolysetUtils::tessellate_faces (inps=..., outps=...) at src/polyset-utils.cc:81
#6  0x000055c62e1f11aa in CGALRenderer::CGALRenderer (this=0x55c63010ea10, geom=...) at /usr/include/c++/9/bits/shared_ptr_base.h:1005
#7  0x000055c62e03b408 in MainWindow::actionRenderDone (this=<optimized out>, root_geom=std::shared_ptr<const Geometry> (use count 6, weak count 0) = {...}) at /usr/include/c++/9/ext/atomicity.h:96
#8  0x000055c62e2698ee in MainWindow::qt_static_metacall (_o=<optimized out>, _c=<optimized out>, _id=<optimized out>, _a=<optimized out>) at /usr/include/c++/9/ext/atomicity.h:96
#9  0x00007f70fc18739c in QObject::event(QEvent*) () from /lib64/libQtCore.so.4
#10 0x00007f70fc5d4a1e in QWidget::event(QEvent*) () from /lib64/libQtGui.so.4
#11 0x00007f70fc9972d0 in QMainWindow::event(QEvent*) () from /lib64/libQtGui.so.4
#12 0x00007f70fc57f3f1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib64/libQtGui.so.4
#13 0x00007f70fc585ed4 in QApplication::notify(QObject*, QEvent*) () from /lib64/libQtGui.so.4
#14 0x000055c62e05daf5 in OpenSCADApp::notify (this=<optimized out>, object=<optimized out>, event=<optimized out>) at src/OpenSCADApp.cc:55
#15 0x00007f70fc16e32f in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /lib64/libQtCore.so.4
#16 0x00007f70fc170f58 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /lib64/libQtCore.so.4
#17 0x00007f70fc19c18d in ?? () from /lib64/libQtCore.so.4
#18 0x00007f70fd39aedd in g_main_context_dispatch () from /lib64/libglib-2.0.so.0
#19 0x00007f70fd39b270 in ?? () from /lib64/libglib-2.0.so.0
#20 0x00007f70fd39b313 in g_main_context_iteration () from /lib64/libglib-2.0.so.0
#21 0x00007f70fc19c2e6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib64/libQtCore.so.4
#22 0x00007f70fc62217b in ?? () from /lib64/libQtGui.so.4
#23 0x00007f70fc16cb13 in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib64/libQtCore.so.4
#24 0x00007f70fc16ce26 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib64/libQtCore.so.4
#25 0x00007f70fc1722be in QCoreApplication::exec() () from /lib64/libQtCore.so.4
#26 0x000055c62e02275e in gui (inputFiles=..., original_path=..., argc=<optimized out>, argv=<optimized out>) at src/openscad.cc:771
#27 0x000055c62de685bd in main (argc=<optimized out>, argv=0x7ffd0232e2e8) at src/openscad.cc:1073

hroncok, 2019-06-09:
That bugzilla should now be public.

t-paul, 2019-06-09:
Yes, it is now. Thanks. Unrelated observation... it seems to be still built against Qt4. It's recommended to use Qt5. Latest Release 2019.05 will be the last one supporting both.

hroncok, 2019-06-09:
We have updated quite recently to qt5 build and I have reproduced it there. The user might have not updated yet.

t-paul, 2019-06-09:
Ahh, that's great. So no suprises when Qt4 support is removed.

knielsen, 2019-06-10:
A bit more details on why this triggers on Fedora 30:

Fedora 30 builds OpenSCAD with `-Wp,-D_GLIBCXX_ASSERTIONS`. This triggers the
following assertion from `/usr/include/c++/9/bits/stl_vector.h`:
```c
  reference
  operator[](size_type __n) _GLIBCXX_NOEXCEPT
  {
    __glibcxx_requires_subscript(__n);
    return *(this->_M_impl._M_start + __n);
  }

# define __glibcxx_requires_subscript(_N)       \
  __glibcxx_assert(__builtin_expect(_N < this->size(), true))

#if defined(_GLIBCXX_ASSERTIONS)
# define __glibcxx_assert(_Condition) __glibcxx_assert_impl(_Condition)
#else
# define __glibcxx_assert(_Condition)
#endif
```

Accessing `&my_vector[0]` on an empty `my_vector` triggers the assertion.

So the combination of `-D_GLIBCXX_ASSERTIONS` with a new enough C++ library
that has the assertion is what triggers the crash. Then the crash can also
be seen for example in the test cgalpngtest_import_stl-test.

t-paul, 2019-06-10:
I think we should fix the API too. It should be possible to return a const reference instead, so there's no poking around in empty vectors. But as initial fix, it still makes sense to just skip the whole function if the input is empty.

Does that mean the offical Fedora build is compiled without `CONFIG-=debug`?. That's not ideal as it causes a quite serious performance hit (see #2953). We have to find a way to untangle that at some point.

hroncok, 2019-06-10:
I haven't changed anything about the debug config. I can look into that as well.

t-paul, 2019-06-10:
Looks like we really need some better documentation for building releases. I'll create a ticket for that...

hroncok, 2019-06-10:
> Accessing `&my_vector[0]` on an empty `my_vector` triggers the assertion.

Assumption: Being build without `_GLIBCXX_ASSERTIONS` this would not crash, but instead it would provide false data. Correct?

t-paul, 2019-06-10:
Not sure if it creates an invalid pointer or not (which will not be used afterwards). I guess that depends on the implementation of vector, if that has at least some valid memory even if size is set to 0. But regardles of that it's still a bug that should be fixed.

knielsen, 2019-06-10:
hroncok: There is no false data provided. It is just returning a pointer value that is never used. So building without _GLIBCXX_ASSERTIONS will avoid the problem seen on Fedora 30.
(But as t-paul writes, better fix the code)

PR #2973 (2019-06-23): Have Reindexer return a const ref instead of a pointer into value array.
Should fix #2965
closed 2019-06-23 by kintel