openscad/openscad #2969: emacs scad-mode crashes on lines starting with #

adrianVmariano, 2019-06-11:
The contrib directory provides a mode for emacs, scad-mode.el.  This mode has lead to random mysterious crashes of emacs.  I managed to repeat the problem: if I open a new file and type "#a=(" then emacs crashes.  

The cause of the problem is that 

     (c-init-language-vars scad-mode)

is never called.   The documentation for c-basic-common-init indicates that you must call c-init-language-vars first.   The above line needs to be inserted above the line that reads 

     (c-basic-common-init 'scad-mode (or scad-indent-style "k&r"))

t-paul, 2019-06-15:
The contrib directory, as the name implies, is external contributed stuff which is not maintained in this repo but provided here for convenience.

So if someone who can actually test this change could provide a pull request that would be great.

adrianVmariano, 2019-06-15:
So I've noticed that I'm getting an error message from the line I added.  So I think it's not actually the solution to the problem, unfortunately.  It "fixed" the problem by breaking the mode.

henwist, 2019-06-18:
I have tested this and can confirm that emacs locked and put CPU on 100% when input "#a=(" in the editor's window.

> I am on  GNU Emacs 26.1 (build 2, x86_64-pc-linux-gnu, GTK+ Version 3.24.4) of 2019-02-03, modified by Debian buster/unstable

adrianVmariano, 2019-06-18:
I've done further testing and found that just typing "#(" causes the crash, and actually instead of '(' any one of []{}() will do the job.   But I have not managed to figure out the correct way to fix it.

PR #3424 (2020-08-30): Make scad-mode derive from c-mode
When extending directly from `prog-mode`, it seems there are some edge cases that cause font-lock to hang until you interrupt with `C-g`, as discussed in #2996 

Deriving from `c-mode` fixes the problem. By default, such derived modes call `c-mode-hook` during initialization, but it's not really desirable for scad-mode, so we disable this hook (unless the user overrides this by customizing `scad-mode-disable-c-mode-hook`).

Fixes #2969
commit db52888b (2020-08-30): Make scad-mode derive from c-mode

Fixes #2969

Ordinarily, such modes would also call c-mode-hook during
initialization, but we disable this hook (unless the user overrides this
by customizing scad-mode-disable-c-mode-hook).
closed 2020-08-30 by t-paul