openscad/openscad #2976: procedure entry point could not be located in the dynamic link library

E-Gwen, 2019-06-24:
Hello,

I tried to run OpenScad 2019.05 32 bit version, but I get this error message : 

"the procedure entry point Direct3DCreate9Ex could not be located in the d3d9.dll dynamic link library"

So I can't run OpenSCAD 2019.05
Such a pity !

I use Windows XP SP3 fully updated.
Notice that OpenSCAD 2015.03-2 32Bit works with that OS.

Could you please fix that issue ?

Best regards.
E-Gwen.

t-paul, 2019-06-24:
I'm not sure that is possible. We use an external build environment ([mxe.cc](https://mxe.cc/) which now has Qt5 version 5.13 and Qt dropped Windows XP support already with Qt 5.7.

I'll leave this open for a bit, maybe someone has an idea. Maybe also post on the forum to see if someone has suggestions.

t-paul, 2019-06-24:
Hmm, looks like MXE has Qt4 still. It might be an option trying to build against that (which will disable a small number of GUI features). However that will be impossible for any future releases as 2019.05 is definitely the last release which will support Qt4.

E-Gwen, 2019-06-24:
Thank for your answer.

Maybe it could help : 

http://rawpedia.rawtherapee.com/Windows

https://github.com/L1L1/cardpeek/blob/master/Makefile.win32

https://docs.libretro.com/development/retroarch/compilation/windowsXP/

Notice that LMMS 1.2.0 runs well with Windows XP although it is compiled with QT5.
You can have a look : 
https://github.com/LMMS/lmms/wiki/Compiling

Sorry I can do nothing more, I am not a developper.

See youi.
E-Gwen.

MichaelAtOz, 2019-11-25:
@t-paul @kintel 
Got another one of these on the Forum.
Are we calling XP unsupported??

t-paul, 2019-11-25:
Pretty much what I said above, for every upcoming full release the answer will be yes.

MichaelAtOz, 2019-11-26:
Was there a build with Qt4? If not, then 2019.05 is unsupported on XP, not 'upcoming' ones. ?

MichaelAtOz, 2019-11-26:
Sorry, I didn't see your Forum post. XP is dead.

nophead, 2019-11-26:
Is the forum no longer linked to the mailing list?

MichaelAtOz, 2019-11-26:
@nophead let me rephrase that, I didn't see tp's Forum post (the Qt not supporting XP bit) in my email until after I posted here (the Qt4 bit). I was probably making coffee when it arrived...
Rudolf mentioned something about delayed Forum visibility, which resolved. That didn't affect me.
Let me know if you have any issues.

andrixnet, 2020-12-20:
I understand all about upcoming future releases. No argument there.

However OpenSCAD download page says: 

System requirements: Windows XP or newer on x86 32/64 bit
OpenSCAD-2019.05
x86 (32-bit) - exe installer - 19 MB

Using an older system for several production tasks which runs very well and very stable, but can't use OpenSCAD 2019.05 on it. 
For the system in question: don't mess with it if it works, besides, specific hardware has no drivers for newer win.

I'd rather have a working OpenSCAD-2019.05 on XP then seeing the requirements text changed at https://www.openscad.org/downloads.html
For reasons such as rotate_extrude(angle = .... 

+1 E-Gwen.
Please post a separate download link for a 2019.05 build that does run on XP.

Thank you.

t-paul, 2020-12-20:
Sorry, there's not much choice at this point other than fixing the requirements statement. Nobody stepped up to provide a build that works on XP, so right now there simply is nothing to link for XP. We struggle to even build for all current OS versions, so supporting very old OSes without help is just not possible.

closed 2020-12-20 by t-paul

MichaelAtOz, 2020-12-20:
@andrixnet if you are not doing OpenSCAD command-line stuff, at least for Windows-7 (client) GUI works via RDP using a S/W OpenGL driver. If you have another PC with W7+ it could be worth a look.
I'm currently FreeRDP'ing from Debian to a W7 VM on a W7 PC for legacy support.

I'm not aware of anyone doing this with XP RDP, a quick google shows it is not [without complications](http://woshub.com/windows-xp-rdp-cant-connect-to-rds-on-server-2012/).