openscad/openscad #2977: Vimlike Mappings or User Defined Mappings

iasj, 2019-06-26:
Hi there. Great software.

It would be wonderful to be able to control the 3D view with some user defined keys. 

I know it is possible to control zoom with `ctrl+[` and `ctrl+]`, but I'd to have more control in order to not use the mouse at all.

As a Vim user for years, I'd like to control for example the draging of the view with the Vim keys as in

```text
       k

       ^
       |
h  <--- --->  l
       |
       v
   
       j
```

This would improve my workflow a lot once I'm always resting my right hand on top of those keys.

I'd also like for this feature to be user defined if possible. I'd like to put zoom action to keys like `i` and `o`.

In fact, I think it is a waste of fingers to use `ctrl+[number]` to change views. Why is it not as it is in Blender, where you can change view directly with number keys. It would make more sense.

Is this already possible in the 2019 version? If so, how?

MichaelAtOz, 2019-06-27:
Is this with the editor window hidden, as it has focus and gets plain keycodes? Or are you proposing a command mode toggle?

There is currently no keyboard remapping built-in.

I agree, some form of keyboard control of the preview window is needed, both translation and rotation on top of the existing zoom. Particularly helpful for [accessibility](https://en.wikipedia.org/wiki/Computer_accessibility).
The Ctrl+n is pretty baked-in, but if there was some toggle, maybe 'n' just could be used.

For info [this list](https://github.com/openscad/openscad/wiki/Keyboard-Shortcuts), is probably a little out of date, but it looks like k/h/l/j are almost unused, i/o not so much. See also the links, in particular to Qt, which has 4 OS flavours which need to be corralled. Choosing key combos is like herding cats on ice [the frozen water variety, I imagine the other would be difficult too..].

iasj, 2019-06-27:
I think a command mode toggle, such as Vim's, would be wonderful, but once I'm not aware about the guts of Openscad's code development, I'm afraid it would be a pain to implement such feature.

Instead, I'm proposing the idea of leaving all it's current default keyboard mappings untouched - so it doesn't change everyone's lives - and implement some sort of API of it's keyboard mappings for advanced users to tweak.

For example, we could have a hidden file in Home folder (.openscadrc) where we put the new mappings and possibly other settings, so every time openscad opens, it reads the file and sets itself based on it. 

And yes, without the command mode toggle, this would only make sense with the editor being hidden.

DLFW, 2019-12-10:
Yes, this is *the one* feature I'm waiting for! It would make the work so much more comfortable.

I think, reading some "hot keys" from a text file would be sufficient and easy to implement for somebody who is familiar with the code. I would even prefer a file based configuration over a "hot key dialog". Key bindings should allow to also use modifiers, especially ‹Ctrl›. Such key bindings should work globally, no matter where the GUI focus is currently located.

I ‐ and I think many others ‐ don't use the editor, but only the viewer and the console. So, vim *modes* are not really necessary. Only the "scene navigation" for the 3D model and showing/hiding, scrolling and maybe resizing the console would be sufficient (and a huge enhancement). Also remapping of the existing hot keys (rendering, STL export...) is of course highly desirable.

Sharma-Hrishabh, 2020-02-16:
I have some view on this like it's good to implement a user-defined keyboard shortcut feature. But I think, allowing users to control modifier 'Ctrl', can lead to trouble. Think of the shortcuts related to the toolbar, like show Edges, Axes, they use 'Ctrl'+'some number' key as a shortcut. We can't allow the user to remove that Ctrl modifier, as qscintilla will override, and instead of action triggering, that number would be typed in the editor. The same thing will happen if, user modify the shortcut by changing 'Ctrl' to 'Shift', as 'Shift' is occupied for special symbols. So, I think, the main modifier in Key-Sequence, here 'Ctrl' should remain fix and the user should be able to modify the second option in Key-Sequence. Any view on this?

t-paul, 2020-02-16:
We may need some sort of validation, e.g. also to prevent setting duplicate values and such. But in general we probably want users to be able to set combinations of modifier characters.

DLFW, 2020-02-16:
Well, I think most of the programs I use allow the use of modifiers for the key bindings. Ranger, alacritty, moc,... Even most GUI/mouse programs offer this freedom. I think it’s a pretty basic feature.

If the user overrides some combination like ‹CTRL›-a, well, then he probably wants to do exactly that and is fine with not having his complete code selected. Most people asking for this feature will anyway come from the VIM mindset and not use the editor at all. Personally, I would prefer to use single letters (h,  j, k, l, u, i,...) for navigating (rotating, zooming) the main view. But people are different and with having this freely configurable, you have the VIM/Tiling WM people happy as well as the mouse guys who just want something like ‹CTRL›-‹cursor-key› to quickly rotate the view while editing in the OpenSCAD editor.

A validation – IMO – is not really necessary. Last definition wins. Most programs do it like this. Of course one has to check the dotfile for valid syntax. :)

thehans, 2020-02-16:
I think what is missing maybe, is a separation of shortcuts depending on widget focus, such as whether the mouse is hovered over text editor or 3d view (or console).   Or in a more strict version, which one it has clicked in last.

Also some shortcuts should probably remain global to the overall window like F5, F6, etc.

Single letters would obviously be interpreted as text input by the text editor. Similarly arrow keys would move the text cursor.  If there was code to enable/disable different groups of shortcuts depending on focus, then that would be the ideal way to enable such 3d navigation shortcuts without interfering with text editor operation.

DLFW, 2020-02-17:
Yes, that would be one option, but I think it also has some downsides to use those “per-widget” hot-keys. For a user who works with the inbuilt editor, changing the focus each time he wants to inspect a change is maybe unnecessarily clumsy. Also, OpenSCAD doesn’t use a focus at the moment: Rotating the view with the mouse leaves the editor focused and active. So, this solution would even require changes of the existing interface. And: If the focus is not clearly visible, users always have to look twice to check where their “key command” is sent to.

As an alternative, just check input events when they enter. If they are a global shortcut, “consume” them. If not, delegate them further to the GUI framework so that the focused widget can do whatever is its standard behavior. Most GUI frameworks provide such a event hook, I guess. Furthermore, I assume that this is much easier to implement, less bug-prone, intuitive for the user and fulfills the purpose perfectly.

In the end, the solution must provide a comfortable way to use OpenSCAD as a viewer without touching the mouse. So if you go for for a “check focus” solution, it should be possible to quickly put the focus wherever the user wants it, without TAB panic and without using the mouse. The focus solution feels somehow cumbersome for me. I would go for a simple and straight “global key” solution. If I – as a user – assign a single letter to an action, then I can live with the result. If I recognize that it was a bad choice for me, I change it back.

itpropro, 2026-02-11:
> > I don't have F keys
> 
> Just out of curiosity, how did you manage that? I don't think I've seen such a keyboard in many many years.
> 
> > and I certainly don't want to take my hands off of my keyboard, how do I trigger a render?
> 
> On Windows, Alt+D, e will trigger a render. Alt+D pops the Design menu, and e is the shortcut keystroke for Render. In general, look for the underlined letter.

I just have a normal split keyboard. None of my split keyboard has F keys. I of course have a layer that I can switch to to use F keys, but I would be much more comfortable with being able to just remap.
I am probably just creating a completely new layer for openscad, but I am just more used to adjust the programs to my keyboard layout than the other way around. Didn't come across a software without remapable keybinds (I mean there is still Firefox, but no comment on that). But it's no big deal.