openscad/openscad #2978: Background Modifier [Version 2019]

iasj, 2019-06-26:
What the hell happened with the [background modifier](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_Language#Background_Modifier)?

* In version 2015 everything was ok

[image]

* In version 2019 it's gone!

[image]

this feature is the only thing that is preventing me to upgrade OpenSCAD. I just can't live without it.

nophead, 2019-06-26:
It still works outside of difference but inside difference it seems broken. For example this:

[image]

With F6 I just get the cylinder but F6 shows the cube minus the cylinder. According to the doc link you posted it should not take part in the difference.

iasj, 2019-06-26:
Even outside it's still weird. It only shows the intersection of the inside shape with the walls of the outsode shape.

2015 version was able to show the whole x-ray view of whatever was inside. It was perfect.

nophead, 2019-06-26:
Not sure what you mean. I can see the inside here:

[image]

iasj, 2019-06-26:
How did you do that? Is it permanen

iasj, 2019-06-26:
Still gonna be a problem if that doesn't work inside difference operator. I use it a lot.

nophead, 2019-06-26:
I just pressed  F5 once. It is permanent until I press F6.

Yes I can see why you might want to see the inside of a difference.

iasj, 2019-06-26:
Sorry for the delay.

I tried to reproduce and indeed it shows the way you described, but not when is inside a difference operator. Look:

* Pressing F12

[image]

* Pressing F9

[image]

Man. I rather have my balls kicked than be without this feature. Someone please fix it.

henwist, 2019-07-05:
Well I can add a few lines of code that can nail down the problem a bit and maybe pinpoint a specific area of interest in the latest master branch - namely F5 in combination with difference and % operator:
Notice that all code are just the same in both pic1 and pic2, but in pic2 the two rows inside the difference are switched with each other (and the % operator is belonging to the same 3d primitive as in pic1). 

-----

> //pic1 and F5:
translate([-25,0,0]) difference() {
       cube(5, center=true);
      %cube(10, center=true);
}
difference() {
  cylinder(h=30, r=6, center=true);
 %sphere(r=10);
}

-------
pic1 and F5:
[image]

---------

> //pic2 and F5:
translate([-25,0,0]) difference() {
          %cube(10, center=true);
          cube(5, center=true);
}
difference() {
  %sphere(r=10);
  cylinder(h=30, r=6, center=true);
}
------
pic2 and F5:
[image]

And voila - in pic1 the inner guts of difference() is showing up again. So here we have a problem that may be interesting to solve. Pressing F6 in both pic1 as well as pic2 does just show up the non prepended % primitive - regardless of which line it stands on. This fact shows that it is handled well in F6, so there should exist a solution for F5 as well.

MichaelAtOz, 2019-07-05:
Note the wiki says

> Because the marked subtree is completely ignored, it might have unexpected effects in case it's used, for example, with the first object in a difference(). In that case this object will be rendered in transparent gray, but it will not be the base for the difference()! 

So I'd say it now has become the base of the difference, what was ignored, isn't.

henwist, 2019-07-06:
@MichaelAtOz that's correct - but should the cube in the following example be rendered yellow?

```
difference()
{
     %union()
     {
         translate([5, 5, 0])
              cylinder(r1=2, r2=3, h=25); 
         
              translate([5, 15, 0])
              cylinder(r1=2, r2=5, h=25);

                       translate([15, 15, 0])
              cylinder(r1=4, r2=3, h=25);
         
                       translate([15, 5, 0])
              cylinder(r1=2, r2=7, h=25);
     }
     
   color("yellow")cube([20, 20, 20]);
}
```

-----

All I get is some yellow spots on the plane and the inside of the rods (rendered with F5):

-----

[image]

----

[image]

I was expecting the rods in gray as they are prepended with % (and they are in gray) and the base to be the cube (it shows up in F6 but not in F5 as the pictures shows).

nophead, 2019-07-06:
According to the documentation the union should be grey and not take part
in the difference, so the cube should be shown in yellow. Currently though,
a difference argument with % does take part in the difference with F5, but
not F6. So it has done a normal difference where the colour of the second
argument colours the faces it cuts.

henwist, 2019-07-07:
I think that F5 would give the same result as F6 - approximately as good as the real rendering. But I can imagine situations where the F5 rendering is not performing as well as F6 - just because it doesn't compute the volumes anyway near the precision needed to do the task accurate enough. Anyway  - here is the latest commits that make the cube yellow and the non rendered volumes gray.

iasj, 2019-07-09:
Now you guys lost me. Have you guys noticed that the modifier have changed behaviour in the last version? 

I see you prepending the modifier in the object tha's being subtracted, but that is not the case.

I often wanna see the object that is being subtracted, and therefore I apply the modifier in the first object in the difference. I usually do this when I'm positioning a nut that is going to live inside a piece, such as in this code:

```openscad
difference(){
  %cube(10,center=true);
  cylinder(d=2,h=2,$fn=6,center=true);
}
```

[image]

which still doesn't show the nut.

In version 2015 I was able to see the nut just by saving the file. It was simple! Just save and I could see the nut like Superman would.

It is important to note that I'm using a separate text editor to write the code. So if this feature still holds, it definitely needs a new documentation on how to use it.

nophead, 2019-07-09:
@iasj, this issue is discussing the fact that using % on the first operand of the difference operator has broken and is being fixed by @henwist to make it work as documented and how it worked in the last release.

I don't know when it broke between 2015 and now. I don't think anybody that uses the development snapshots uses it.

iasj, 2019-07-09:
Oh, I see now. Sorry for the despair. It just happens that this way of using the x-ray is the core of my desing patterns, such as follows:

```openscad

Holder();

module Holder(){
  
  difference(){
    shape();
    cut();
  }

  module shape(){
    // Build the shape
  }

  module cut(){
    // Position and build the holes to be cut
  }

}
```

so for example if I have to desing a holder for something, the code above would be my design pattern (the skeleton of each piece), and I often use the x-ray (which goes on the `shape();` call) to position and see the final places of each element in the `cut()` module.

I'll be more pacient now that I understood the trouble. Hope you guys find a solution. Thanks for the help.

iasj, 2019-09-19:
Any developments on this matter?

henwist, 2019-10-31:
Yes, the pull request is awaiting to be incorporated into the master branch. It is out of my control, but is works as expected and I hope that it will be endorsed by the team.
/br Henrik

iasj, 2019-11-06:
@henwist . Thank you very much. It will really improve my workflow. I hope they accept, because the openscad 2015's dependencies are already getting upgraded. So it is not going to work for long if I don't freeze all the dependencies. I don't wanna do that, so I hope your changes come through their filter. Appreciate the effort!

t-paul, 2020-06-05:
Closed via #2983.

closed 2020-06-05 by t-paul