openscad/openscad #2980: OpenSCAD (Not Responding)

agnunez, 2019-06-27:
Just intalled OpenSCAD-2019.05-x86-64 in windows 10 and I work with it for one day withiot problem. Next day, it opens and I can load my scripts or even examples, but just after few second it hangs showing (Not Responding). Re-installed both, msi and zip. Unplug 3dconnexion hand controller, just in case it was the cause. No error in Event Viewer. Only way to exit is to "End Task" in Task Manager.
Any place to look for error logging to solve it?

thehans, 2019-06-27:
If you run `openscad.com` from the command line does it print out anything to the console?
If openscad directory is not in your PATH, then you may need to give the full path: 
```"c:\Program Files\OpenSCAD\openscad.com"```

Also you can try `openscad.com --debug all` to which will log a ton of various messages to console.

Lastly, pasting the output from Help -> Library Info here might help narrow things down some.

agnunez, 2019-06-28:
I did openscad.com --debug all  in its directory within CMD. It opesn the program, pop up choice for projects works, choose anyone (Basic/logo.scad), and log many setting options lines to the console until it stop at the middle of word SettingWriter ....
....
SettingsWriter: SettingsWriter D: input/button0
SettingsWriter: SettingsWriter D: input/button1
SettingsWriter: SettingsWriter D: input/button2
Settin
...
full listing: https://drive.google.com/file/d/14r0Rc3Z8ZMahC2b270Yxv3YVEqxdGzVY/view?usp=sharing

Mouse change to waiting icon and (Not Responding) at top program name.

Thanks

MichaelAtOz, 2019-06-29:
One would question why it goes through SettinsWriter sooo many times...

> Unplug 3dconnexion hand controller

Which specific device, they have different driver parameters.
Do you have the 3Dconnection drivers loaded (ie for some Windows app)?

Were you running any of the 2019 pre-release snapshots with SpaceMouse, or came straight from 2015.03?

> Lastly, pasting the output from Help -> Library Info here might help narrow things down some.

thehans, 2019-06-29:
Another thing I would recommend trying is to disable/enable the different input drivers from OpenSCAD preferences and see if any of those are the cause (change and then restart OpenSCAD).  

If you can't get to the menu before freezing, you can toggle these in registry settings: 
run regedit.exe
Look under `Computer\HKEY_CURRENT_USER\Software\OpenSCAD\OpenSCAD\input` and try setting the enable* registry keys to false/true.

Some past issues with input driver related code have caused similar problems: #2854 #2578

agnunez, 2019-07-01:
> One would question why it goes through SettinsWriter sooo many times...
>
>
> Which specific device, they have different driver parameters.
> Do you have the 3Dconnection drivers loaded (ie for some [...]

I installed stable 2015.03. Previously my W10 was having the 3DConnexion SpaceMouse Compact edition. Do you recommend to test any pre-release version? Thanks

agnunez, 2019-07-01:
> Another thing I would recommend trying is to disable/enable the different input drivers from OpenSCAD preferences and see if any of those are the cause (change and then restart OpenSCAD).
>
> If you [...]

Sorry, but I don't have input key as children of OpenSCAD one. Only register keys are 3dview, advanced, desing, editor, launcher, view, window
Editor contains ctrlmousewheelzoom true. Switched of to false, and relaunched, with no effect.
Thanks

agnunez, 2019-07-01:
> If you run `openscad.com` from the command line does it print out anything to the console?
> If openscad directory is not in your PATH, then you may need to give the full path:
> `"c:\Program [...]

I tried a zip of 2019.05.19.ci2704 version with same frozen result. This time log stops later:
QSettingsCached: QSettings::setValue(): recentFileList = ''
QSettingsCached: QSettings::setValue(): view/hideEditor = 'false'
QSettingsCached: QSettings::setValue(): view/hideEditor = 'false'
QSettingsCached: QSettings::setValue(): view/hideCustomizer = 'false'
QSettingsCached: QSettings::setValue(): view/hideCustomizer = 'true'
QSettingsCached: QSettings::setValue(): view/hideConsole = 'false'
QSettingsCached: QSettings::setValue(): view/hideConsole = 'false'
QSettingsCached: QSettings::setValue(): view/hideConsole = 'false'
QSettingsCached: QSettings::setValue(): view/hideEditor = 'false'
QSettingsCached: QSettings::setValue(): view/hideToolbar = 'false'
QSettingsCached: QSettings::setValue(): view/hideCustomizer = 'true'
QSettingsCached: QSettings::setValue(): view/showAxes = 'true'
QSettingsCached: QSettings::setValue(): view/showScaleProportional = 'true'
QSettingsCached: QSettings::setValue(): view/orthogonalProjection = 'false'
QSettingsCached: QSettings::setValue(): design/autoReload = 'true'

but freeze anyway.  I cannot access to help->library menu because it hang before allowing me to...
thanks

MichaelAtOz, 2019-07-01:
> I installed stable 2015.03. Previously my W10 was having the 3DConnexion SpaceMouse Compact edition. Do you recommend to test any pre-release version? Thanks

Pre-release versions? No, not at this stage. As you came from 2015.03, you had not had any previous SpaceMouse OpenSCAD use.

@t-paul did you get exposed to the 'compact edition'? 

> Previously my W10 was having the 3DConnexion SpaceMouse Compact edition. 

So, as you were NOT using SpaceMouse for OpenSCAD, you were presumably using it for some other Windows app?
Reason I ask, to do so you must have loaded the 3dconnextion Windows Driver? Correct?

agnunez, 2019-07-01:
> Pre-release versions? No, not at this stage. As you came from 2015.03, you had not had any previous SpaceMouse OpenSCAD use.
>
> @t-paul did you get exposed to the 'compact edition'?
>
>
> So, as you were [...]

Yes, I use 3dconnexion with other CAD program and was installed previously to  install OpenScad   2015.03 in this pc. I use to work with older version in other computers. The weird thing is that I used for a day or two (without 3dconnexion) and following that, it came this "Not Responding" issue, without altering the configuration.

t-paul, 2019-07-01:
Not sure if all issues are solved. It should be detected and at least do something. The tests were on MacOS though: http://forum.openscad.org/Inputdriver-on-MacOS-td25528i20.html

MichaelAtOz, 2019-07-02:
You can get the Help->Library info from the command line `openscad --info`

> I don't have input key as children of OpenSCAD one. Only register keys are 3dview, advanced, desing, editor, launcher, view, window

`input` only appears after you check any `Driver selection` boxes in `Preferences/Axis`.
But `InputDriver` should be there.
It is hanging at `QSettingsCached: QSettings::setValue(): design/autoReload = 'true'` the next registry key is `editor`...it is not getting to output the `InputDriver` key. However, I just tried `--debug all` and got to `New`, t crashed, but theere was only a sort debug outout, so I think there is some output lost. But apparently it was working for a day, so I don't know why those keys are NOT there... 

So start simple.
Change  registry key `HKEY_CURRENT_USER\Software\OpenSCAD\OpenSCAD\design\autoReload` to false.
This will prevent auto F5 after loading a .scad.
Run OpenSCAD, select 'New', if it starts, do File/Exit. This hopefully writes a full set of registry keys.
Try again, type `cube();` press F5. Press F6. This is trying to see at what step it hangs.
If that works load something complex, F5, F6.

agnunez, 2019-07-02:
> You can get the Help->Library info from the command line `openscad --info`
>
>
> `input` only appears after you check any `Driver selection` boxes in `Preferences/Axis`.
> But `InputDriver` should be [...]
here it is:

openscad --info
OpenSCAD Version: 2019.05
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 12 CPUs 15.85 GB RAM
User Agent: OpenSCAD/2019.05 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.3.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.2.5
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.11.2
QScintilla version: 2.10.4
InputDrivers:
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.9.1
harfbuzz version: 2.0.2
lib3mf version: 1.8.0
Application Path: C:/Users/anunez/Downloads/OpenSCAD-2019.05-x86-64/openscad-2019.05
Documents Path: C:\Users\anunez\Documents
User Documents Path: C:\Users\anunez\Documents
Resource Path: C:/Users/anunez/Downloads/OpenSCAD-2019.05-x86-64/openscad-2019.05
User Library Path: C:/Users/anunez/Documents/OpenSCAD/libraries
User Config Path: C:\Users\anunez\AppData\Local/OpenSCAD
Backup Path: C:/Users/anunez/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  C:/Users/anunez/Documents/OpenSCAD/libraries
  C:/Users/anunez/Downloads/OpenSCAD-2019.05-x86-64/openscad-2019.05\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  C:/Windows/fonts
  C:/Users/anunez/.local/share/fonts
  C:/Users/anunez/.fonts

GLEW version: 1.12.0
OpenGL Version: 4.5.0 - Build 25.20.100.6374
GL Renderer: Intel(R) UHD Graphics 630
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: WGL
PNG generator: lodepng
OS info: Microsoft(TM) Windows(TM) 10 0 17134 0x309ecb4 amd64
Machine: 8664

agnunez, 2019-07-02:
> You can get the Help->Library info from the command line `openscad --info`
>
>
> `input` only appears after you check any `Driver selection` boxes in `Preferences/Axis`.
> But `InputDriver` should be [...]

I changed HKEY_CURRENT_USER\Software\OpenSCAD\OpenSCAD\design\autoReload to false, and pass the pop up with "New" (or any sample .scad), but the app windows appear and mouse already is irresponsible and cannot click on File/Exit or any place.
Thanks

MichaelAtOz, 2019-07-02:
> Resource Path: C:/Users/anunez/Downloads/OpenSCAD-2019.05-x86-64/openscad-2019.05

Are you trying to run it from that location?
Windows requires programs to be under `C:/Program Files`.

agnunez, 2019-07-02:
> > Resource Path: C:/Users/anunez/Downloads/OpenSCAD-2019.05-x86-64/openscad-2019.05
> 
> Are you trying to run it from that location?
> Windows requires programs to be under `C:/Program Files`.

I've tried everything, run with installer from c:/Program Files.. , run zip version from a directory.. run other versions..

agnunez, 2019-07-02:
> > Resource Path: C:/Users/anunez/Downloads/OpenSCAD-2019.05-x86-64/openscad-2019.05
> 
> Are you trying to run it from that location?
> Windows requires programs to be under `C:/Program Files`.

running from standard installation... same results.

C:\Program Files\OpenSCAD>openscad --info
OpenSCAD Version: 2019.05
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 12 CPUs 15.85 GB RAM
User Agent: OpenSCAD/2019.05 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.3.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.2.5
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.11.2
QScintilla version: 2.10.4
InputDrivers:
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.9.1
harfbuzz version: 2.0.2
lib3mf version: 1.8.0
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\anunez\Documents
User Documents Path: C:\Users\anunez\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/anunez/Documents/OpenSCAD/libraries
User Config Path: C:\Users\anunez\AppData\Local/OpenSCAD
Backup Path: C:/Users/anunez/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  C:/Users/anunez/Documents/OpenSCAD/libraries
  C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  C:/Windows/fonts
  C:/Users/anunez/.local/share/fonts
  C:/Users/anunez/.fonts

GLEW version: 1.12.0
OpenGL Version: 4.5.0 - Build 25.20.100.6374
GL Renderer: Intel(R) UHD Graphics 630
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: WGL
PNG generator: lodepng
OS info: Microsoft(TM) Windows(TM) 10 0 17134 0x309ecb4 amd64
Machine: 8664

MichaelAtOz, 2019-07-02:
hmmm, I'm grasping at straws now...
Try disabling any antivirus.
Try removing the 3Dconnexion drivers.
Backup the Registry, then delete the whole `HKEY_CURRENT_USER\Software\OpenSCAD` branch.
Try the 32bit version.
See if there is a newer display driver.

agnunez, 2019-07-02:
Great! At least you indicated me where to grasp. Finally is a interaction with NVDIA GTX 1060 second monitor. If I start with the hdmi of this external monitor unplugged, it works! Later, I can plug in it again and it works even in the second one. I even enabled HIDAPI and SpaceMouse is working. Thanks to your support.

closed 2019-07-02 by agnunez

MichaelAtOz, 2019-07-02:
Great.
For the next time, did you need to do anything, like removing the 3Dconnexion drivers, to fix it, or JUST unplug the 2nd monitor? 
Did you have this 2nd monitor unplugged when it worked for a day?

Could you post a Help/Library info from the GUI, it usually has more monitor info.

agnunez, 2019-07-02:
Hi,
I got another simpler solution. I actually have 3 monitors (2 external hdmi+displayport+laptop). The problem appears when OpenScad start on one of the external monitors (open the windows on that monitor). For some reason, the 1st day I was working, I shift OS windows to my hdmi monitor and close the app there, so next day I launched, it also tried to open the windows there and cause the issue. 
The first time I unplug the monitor, it again have to start the windows in the laptop internal display and works there, and once is working, I can shift the OS windows to my hdmi larger monitor without problem. So long story short, the solution is just to always close the OS app inside the laptop internal monitor, and open it up again on the same... later.. I can move it anywhere. thanks

This is the help/library info with 3 monitors:

OpenSCAD Version: 2019.05
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 12 CPUs 15.85 GB RAM
User Agent: OpenSCAD/2019.05 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.3.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.2.5
CGAL version, kernels: 4.11, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.11.2
QScintilla version: 2.10.4
InputDrivers: HidApiInputDriver (256f:c635 - 3Dconnexion Space Mouse Compact)*
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.9.1
harfbuzz version: 2.0.2
lib3mf version: 1.8.0
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\anunez\Documents
User Documents Path: C:\Users\anunez\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/anunez/Documents/OpenSCAD/libraries
User Config Path: C:\Users\anunez\AppData\Local/OpenSCAD
Backup Path: C:/Users/anunez/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
C:/Users/anunez/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
C:/Windows/fonts
C:/Users/anunez/.local/share/fonts
C:/Users/anunez/.fonts
GLEW version: 1.12.0
OpenGL Version: 4.5.0 - Build 25.20.100.6374
GL Renderer: Intel(R) UHD Graphics 630
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QGLWidgetGL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_depth_clamp_separate
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
[... 185 lines cut ...]
GL_IBM_texture_mirrored_repeat
GL_INTEL_conservative_rasterization
GL_INTEL_fragment_shader_ordering
GL_INTEL_framebuffer_CMAA
GL_INTEL_map_texture
GL_INTEL_multi_rate_fragment_shader
GL_INTEL_performance_query
GL_KHR_blend_equation_advanced
GL_KHR_blend_equation_advanced_coherent
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_texture_compression_astc_hdr
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

thanks

1klap, 2019-08-30:
I can confirm this. I have a PC with dual screen setup, and if the OpenSCAD window with the 3d-view is opened on the secondary screen, the program will hang.  Unplugging the second monitor helps. And as @agnunez writes, if the program is closed on one screen, Windows seems to remember to reopen it again on that one, so the monitor does not need to be unplugged every time, as long as OpenSCAD is closed on the main screen.

OpenSCAD Version: 2019.05
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 4 CPUs 15.93 GB RAM
User Agent: OpenSCAD/2019.05 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.3.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.2.5
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.11.2
QScintilla version: 2.10.4
InputDrivers:
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.9.1
harfbuzz version: 2.0.2
lib3mf version: 1.8.0
Application Path: c:/Program Files/OpenSCAD
Documents Path: C:\Users\<edited>\Documents
User Documents Path: C:\Users\<edited>\Documents
Resource Path: c:/Program Files/OpenSCAD
User Library Path: C:/Users/<edited>/Documents/OpenSCAD/libraries
User Config Path: C:\Users\<edited>\AppData\Local/OpenSCAD
Backup Path: C:/Users/<edited>/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  C:/Users/<edited>/Documents/OpenSCAD/libraries
  c:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  C:/WINDOWS/fonts
  C:/Users/<edited>/.local/share/fonts
  C:/Users/<edited>/.fonts

GLEW version: 1.12.0
OpenGL Version: 4.6.0 NVIDIA 436.02
GL Renderer: GeForce GTX 980 Ti/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: WGL
PNG generator: lodepng
OS info: Microsoft(TM) Windows(TM) 10 0 18362 0x309ec84 amd64
Machine: 8664

jhoffmann2, 2024-12-13:
> Great. For the next time, did you need to do anything, like removing the 3Dconnexion drivers, to fix it, or JUST unplug the 2nd monitor? Did you have this 2nd monitor unplugged when it worked for a day?
> 
> Could you post a Help/Library info from the GUI, it usually has more monitor info.

oh my god I had the same issue and this fixed it. What a strange bug! I wish the dev team luck finding it! 🤣