openscad/openscad #2981: 2019.05 not opening on Mac OS X 10.14.5

etjones, 2019-06-27:
Hi - I'm unable to run OpenSCAD.app for the 2019.05 Mac release. The message I get is `"OpenSCAD.app" can't be opened because the identity of the developer cannot be confirmed.`

* I have permissions set to allow all Apps, regardless of developer.
* Running the executable from the command line works. (`/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD scad_file.scad`) 
* Running from the command line with `open -a OpenSCAD scad_file.scad` yields the same developer signing message. 

* It looks like there's an absolute path referring to Marius' machine embedded in the build somewhere. The following lines are from my Console when trying to open OpenSCAD via a double-click on the app icon:
```
default	15:19:46.892402 -0500	XprotectService	File /Users/evan.jones/Applications/OpenSCAD.app/Contents/MacOS/../Frameworks/libqscintilla2_qt5.dylib failed on rPathCmd /Users/kintel/code/OpenSCAD/libraries/install/lib/QtPrintSupport.framework/Versions/5/QtPrintSupport
default	15:19:48.930348 -0500	loginwindow	-[PersistentAppsSupport applicationQuit:] | for app:OpenSCAD, _appTrackingState = 2
default	15:19:48.930378 -0500	loginwindow	-[PersistentAppsSupport applicationQuit:] | App: OpenSCAD, quit, updating active tracking timer
```
* Maybe this is related to #2805?

pchar, 2019-07-01:
run this command from your terminal and it will works ( for me )
sudo spctl --master-disable

dstulken, 2019-07-10:
For what it's worth, I'm running OpenSCAD 2019.05 successfully on MacOS 10.14.5, and I didn't have to run any special commands or launch from the command line to do so.  

I'm not saying there isn't a bug in the build... just that for any devs attempting to reproduce this, there must be some other variable in the environment that triggers/exposes it, not just the combination of this particular OpenSCAD and MacOS versions.

etjones, 2019-08-01:
Hm... Thanks for looking at this, y'all. I've just run into the same behavior on a second machine, but I haven't located what is different about my system that's causing the objections. I did already have `sudo spctl --master-disable` set as pchar recommended, and that didn't resolve the issue. Maybe my environment's set up to link to an incorrect version of QT? Anyway...

closed 2019-08-01 by etjones

Aniwax, 2020-12-04:
So what is the solution to this issue? I am still having similar problems.

etjones, 2020-12-04:
I never resolved it. My workaround has been to open the program from the command line. I made an alias something like:

`alias scad=“/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD $1”`

(May not be exactly correct, I’m on mobile, but you get the picture)

It’s not perfect, but it gets the job done.

haimoz, 2021-07-06:
This happened for me recently, with macOS version 11.3.1 (Big Sur) and OpenSCAD version 2021.01 (64 bit Intel).

What worked for me is the following:
1. try launching OpenSCAD, got prompted that the developer could not be verified
2. open "System Preferences", go to "Security & Privacy > General"
3. in the section on the bottom half of the panel ("Allow apps downloaded from:"), you will find a button to launch OpenSCAD anyway
4. click that button and I was all good to go

Hope this helps.