openscad/openscad #2984: Shift-Option-arrow keys not working in QScintilla editor on OSX

QuackingPlums, 2019-07-05:
Submitting this issue because I believe that overriding standard OSX behaviour should not be acceptable.

Option-Left and Option-Right are standard key combinations on OSX for moving the cursor a whole word to the left or right. The equivalent in Windows is Ctrl-arrow.

Adding the Shift modifier (Shift-Option-arrow) allows selection by word, just as Shift-Ctrl-arrow does in Windows. This is also a standard key combination.

Issue #2885 seems to imply that '_we have to hijack a reserved combination_' and that the Shift-Option-arrows '_looks reasonable_'. 
I disagree with this last statement. I have avoided activating the QScintilla editor because of this bug alone, and if the basic editor is going away in a future release then I don't agree that we can simply override a reserved key combination and inconvenience an entire subsection of users just because they are in the minority.

shish, 2019-08-18:
Also: cmd-left / cmd-right for start of line / end of line. I never realised that I used these shortcuts hundreds of times a day, every day, until I tried openscad and they didn't work...

QuackingPlums, 2022-01-20:
So the standard editor is no longer an option in 2021.01. It's a complete PITA not to have the editor behave like every other code editor on Mac. Is this not considered a worthwhile 'enhancement' any more?

t-paul, 2022-01-20:
No, the problem is that Apple does not allow development (or even just running MacOS) on anything other than Apple hardware. This seriously reduces options for working on MacOS specific issues.