openscad/openscad #2985: difference() doesn't commute

Vector5233, 2019-07-05:
Hi,

New to openSCAD, so I apologize if this is user error.

I run this and get desired behavior (image 1):

`translate([-150,-150]) {

difference() {
    color("yellow") square(22);
    color("yellow") translate([2,2]) square(18);
}
color("black", alpha = 0.05) square(size = [355,295], center = false);
color("yellow", alpha = 0.1) translate([-90,162]) square(size = [90,73],center=false);
}`  

I move the difference to a later point and it fails to subtract the inner square (image 2):

`color("black", alpha = 0.05) square(size = [355,295], center = false);

difference() {
    color("yellow") square(22);
    color("yellow") translate([2,2]) square(18);
}
color("yellow", alpha = 0.1) translate([-90,162]) square(size = [90,73],center=false);
}`

[image]
[image]

I can work around this by creating the difference first ... but why is it necessary??

t-paul, 2019-07-05:
I think that is just Z-Figthing ([see FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview?))

If I zoom in to that area, it start to tear the colors as the graphics card can't decide which of the planes to draw.
[image]

t-paul, 2019-08-18:
Closing as there's no feedback.

closed 2019-08-18 by t-paul