openscad/openscad #2986: Code causes crash

lukehutch, 2019-07-06:
The following code reliably causes openscad-2019.05-2.fc30.x86_64 to crash:

```
depths =
[[2.05, 2.6, 3.5, 4.55, 6.22, 8.15],
[2.03, 2.65, 3.75, 4.9, 6.63, 8.97],
[2.05, 2.88, 3.98, 5.55, 7.65, 10.45],
[2.07, 3.03, 4.16, 5.87, 8.21, 11.25],
[2.12, 3.03, 4.22, 5.95, 8.32, 11.35],
[2.11, 2.8, 3.98, 5.35, 7.55, 10.3],
[2.1, 2.6, 3.65, 4.75, 6.54, 8.71],
[2.1, 2.38, 3.15, 4.04, 5.62, 7.4],
[2.09, 2.37, 3.05, 3.9, 5.38, 6.94]];

function makePoint(r, c) =
    let(a = r/8*180, cc = c == 6 ? 5 : 6, radius = (cc * 2 + 1) / 11 * 45.75)
    [radius * sin(a), radius * cos(a), c < 6 ? depths[r][cc] : 0];

points = [ for (r = [0:8]) for (c = [0:6]) makePoint(r, c) ];

function pointIdx(r, c) = c + r * 7;

function makeFace(r, c) =
    [pointIdx(r, c), pointIdx(r, c + 1), pointIdx(r + 1, c + 1), pointIdx(r + 1, c)];

faces = [ for (r = [0:7]) for (c = [0:5]) makeFace(r, c) ];

polyhedron( points = points, faces = faces );
```

Weirdly, replacing `depths[r][cc]` with `depths[r][c]` (which should be equivalent) stops the code from crashing.

nophead, 2019-07-06:
It doesn't crash for me with OpenSCAD 2019.05. I get:
```
ERROR: Unable to convert point at index 0 to a vec3 of numbers, in file ./, line 25
WARNING: PolySet has degenerate polygons
```

This is because cc = 6 most of the time and indexes off the end of the row. I think perhaps you meant to write ```cc = c == 6 ? 5 : c```.

lukehutch, 2019-07-06:
Yes, that was the correct fix (fixed after pasting the code here). I assume Fedora is based on an old build, so I'll close this as already fixed. Thanks for testing.

closed 2019-07-06 by lukehutch

t-paul, 2019-07-06:
Probably related to #2965.