openscad/openscad #2987: Enhancement request - position-able rulers

thawkins, 2019-07-07:
It would be very cool to be able to render a set of x/y/z rulers at a given position 

The ruler function should use the current transformations (scale/rotate/translate) and render the ruler on the effective plane. 

something like:

translate([0,0,10]) rotate ([0,45,0]) scale([1.1.0.4]) 
                                   ruler(units=1);

units would be in base units as per everything else;

the code to render the existing axis could be pressed into operation to do this. 

Justification: be able to measure the distance between objects in the graphics context of an object.

t-paul, 2019-07-07:
Thingiverse has a couple of those, e.g. https://www.thingiverse.com/thing:1741429

MichaelAtOz, 2019-07-07:
There are two here http://forum.openscad.org/A-small-script-to-make-a-ruler-td22313.html 

(The second is by [Ronaldo](http://forum.openscad.org/suggestion-measurement-widget-td21206.html))

MichaelAtOz, 2019-07-07:
I just noticed this in MeshLab, Render/Show-quoted-box
[image]
The rulers adjust as you move around.
Maybe the code is transportable?

t-paul, 2019-07-07:
That MeshLab box looks nice.

I'm closing the issue as rendering a ruler should be fully supported by a user space library (including the use of `%` modifier so it gets automatically removed from the mesh generation).

closed 2019-07-07 by t-paul

lvta0909, 2020-01-08:
@thawkins @MichaelAtOz 
I find incovenient using the scripts to render the ruler on the models, mainly because its a try and error until I get the positioning of the ruler right, I did a quick patch to add an "auxiliary axes" that you can on the fly, on the viewport, using the mouse:
- ctrl+click right button + drag mouse left-right: mouse ruler on x axes
- alt+click right button + drag mouse left-right: mouse ruler on y axes
- ctrl+alt+click right button + drag mouse left-right: mouse ruler on z axes
- shift + one of the above: fine positioning = speed/10

You can enable/disable it on the "View->show auxiliary axes" menu option.
Here is a short video of a demo: https://youtu.be/a-LJpqMB1sc

The source code is on https://github.com/lvta0909/openscad

@t-paul Should I fire a merge request?