openscad/openscad #2988: File/Recent-files does not handle '&' in filename

MichaelAtOz, 2019-07-10:
at least on Windows.
`&` in menu titles flags the following character as the keyboard shortcut and displays it as underscored.
Having previously opened a file with `&` in the name, shows that file in the Recent-files list, wrongly.
[image]
Note the actual filename in the title bar.
Whatever the fix is, it should probably be also applied to the examples menu, as it is possible such a file could be there in the future.

RomitKumar, 2019-07-10:
Made pr #2989 solving the issue.

MichaelAtOz, 2019-07-12:
[image]

Fixed, thanks.

MichaelAtOz, 2019-07-12:
Should I close such issues or leave open so it gets included in a release, or add it to some list?

closed 2019-07-12 by MichaelAtOz

t-paul, 2019-07-13:
Yes, closing is fine. Using `(fixes #2988)' in the checkin comment would even close automatically when merged to master.