openscad/openscad #2990: IDE shortcut suggestions

dimitre, 2019-07-11:
first of all congrats on the project, beautiful tool for 3D design.
I understand each programming IDE have it's own particularities but most I use have standard shortcuts for commenting lines (Command + /) 
and ident, unident (Command + [  and Command + ])

I understand it is totally different here but it would be great to be able to customize it in more standard ways (sublimetext, vscode, xcode, atom, etc)

thehans, 2019-07-13:
For indent/unindent I've always used Tab/Shift-Tab which works fine in OpenSCAD and pretty much every other editor I can think of, not sure another shortcut is really necessary for that, but also probably wouldn't really hurt anything to support multiple ways of doing it.

t-paul, 2019-07-15:
Hmm, I guess as this is probably heading into the "make shortcuts customizable" direction, we can close as duplicate as #2977.