openscad/openscad #2994: echo() outputs wrong exponent

indazoo, 2019-07-14:
Hi
As a weekend OpenScad warrior, this may be my fault too, but my brain cells didn't triggered a facepalm reaction so far.

```
echo("Seems wrong: ", pow(3,630), pow(3,629));
for(p = [620:1:650])
	echo(p,pow(3,p),1000/pow(3,p));
```

The output of echo() seems wrong, the value itself in the variable seems to be right:

```
ECHO: "Seems wrong: ", 3.85825e+3, 1.28608e+3
ECHO: 620, 6.53398e+295, 1.53046e-293
ECHO: 621, 1.96019e+296, 5.10153e-294
ECHO: 622, 5.88058e+296, 1.70051e-294
ECHO: 623, 1.76418e+297, 5.66837e-295
ECHO: 624, 5.29253e+297, 1.88946e-295
ECHO: 625, 1.58776e+298, 6.29819e-296
ECHO: 626, 4.76327e+298, 2.0994e-296
ECHO: 627, 1.42898e+299, 6.99799e-297
ECHO: 628, 4.28695e+299, 2.33266e-297
ECHO: 629, 1.28608e+3, 7.77554e-298
ECHO: 630, 3.85825e+3, 2.59185e-298
ECHO: 631, 1.15748e+301, 8.63949e-299
ECHO: 632, 3.47243e+301, 2.87983e-299
ECHO: 633, 1.04173e+302, 9.59944e-3
ECHO: 634, 3.12518e+302, 3.19981e-3
ECHO: 635, 9.37555e+302, 1.0666e-300
ECHO: 636, 2.81266e+303, 3.55535e-301
ECHO: 637, 8.43799e+303, 1.18512e-301
ECHO: 638, 2.5314e+304, 3.95039e-302
ECHO: 639, 7.5942e+304, 1.3168e-302
ECHO: 640, 2.27826e+305, 4.38932e-303
ECHO: 641, 6.83478e+305, 1.46311e-303
ECHO: 642, 2.05043e+306, 4.87702e-304
ECHO: 643, 6.1513e+306, 1.62567e-304
ECHO: 644, 1.84539e+307, 5.41891e-305
ECHO: 645, 5.53617e+307, 1.8063e-305
ECHO: 646, 1.66085e+308, 6.02101e-306
ECHO: 647, inf, 0
ECHO: 648, inf, 0
ECHO: 649, inf, 0
ECHO: 650, inf, 0
```

Tested in:
OpenSCAD version 2019.05.19.ci2703 (git 92550de)
OpenSCAD version 2019.05 (official release)

MichaelAtOz, 2019-07-14:
Confirmed 2019.05 on Win7.
echo(pow(10,300)); // 1e+300 - so it  not just trailing 0's

I seem to recall a recent change in echo().

MichaelAtOz, 2019-07-14:
Probably #2950. Looks fixed from 2019.06.05.

t-paul, 2019-07-14:
Yep, that's fixed, nightly reports:
```
ECHO: "Seems wrong: ", 3.85825e+300, 1.28608e+300
```

closed 2019-07-14 by t-paul