openscad/openscad #2998: Another precision problem: pow()

indazoo, 2019-07-19:
Hi,
Let me first thank you for your support with my OpenScad issues. Unfortunately I have another one.
I just wanted to make a stepper motor clamp. Yes, on a weekend....well....
-----------------------------------------
Story, which you can skip  ;-)
1. type some clamp code
2. include a stepper motor library from the web
3. does not have my motor size => Improve stepper motor library (modularize it for stepper motor sizes)...why is this not already there .... argh!!
5. library creates warning on console. It uses type detection an in there somewhere by the len() statement which seem no longer to work for integers (t-Paul mentioned this in another issue here)
6. So I am fixing now the type detection...instead of the clamp
7. To be able to work without the len() statement I use string parsing a bit more in the type detection from the same lib.
---------------------------------------------------------
8. Because of fun, I improve string parsing and stumble about this:

echo(5.59543e-11 - 5.59543*pow(10,-11));
===>   -6.46235e-26    but should be zero

clamp still not finished.....

nophead, 2019-07-19:
Not sure why you expect floating point calculations to be exact.  They rarely are. In particular 0.1, and so 10e-11 cannot be represented exactly in binary just like 1/3 can't be represented in decimal. The only exact fractions in binary are those with denominators that are a power of two.

To replace len() you can use is_list(), but I agree it is a pain.

indazoo, 2019-07-19:
Thanks for the is_list() tip!
To the issue:
I expect it to be exact because I'm a user. As a user I have some math knowledge and this math knowledge tells me : "math is defined". There is no "feel good" math. In public may be a silly discussion about genders but math is luckily defined (at least that part of math I know). 
In each OpenScad "precision" discussion I found, for example the thing with the 0.2 steps in ranges, there is the "user" view and the "expert" view. The user wants it as logic would dictate, the expert argues with bits. I have also in my profession a hard time with people not accepting the customers needs. It is a common behaviour. I understand it. The work which must be done would require great effort. 
I only paid once a bounty for OpenScad and it is free. So, I'm no in a position to want a thing from you great OpenScad programmers.
But it is my position, that such things are not good for OpenScad and have created many frustrated ex-users.
I must look for a guide to compile OpenScad on Windows. The clamp must wait..... :-)

closed 2019-07-19 by indazoo

t-paul, 2019-07-19:
OpenSCAD is not in the position to completely rework understanding of how numbers are handled in computers. Yes, behavior of floating point numbers are not exact math and this is certainly an issue, but one that applies to pretty much all languages except maybe a few math focused solvers that can do symbolic evaluation.

The probably most detailed explanation of the status quo is probably [What Every Computer Scientist Should Know About Floating-Point Arithmetic ](https://docs.oracle.com/cd/E19957-01/806-3568/ncg_goldberg.html). I would not dare claiming I understand everything shown there, that's a bit over my math knowledge.

Just some random examples:
```
$ python3 -c "print(0.2 * 0.2)"
0.04000000000000001

$ perl -e 'print 0.04 - 0.2 * 0.2, "\n"'
-6.93889390390723e-18

$ yabasic -e 'print 1e-10 - 0.1^10'
-5.16988e-26
```

There is documentation about building on Windows in the Wiki: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Building_on_Microsoft_Windows

nophead, 2019-07-19:
The funny things is my Casio calculator that I have from being a child gets
these things right. I think it calculates to an extra digit of precision
and rounds to nearest at every step.Or perhaps it uses floating point BCD.
I wish somebody would make that into a library

indazoo, 2019-07-20:
Hi,
After hitting the THEY_HAVE_NO_VISUAL_STUDIO_SOLUTION wall, I was found by my wife in a fetal position in my man cave. Then I remembered to fight and found a solution with the good old round().
Search for round(in the code below)
This works for me.
It seems, when multipling a number with a smaller than zero value an error is being introduced.
Since I don't want to do a doctorate on this, I let it be. I mean, the clamp wants to be designed.

Sorry for the formatting, I have no clue why it breaks in between
n and NO I'wont fix this too. The clamp!!!
``//*************************************************************************************
//
// string Parsing
//
//*************************************************************************************

function parse_int(string="", base=10) = 
	_pre_parse_number(string=string, base=base, parse_decimal = false, parse_exponent=false);

function parse_real(string="", base=10) = 
	_pre_parse_number(string=string, base=base, parse_decimal = true, parse_exponent=false);

// parse_float is untested for base != 10
function parse_float(string="", base=10) = 
	_pre_parse_number(string=string, base=base, parse_decimal = true, parse_exponent=true);
		
function _pre_parse_number(string=undef, base=10, 
			parse_decimal = false, parse_exponent=false) =
	string == undef 
		? undef
	//: type(string) != "string"
	//	?	string
	: len(string)==0
		? undef
	: let(str_sign = _parse_sign(string=string))
	  len(str_sign[1]) == 0
		? undef
	:
		str_sign[0]*_parse_number(string=str_sign[1], base=base, 
		parse_decimal = parse_decimal, parse_exponent=parse_exponent,
		i=len(str_sign[1])-1 );
		
function _parse_sign(string="") = 
	string[0] == "-" 
		? [-1, substring(string,1,len(string)-1)] 
	:string[0] == "+" 
		? [1, substring(string,1,len(string)-1)]
	: [1, string]
	;

function _parse_number(string="", base=10, parsed_sign = undef,
		parse_decimal = true, parse_exponent=true, exponent = 0, exponent_i = -1,
		i=0, nb=0, decimalpoint_i = -1) = 
	i == -1
		//finishing work
		? (valid_i(exponent_i) 
			? //with an exponent then number shall be larger/smaller
			  let(len_decimals = decimalpoint_i < 0 
					? 0 //no decimals
					:exponent_i<0 
						? len(string)-decimalpoint_i-1
						: exponent_i-decimalpoint_i-1,
					prep_nb=round(nb * pow(base,len_decimals)),
					prep_exp = exponent - len_decimals
					
					)
				//prep_nb * pow(base,prep_exp)
				//there was a floating point precision issue  
				//==> use round()
				prep_exp >= 0 
					? prep_nb * pow(base,prep_exp)
					: prep_nb / pow(base,abs(prep_exp))
			: nb) 
	: string[i] == "."  	
		?	//current character is a point (decimal).
			let(len_dec = (valid_i(exponent_i)? exponent_i-1 : (len(string)-1))  -i)
			_parse_number(string=string, base=base, 
				parse_decimal = parse_decimal, parse_exponent=parse_exponent, 
				exponent = exponent, exponent_i = exponent_i,
				i=i-1,nb=(parse_decimal
						? //Convert whole number decimals to real decimals
							nb/pow(base,len_dec)
						: 0 //Decimals not needed
					),
				decimalpoint_i = i)
	: string[i]=="e" 
		? // A floating point number indicator without sign ==> e
		_parse_number(string=string, base=base, 
			parse_decimal = parse_decimal, parse_exponent=parse_exponent, 
			exponent = nb, exponent_i = i,
			i=i-1, nb=0,
			decimalpoint_i = decimalpoint_i)
	: (i-1 > 0 && string[i-1]=="e" && (string[i]=="+" || string[i]=="-"))
		? // A floating point number indicator with sign ==> +-e		
		_parse_number(string=string, base=base, 
			parse_decimal = parse_decimal, parse_exponent=parse_exponent, 
			exponent = nb *(string[i] == "-" ? -1 : 1),
			exponent_i = i-1,
			i=i-2, nb=0,
			decimalpoint_i = decimalpoint_i)
	:
		let(p = calc_pow(index = i, len_string = len(string), decimalpoint_i=decimalpoint_i, exponent_i=exponent_i))
		_parse_number(string=string, base=base, 
			parse_decimal = parse_decimal, parse_exponent=parse_exponent, 
			exponent = exponent, exponent_i = exponent_i,
			i=i-1, nb=nb+search(string[i],"0123456789ABCDEF")[0]*pow(base,p),
			decimalpoint_i = decimalpoint_i)
	;

function calc_pow(index = 0, len_string = -1, decimalpoint_i=-1, exponent_i=-1) =
	valid_i(decimalpoint_i)
		? // The decimal point has been detected, so the front digits are > 0
			abs(index-(decimalpoint_i-1))
	: valid_i(exponent_i)
		? // There is an exponent detected put no decimal point so far. 
		  // So it is either the decimals or a number without decimals
		  // Initially decimals are calculated as natural number and later on converted to decimals.
		  abs(index-exponent_i+1)
	: //No decimal, no exponent
		(len_string-1)-index
;
	
function valid_i(i) = (i>=0);
function valid_i_to_add(i) = i >= 0 ? i : 0;

strngs_test_parsing();
module strngs_test_parsing()
{
	tests = [
		["123", 123, "int",10],
		["101.", 101, "int",10],
		["131.0", 131, "int",10],
		["0.2223", 0, "int",10],
		[".2223", 0, "int",10],
		["-", undef, "int",10],
		["", undef, "int",10],
		["101", 5, "int",2],
		["101.01", 5, "int",2],
		["1011", 11, "int",2],
		["101", 5, "real",2],
		["101.01", 5.25, "real",2],
		["123", 123, "real", 10],
		["-139.939", -139.939, "real", 10],
		["1e8", 1e+8, "float", 10],
		["2e8", 2e+8, "float", 10],
		["3.0739e+12", 3.0739e+12, "float", 10],
		["3.073943e+2", 3.073943e+2, "float", 10],
		["5.59543e-11", 5.59543e-11, "float", 10],
		["-5.59543e-11", -5.59543e-11, "float", 10],
		["2.44e-3", 0.00244, "float", 10],
		["1e+3", 1000, "float", 10]
		];
	for(test = tests)
		_test_parse(test);
}

module _test_parse(test) 
{
	let(parsed = 
			(test[2] == "int"
				? parse_int(string=test[0], base = test[3])
			:test[2] == "real"
				? parse_real(string=test[0], base = test[3])
			:test[2] == "float"
				? parse_float(string=test[0], base = test[3])
			:
				parse_int(string=test[0], base = test[3])
			)
		)
	echo(parsed==test[1], test[0], "as" , test[2], "expected:", test[1], 
		"result:",parsed, parsed-test[1]);
}``

nophead, 2019-07-20:
I have to ask: what has parsing floating point numbers got to do with designing a clamp?

I imagine pow() is implemented with logarithms because it is general and handles fractional powers. For integer powers of 10 you might be better with a recursive function that does repeated multiplies by 10, That will give an integer as long as the value fits in a double mantissa, so should be exact. Then for negative powers take the reciprocal. That will be inexact but as close as it can be in floating point.

MichaelAtOz, 2019-07-20:
It's missing `substring()`.

nophead, 2019-07-20:
Given OpenSCAD cannot read files or accept user input, why would you ever need to parse strings to numbers?

MichaelAtOz, 2019-07-20:
Customiser?

nophead, 2019-07-20:
You can type floating point into cusomiser and OpenSCAD will parse it and assign it to a variable. I can't see why you would need to input it as a string and parse it to float yourself.

indazoo, 2019-07-20:
> its missing substring
Uhh Ohhh: The code is based on bolts library created by  Johannes Reinhardt.
I use common.scad and strings.scad
https://github.com/boltsparts/BOLTS

Best would be you download it an paste my code in strings.scad.

Below is the type detection which might be necessary too.
No guarantees....its a wild world..

//*************************************************************************************
//
// Type Testing 
//
//*************************************************************************************

//see http://rocklinux.net/pipermail/openscad/2013-September/005522.html
/*
echo("1", type([2, 3])== "vector");
echo("2", type([])== "vector" );
echo("3", type(-4)== "number" );  //int
echo("4", type(0)== "number" );//int
echo("5", type(0.3)== "number" );//float
echo("6", type(-0.3)== "number" );//float
echo("7", type(2e-6)== "number" );//float
echo("8", type(2e3)== "number" );//int
echo("9", type(2e-06)== "number" );//float
echo("10", type(2e+03)== "number" );//int
echo("11", type("a")== "string" );
echo("12", type("10")=="number" );
echo("12", type(" 10 ")=="number" );
echo("13", type(true)== "boolean" );
echo("14", type(false)== "boolean" );
echo("15", type("true")== "boolean" );
echo("16", type("false")== "boolean" );
echo("17", type(undef)=="undef" );
echo("18", type([1:2:10])== "range" );// 
echo("19", type([1:1.5:10])== "range"); //
echo("20", type("[44 : 3 : 47]")== "range" );
echo("20", type("[44:3:47]")== "range" );
echo("21", type("[4;6;ewerw]")== "string" );
echo("22", type(12345678901234500)== "number" );// int
echo("23", type("M2.5")== "string" );
echo("24", type("")== "string" );
echo("25", type([])== "vector" );
echo("26", type("bumpf")== "string" );
echo("27", type([1:10])== "range" );// 

echo(type("[44 : 3 : 47]"));
echo(split("44:3:47",":"));
echo(type("10"));
echo(type("a"));
*/
function type(P) =
	P==undef 
		? "undef"
	:(P == true || P == false)
		? "boolean"
	: let(asStr = str(P))
	  len(asStr) == 0 
		? "string" //empty string
	: (substring(asStr, 0, 1) == "[" && substring(asStr, len(asStr)-1,1) == "]" )
		// A range or vector or string like "[***]"
		? len(asStr) == 0 
			? "vector"
		: (P + [1] == undef)
			? (	// A string or range
				let(content = substring(asStr, 1, len(asStr)-2),
					splitted = split(content,":")
				)
				( (len(splitted) == 3 || len(splitted) == 2)
					&&  all( [ for(x=splitted) (type(parse_float(trim(x)))=="number") ] )
				)
					? "range"
					: "string"
				) //end: A string or range
			:	"vector"  
	: str(P)==P  //Attention: str("1.5") equals to "1.5"
				// str([1:5]) equals to "[1:5]"
		? (str(P)=="false" || str(P)=="true") 
			? "boolean" 
			: let(p2=parse_float(trim(str(P))))
			  p2 == undef 
				? "string"
				: "number"
	:
	(abs(P)+1>abs(P)) 
		? "number" //for this implementation i use "number" instead of "float" 
	: floor(P)==P 
		? "number"
	//Unknown, most likely a string
	: "string"
	;

indazoo, 2019-07-20:
> why would you ever need to parse strings to numbers?

We users do things the designers never imagined !!!

indazoo, 2019-07-20:
Nophead I missed your question:
>what has parsing floating point numbers got to do with designing a clamp?

The thing was, the stepper motor library I used made use of the bolts library and the bolts library used len(8) or something and this no longer worked in the newer OpenScad versions and len(8) was used in its type() function and the type function used parse_int(). Then I coded a bit more than was really needed. I admit that.

indazoo, 2019-07-20:
@MichaelAtOz 
Thanks for the correction!

nophead, 2019-07-20:
The release version now has:
```
is_undef
is_bool
is_num
is_string
is_list
```

built in functions, which should replace or simplify the type() function.

indazoo, 2019-07-20:
good.

t-paul, 2019-07-20:
Note that there is intentially no `is_range`, see #1584/#1583. I'd recommend not relying on a difference between range and list/vector.

indazoo, 2019-07-21:
I forgot, the strings library was from 
https://www.thingiverse.com/thing:526023
Unfortunately it was not in MCAD

thehans, 2019-07-21:
@indazoo If you want to build OpenSCAD using Visual Studio, I created a solution for this a while ago.

See here for the original build instructions: #2814 .  I used the free Visual Studio Community 2017 for this (As I understand there is now a 2019 version, which I have not tried yet.)

I haven't run the VS build in a few months, so its possible that other changes to the project may have broken compatibility in the meantime.  If you have any questions/comments on this build process, please put them in the current tracking issue here: #2857