openscad/openscad #2999: Color is not working in render mode / ubuntu (2019.01-RC2)

reubenfirmin, 2019-07-20:
Here is my model:

```
color([0,1,0.99]) {
    difference() {
        union() {
            cube([20, 120, 20]);
            translate([60, 0, 0]) {
                cube([20, 120, 20]);
            }
            cube([80, 120, 5]);
            translate([0, 120, 0]) {
                cube([80, 20, 20]);
            }
        }   
        translate([20, 100, 0]) {
            cube([40, 20, 5]);
        }
    }
}
```

In preview I get:

[image]

In render I get:

[image]

t-paul, 2019-07-20:
Color is currently only supported in Preview mode.

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#color

closed 2019-07-20 by t-paul

reubenfirmin, 2019-07-20:
So why not support it in image export? Just because STL has a limitation doesn't mean that you should cripple the application.

t-paul, 2019-07-20:
It's simply not implemented. It hopefully will be at some point, right now the information is simply not there after the mesh calculation process.

MichaelAtOz, 2019-07-20:
> So why not support it in image export?

You can File/Export/as-image after F5 preview.

nophead, 2019-07-21:
You can also do color(...) render() object() in preview mode, which will
give exactly the same view as render but with colour, but will be just as
slow.