openscad/openscad #3000: Object not rendered under specific circumstances

Erhannis, 2019-07-20:
Consider the following code.

```
cube();
linear_extrude(height=10) {
  for (i = [0:8]) {
    rotate([0,0,i*360/9])
      polygon(points=[[0,0],[0,10],[1,9]]);
  }
};
```

When rendered, it should display a series of blades resembling a fan, an a small cube at the base near the center.  Instead, only the cube is rendered.  (If you preview it, the blades ARE rendered correctly.)  If you comment out the cube, however, the blades are correctly rendered.  Experimentation shows that removing the cube, extrusion, for-loop, rotation, reference to `i `, or `polygon` all dispel the bug.  Even swapping the nesting of the for-loop and the extrusion fixes it (which is how I've worked around it in my project).  It's such a brittle bug, it makes me a little concerned there might be some fundamental flaw in the underlying mechanics that allow all these factors to combine in such a way.  I've tried it in 2019.05.16, and commit 6cc7e911.  Any idea what's up?

[image]
[image]

MichaelAtOz, 2019-07-21:
When I F6, I get
```
ERROR:  error in CGAL_Nef_polyhedron3(): 
CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h 
Line: 326
```

This is because CGAL has problems in the centre singularity, it can't tell whether the centre point of each object is inside or outside.

> swapping the nesting of the for-loop and the extrusion fixes it

Looks like it is easy to tell in 3D...

You are likely to get similar CGAL errors whenever you just butt-up two objects, face-to-face, as it doesn't know whether they should be two objects or joined. It is good practice to either have a gap or embed one object into the surface of the other, by a tiny amount (e.g. 0.001 or nophead will say use 1/128 as it is a factor of 2 so doesn't get floating point issues).

In your case, avoid the singularity, move the origin out by a tiny amount and fill it with a skiny cylinder or something.

Note that the cube() triggers the problem, as OpenSCAD only calls CGAL when it needs to do CSG, so if there is nothing to union, difference, intersect etc, it doesn't get called.

nophead, 2019-07-21:
You can but objects face to face in CGAL as long as they touch exactly
after floating point considerations. What you can't do is have 3D shapes
that share one edge. That is physically impossible and is non-manifold.
Before the linear_extrude the shape made is a self intersecting polygon
which doesn't make a valid 3D shape when extruded because the centre is
infinitely thin.

Erhannis, 2019-07-23:
Hmm.  Is there an in-code fix for this?  Seems like the logical solution to the root problem would be to leave the two objects separate, as they would be were their edges just a tiny bit apart.  It's all well and good to say "move the objects slightly", except that NOW I'm dealing with a 400 line file, and to say "two things are touching and they shouldn't be" isn't especially helpful.

MichaelAtOz, 2019-07-23:
> to say "two things are touching and they shouldn't be" isn't especially helpful. Not to mention, adding small slop values to satisfy the software isn't a very clean solution.

It's called SCIENCE, mathematics in particular. It is not to satisfy the software.

Non-manifold and self-intersections will be the bane of your 3D future if you do not work to avoid them.

[This is a good explanation](http://3dprintingninja.blogspot.com/2014/07/non-manifolds-your-worst-nightmare.html).

The technicality is described in this [CGAL manual](https://doc.cgal.org/latest/Polygon_mesh_processing/index.html#coref_bolop_subsec), as:

> The corefinement of two triangulated surface meshes can naturally be used for computing Boolean operations on volumes. Considering two triangulated surface meshes, each bounding a volume, the functions 
CGAL::Polygon_mesh_processing::corefine_and_compute_union(), CGAL::Polygon_mesh_processing::corefine_and_compute_intersection() and CGAL::Polygon_mesh_processing::corefine_and_compute_difference() 
respectively compute the union, the intersection and the difference of the two volumes. ...
>
> There is no restriction on the topology of the input volumes. However, there are some requirements on the input to guarantee that the operation is possible. First, the input meshes must not self-intersect. Second, the operation is possible only if the output can be bounded by a manifold triangulated surface mesh. In particular this means that the output volume has no part with zero thickness. Mathematically speaking, the intersection with an infinitesimally small ball centered in the output volume is a topological ball. At the surface level this means that no non-manifold vertex or edge is allowed in the output. For example, it is not possible to compute the union of two cubes that are disjoint but sharing an edge.
>

While nophead is technically correct on sharing one edge, the common first contact with this is butting two faces, the limitation of 'touch exactly after floating point considerations' (in particular any use of rotation) will get you when it is most inconvenient. So it is a real good rule of thumb to make sure faces don't just 'touch', either have a gap or embed one inside the other.

> NOW I'm dealing with a 400 line file

Hopefully you are (will) use modules to break up your code, it's much easier to change a module once, than the same chunk of code in 10 places.

MichaelAtOz, 2019-07-23:
[This has some good examples](http://www.maruf.ca/files/caadoc/CAATobTechArticles/TopoConcepts.htm#Manifold) of manifoldness in 2D & 3D, specifically your 2D polygons meeting at one point.

jbash, 2019-11-13:
I ran into something like this, found a few odd allusions to similar things, and I don't think it's an issue of anything lining up exactly with anything else. Maybe it's a real CGAL bug? Or at least maybe there's a class of designs  similar to the original report that trigger a real CGAL bug?

In OpenSCAD 2009.05, the following code fails with "ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: cet->get_index() == ce->twin()->get_index() File: /usr/include/CGAL/Nef_3/SNC_external_structure.h Line: 1220":

```
union () {
    cylinder (d=25, h=100, $fn=36);
    translate ([0,0,0.1])
        linear_extrude (height=99.8, twist=60, $fn=12)
            translate ([12,0,0])
                square (size=[25,1],center=true);
}
```

That previews OK; it just doesn't render. The edge of the extruded vane is well inside the cylinder, and the top and bottom of the extruded vane are half a unit below/above the top or bottom of the cylinder.

Reducing the twist to any integer number of degrees up to 12 makes that code work. 13, 14, and 15 degrees don't work, but 16 through 22 *do* work. Nothing above that through 35 works, and I got tired there.

Going back to 60 degrees of twist, setting $fn for the extrude to 72 makes it work. I didn't try any other values.

Maybe a point is lining up somewhere... but I would think that every point in a 12-point extrusion would also be in a 72-point extrusion, and a 13-degree angle isn't a very good candidate to make any numbers identical.

nophead, 2019-11-13:
It works if you add slices = 20.

It isn't a CGAL bug. The problem is with so few slices linear_extruding a thin object causes self intersections due to the course  triangulation.

NickDeBeenSAE, 2023-10-27:
Shapes don't show up at all for me.

I even copied the code at the top of the issue here, and nothing, it never renders.

kintel, 2024-03-31:
Status as of 2024-03-24:
* First example: Still broken with CGAL, but works with Manifold
* example from jbash: Works with both CGAL and Manifold

closed 2026-03-30 by github-actions[bot]