openscad/openscad #3021: No 3d viewer

AppDevMichael, 2019-07-30:
There seems to be a problem with the 3d viewer in 2019.05 for me at least.

[image]

AppDevMichael, 2019-07-30:
I have tried reinstalling and restarting OpenSCAD. However this did nothing. 

I was working yesterday.

t-paul, 2019-07-30:
We are talking Windows? If it was working yesterday, it could be either some update of the graphics driver causing issues, or maybe the registry settings got messed up.

Maybe have a look at `HKEY_CURRENT_USER\Software\OpenSCAD` and temporarily rename that to see if an empty config restores things.

Otherwise please post the Help -> Library Info which could help narrowing things down.

AppDevMichael, 2019-07-30:
Renaming that registry fixed it thanks!

Any idea why this happened?

AppDevMichael, 2019-07-30:
This is the Library and Build Information:

```
OpenSCAD Version: 2019.05
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 8 CPUs 15.89 GB RAM
User Agent: OpenSCAD/2019.05 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.3.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.2.5
CGAL version, kernels: 4.11, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.11.2
QScintilla version: 2.10.4
InputDrivers: 
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.9.1
harfbuzz version: 2.0.2
lib3mf version: 1.8.0
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\micha\Documents
User Documents Path: C:\Users\micha\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/micha/Documents/OpenSCAD/libraries
User Config Path: C:\Users\micha\AppData\Local/OpenSCAD
Backup Path: C:/Users/micha/Documents/OpenSCAD/backups
OPENSCADPATH: C:\Users\micha\Documents\OpenSCAD\libraries
OpenSCAD library path:
C:/Users/micha/Documents/OpenSCAD/libraries
C:/Users/micha/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
C:/WINDOWS/fonts
C:/Users/micha/.local/share/fonts
C:/Users/micha/.fonts
GLEW version: 1.12.0
OpenGL Version: 4.4.0 - Build 22.20.16.4708
GL Renderer: Intel(R) HD Graphics 630
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QGLWidgetGL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_depth_clamp_separate
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
[... 197 lines cut ...]
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_ARB_buffer_region
WGL_ARB_context_flush_control
WGL_ARB_create_context
WGL_ARB_create_context_profile
WGL_ARB_create_context_robustness
WGL_ARB_extensions_string
WGL_ARB_framebuffer_sRGB
WGL_ARB_make_current_read
WGL_ARB_multisample
WGL_ARB_pbuffer
WGL_ARB_pixel_format
WGL_ARB_pixel_format_float
WGL_EXT_create_context_es2_profile
WGL_EXT_create_context_es_profile
WGL_EXT_depth_float
WGL_EXT_extensions_string
WGL_EXT_pixel_format_packed_float
WGL_EXT_swap_control
WGL_INTEL_cl_sharing
WGL_NV_DX_interop
WGL_NV_DX_interop2
```

AppDevMichael, 2019-07-30:
With a bit of debugging I have come to the conclusion that the 3d viewer will not work on a second, or third monitor, it only works on the main monitor. 

I don't know how to fix this as of yet.

t-paul, 2019-07-30:
That sounds like a Qt5 bug, there were similar issues with the application just locking up. This variant of messing up the window state is new though.

t-paul, 2019-07-30:
Hmm, only found https://bugreports.qt.io/browse/QTBUG-70721 so far which does not really match the issue here. Are the monitors configured for different scaling factors?

AppDevMichael, 2019-07-30:
The main display was set at a custom scale factor however if I turn it off and set them all to 100% scale, the issue still remains.

closed 2026-04-02 by github-actions[bot]